margelo/react-native-vision-camera #3187: 🔧 Unable to find a specification for `react-native-worklets-core` depended upon by `VisionCamera/FrameProcessors`

phazei, 2024-09-11:
### How were you trying to build the app?

pod install

### Full build logs

```tsx
> pod install

RNFBAnalytics: Using Firebase/AnalyticsWithoutAdIdSupport pod in place of default Firebase/Analytics
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] $VCEnableLocation is not set, enabling CLLocation APIs by default...
[VisionCamera] $VCEnableFrameProcessors is not set, enabling Frame Processors if Worklets is installed...
[VisionCamera] react-native-worklets-core found at ., Frame Processors are enabled!
Auto-linking React Native modules for target `Abstract`: RNBootSplash, RNCAsyncStorage, RNCClipboard, RNCMaskedView, RNConvertPhAsset, RNDateTimePicker, RNDeviceInfo, RNFBAnalytics, RNFBApp, RNFBCrashlytics, RNFBDynamicLinks, RNFBMessaging, RNFS, RNFileViewer, RNGestureHandler, RNLocalize, RNNotifee, RNPermissions, RNReanimated, RNSVG, RNScreens, RNStaticSafeAreaInsets, RNVectorIcons, VisionCamera, react-native-audio, react-native-biometrics, react-native-cameraroll, react-native-contacts, react-native-detector, react-native-document-picker, react-native-heic-converter, react-native-maps, react-native-mov-to-mp4, react-native-netinfo, react-native-orientation-locker, react-native-paste-input, react-native-restart, react-native-safe-area-context, react-native-secure-key-store, react-native-startup-time, react-native-video, and toolbar-android
Framework build type is static library
[Codegen] Adding script_phases to React-Codegen.
[Codegen] Generating ./build/generated/ios/React-Codegen.podspec.json
[Codegen] Analyzing /Users/adamsegal/Projects/evertel-frontend-nx/apps/evertel-rn/package.json
[Codegen] Searching for codegen-enabled libraries in the app.
[Codegen] The "codegenConfig" field is not defined in package.json. Assuming there is nothing to generate at the app level.
[Codegen] Searching for codegen-enabled libraries in the project dependencies.
[Codegen] Found @react-native-async-storage/async-storage
[Codegen] Found @react-native-clipboard/clipboard
[Codegen] Found @react-native-community/datetimepicker
[Codegen] Found react-native
[Codegen] Found react-native-document-picker
[Codegen] Found react-native-gesture-handler
[Codegen] Found react-native-reanimated
[Codegen] Found react-native-safe-area-context
[Codegen] Found react-native-screens
[Codegen] Found react-native-svg
[Codegen] Found react-native-vector-icons
[Codegen] Found @react-native-camera-roll/camera-roll
[Codegen] Found react-native-permissions
[Codegen] Found @mattermost/react-native-paste-input

[Codegen] >>>>> Searching for codegen-enabled libraries in react-native.config.js
[Codegen] Processing rnasyncstorage
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, macos, tvos, visionos for rnasyncstorage
[Codegen] Processing rnclipboard
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, macos, tvos, visionos for rnclipboard
[Codegen] Processing RNDateTimePickerCGen
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, visionos for RNDateTimePickerCGen
[Codegen] Processing FBReactNativeSpec
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Processing rncore
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Processing rndocumentpicker
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios for rndocumentpicker
[Codegen] Processing rngesturehandler_codegen
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, macos, tvos, visionos for rngesturehandler_codegen
[Codegen] Processing rnreanimated
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, macos, tvos, visionos for rnreanimated
[Codegen] Processing safeareacontext
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, tvos, visionos for safeareacontext
[Codegen] Processing rnscreens
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, tvos, visionos for rnscreens
[Codegen] Processing rnsvg
[Codegen] Searching for podspec in the project dependencies.
[Codegen] Supported Apple platforms: ios, macos, tvos, visionos for rnsvg
[... 285 lines cut ...]
```

### VisionCamera Version

4.5.3

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

phazei, 2024-09-11:
We do not use or need Frame Processors, so we don't want them enabled, but it seems to be thinking it's finding it and enabling it, then failing because it's not there.

`pod install --repo-update` didn't help

phazei, 2024-09-11:
I debugged it and discovered the issue.  It's a bug in the Ruby because depending on how it's run, sometimes it returns "undefined" but other times it returns `undefined` which is not the string.

I modified the podspec like so:
```
def Pod::getWorkletsLibraryPath
  installation_root = Pod::Config.instance.installation_root.to_s
  Pod::UI.puts "[VisionCamera Debug] Installation root: #{installation_root}"
  
  command = "cd \"#{installation_root}\" && node --print \"try { console.log(require.resolve('react-native-worklets-core/package.json')) } catch(e) { console.log('PACKAGE_NOT_FOUND') }\""
  Pod::UI.puts "[VisionCamera Debug] Executing command: #{command}"

  output = `#{command}`
  Pod::UI.puts "[VisionCamera Debug] Command output: #{output}"
  Pod::UI.puts "[VisionCamera Debug] Output lines: #{output.strip.lines.map(&:inspect).join(', ')}"
  
  if output.lines.any? { |line| line.strip == 'PACKAGE_NOT_FOUND' }
    Pod::UI.puts "[VisionCamera Debug] Package not found"
    return nil
  elsif output.lines.any? { |line| line.include?('node_modules/react-native-worklets-core/package.json') }
    path = output.lines.find { |line| line.include?('node_modules/react-native-worklets-core/package.json') }.strip
    Pod::UI.puts "[VisionCamera Debug] Package found at: #{path}"
    return File.dirname(path)
  else
    Pod::UI.puts "[VisionCamera Debug] Unexpected output, assuming package not found"
    return nil
  end
end
```

It outputs:
```
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] $VCEnableLocation is not set, enabling CLLocation APIs by default...
[VisionCamera] $VCEnableFrameProcessors is not set, enabling Frame Processors if Worklets is installed...
[VisionCamera Debug] Installation root: /Users/username/Projects/frontend-nx/apps/evertel-rn/ios
[VisionCamera Debug] Executing command: cd "/Users/username/Projects/frontend-nx/apps/rn/ios" && node --print "try { console.log(require.resolve('react-native-worklets-core/package.json')) } catch(e) { console.log('PACKAGE_NOT_FOUND') }"
[VisionCamera Debug] Command output: PACKAGE_NOT_FOUND
undefined
[VisionCamera Debug] Output lines: "PACKAGE_NOT_FOUND\n", "\e[90mundefined\e[39m"
[VisionCamera Debug] Package not found
[VisionCamera] react-native-worklets-core not found - Frame Processors are disabled!
```

The bug in the original code is "output.strip" doesn't actually work properly in all cases when it outputs multiple lines which can happen.  I tried it without outputting text, but it was inconsistent.  Please fix, thank you.

neha-s999, 2024-10-30:
I am still facing this error 
❌  Pods/react-native-vision-camera-v3-text-recognition: 'VisionCameraProxy.h' file not found with <angled> include; use "quotes" instead
   └─/Users/tribalscale/script-runner-driver-mobile/node_modules/react-native-vision-camera-v3-text-recognition/ios/VisionCameraV3TextRecognition.m:9:9
❌  Pods/react-native-vision-camera-v3-text-recognition: 'condition_variable' file not found
   └─/Users/tribalscale/script-runner-driver-mobile/node_modules/react-native-worklets-core/cpp/dispatch/WKTDispatchQueue.h:3:10
› Executing expo-modules-core Pods/ExpoModulesCore » Copy generated compatibility header

This is my package.json
{
  "name": "scriptrunner-driver-mobile",
  "main": "expo-router/entry",
  "version": "1.0.0",
  "scripts": {
    "start": "expo start",
    "reset-project": "node ./scripts/reset-project.js",
    "android": "expo run:android",
    "ios": "expo run:ios",
    "web": "expo start --web",
    "test": "jest --watchAll",
    "lint": "eslint .",
    "prepare": "husky"
  },
  "jest": {
    "preset": "jest-expo"
  },
  "dependencies": {
    "@aws-amplify/react-native": "^1.1.6",
    "@aws-amplify/ui-react-native": "^2.2.14",
    "@expo/vector-icons": "^14.0.2",
    "@react-native-async-storage/async-storage": "^2.0.0",
    "@react-native-community/geolocation": "^3.4.0",
    "@react-native-community/netinfo": "^11.4.1",
    "@react-navigation/native": "^6.1.18",
    "aws-amplify": "^6.6.4",
    "axios": "^1.7.7",
    "expo": "~51.0.28",
    "expo-camera": "^15.0.16",
    "expo-constants": "~16.0.2",
    "expo-font": "~12.0.9",
    "expo-image": "^1.12.15",
    "expo-image-manipulator": "^12.0.5",
    "expo-linking": "~6.3.1",
    "expo-location": "^17.0.1",
    "expo-router": "~3.5.23",
    "expo-splash-screen": "~0.27.5",
    "expo-status-bar": "~1.12.1",
    "expo-system-ui": "~3.0.7",
    "expo-web-browser": "~13.0.3",
    "react": "18.2.0",
    "react-dom": "18.2.0",
    "react-native": "0.74.5",
    "react-native-config": "^1.5.3",
    "react-native-error-boundary": "^1.2.5",
    "react-native-geolocation-service": "^5.3.1",
    "react-native-gesture-handler": "~2.16.1",
    "react-native-get-random-values": "^1.11.0",
    "react-native-image-picker": "^7.1.2",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-maps": "^1.18.0",
    "react-native-paper": "^5.12.5",
    "react-native-permissions": "^4.1.5",
    "react-native-reanimated": "~3.10.1",
    "react-native-safe-area-context": "^4.10.5",
    "react-native-screens": "3.31.1",
    "react-native-signature-capture": "^0.4.12",
    "react-native-svg": "^15.8.0",
    "react-native-svg-transformer": "^1.5.0",
    "react-native-vision-camera": "4.0.5",
    "react-native-vision-camera-v3-text-recognition": "^1.1.1",
    "react-native-web": "~0.19.10",
    "react-native-webview": "^13.12.2",
    "react-native-worklets-core": "^1.3.3"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@types/jest": "^29.5.13",
    "@types/react": "~18.2.45",
    "@types/react-native": "^0.72.8",
    "@types/react-native-maps": "^0.24.2",
    "@types/react-test-renderer": "^18.0.7",
    "@typescript-eslint/eslint-plugin": "^8.7.0",
    "@typescript-eslint/parser": "^8.7.0",
    "eslint": "^8.57.1",
    "eslint-config-expo": "^7.1.2",
    "husky": "^9.1.6",
    "jest": "^29.7.0",
    "jest-expo": "~51.0.3",
    "lint-staged": "^15.2.10",
    "react-test-renderer": "18.2.0",
    "ts-jest": "^29.2.5",
    "typescript": "~5.3.3"
  },
  "lint-staged": {
    "*.{js,jsx,ts,tsx}": [
      "npm run lint"
    ]
  },
  "private": true
}

ameesme, 2024-11-11:
I have been experiencing the same issue. Upgrading to version 4.6.1 addressed the issue for me.

karimsallam, 2025-05-19:
I have the same issue running `pod install` via `nx`, @phazei patch fixes the issue. My Ruby is setup via `proto`.

phazei, 2025-05-19:
@mrousavy could that PR that fixes this issue be merged?  it's pretty simple and straight forward.

mrousavy, 2025-05-25:
yea send me a PR

phazei, 2025-05-26:
https://github.com/mrousavy/react-native-vision-camera/pull/3188

phazei, 2025-10-17:
@mrousavy  the PR's been sitting a while, still needed https://github.com/mrousavy/react-native-vision-camera/pull/3188 and @marwan362 approved it