margelo/react-native-vision-camera #3191: 💭 Production Build Error with 4.5.3

SKHRAPP, 2024-09-11:
### Question

Getting error when try to make production Build

```
* What went wrong:
Execution failed for task ':app:packageStagingReleaseBundle'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.PackageBundleTask$BundleToolWorkAction
   > Unrecognized native architecture for directory 'lib/riscv64'.
```

how to solve this issue 

### What I tried

 "dependencies": {
   "react-native": "0.74.2",
   "react-native-reanimated": "^3.14.0",
   "react-native-vision-camera": "^4.5.3",
   "react-native-worklets-core": "^1.3.3",
  }

### VisionCamera Version

4.5.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-09-12:
closing because you copy & pasted an identical issue again immediately afterwards.

mrousavy, 2024-09-12:
And stop pinging me for every issue you have.

closed 2024-09-12 by mrousavy