margelo/react-native-vision-camera #3192: 🔧 Production Build Error with 4.5.3

SKHRAPP, 2024-09-11:
### How were you trying to build the app?

Getting error when try to make production Build

```
* What went wrong:
Execution failed for task ':app:packageStagingReleaseBundle'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.PackageBundleTask$BundleToolWorkAction
   > Unrecognized native architecture for directory 'lib/riscv64'.
```

how to solve this issue @mo

### Full build logs

```tsx
.
```

### Project dependencies

```json
"dependencies": {
   "react-native": "0.74.2",
   "react-native-reanimated": "^3.14.0",
   "react-native-vision-camera": "^4.5.3",
   "react-native-worklets-core": "^1.3.3",
  }
```

### VisionCamera Version

4.5.3

### Target platforms

Android

### Operating system

Linux

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

MahmoudHemaid, 2024-09-12:
I have same issue, and i can see many issues opened for same issue @mrousavy

mrousavy, 2024-09-12:
This is not a VisionCamera error. In the logs you posted here there isn't even a single mention of VisionCamera. Please don't ping me everytime you have an issue @MahmoudHemaid .

closed 2024-09-12 by mrousavy

MahmoudHemaid, 2024-09-12:
How about this issue?
https://github.com/mrousavy/react-native-vision-camera/issues/3177

The issue is related to v4.5.3, people where able to make build works by downgrading to 4.5.2 so it's related to VisionCamera

mrousavy, 2024-09-12:
replied there.