margelo/react-native-vision-camera #3193: ✨ Document Scanner

Asdfchill, 2024-09-11:
### What feature or enhancement are you suggesting?

Maybe it's a bit of a stretch, but I think the camera package could really benefit from having an integrated "document scanning" function in addition to the code scanner.

So, basically, detect document boundaries and run a perspective transformation to get a normalized image.
[image]

What's the reason for asking for it? 
I'd like to use the Vision Camera for all our camera-related use cases.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I did find some other packages/frame processors that were apparently built on Vision Camera, but they depend on a third-party plugin and license. 

Otherwise, you could use the native OS functionalities, like VisionKit for iOS, for instance.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-09-12:
Hey!

Thanks for the feature request, but I don't want this to be part of VisionCamera. VisionCamera is a lean & lighweight package, any specific product features can easily be built using the extensible plugin system (Frame Processors)

So a document scanner will not be part of VisionCamera's core codebase.

xulihang, 2024-09-19:
This can be done using this frame processor plugin: https://github.com/tony-xlh/vision-camera-dynamsoft-document-normalizer

maccomaccomaccomacco, 2025-11-07:
> This can be done using this frame processor plugin: https://github.com/tony-xlh/vision-camera-dynamsoft-document-normalizer

it would be nice to have a free solution