margelo/react-native-vision-camera #3194: 🐛 i face this error when trying to upload the app to Play Store v4.5.2 and v4.5.3

amir-ibrahim, 2024-09-11:
### What's happening?

i face this error when trying to upload the app to Play Store v4.5.2 and v4.5.3 the error is :
'riscv64' architecture is only supported for application targeting WearOS.

### Reproduceable Code

```tsx
try to upload the app to app store
```

### Relevant log output

```shell
Your app declares support for 'riscv64' architecture, but the following libraries are missing:[libVisionCamera.so, libc++_shared.so, libcrashlytics-common.so, libcrashlytics-handler.so, libcrashlytics-trampoline.so, libcrashlytics.so, libfabricjni.so, libfb.so, libfbjni.so, libfolly_runtime.so, libgifimage.so, libglog.so, libglog_init.so, libhermes.so, libhermes_executor.so, libhermesinstancejni.so, libimagepipeline.so, libjniPdfium.so, libjscinstance.so, libjserrorhandler.so, libjsi.so, libjsijniprofiler.so, libjsinspector.so, liblogger.so, libmapbufferjni.so, libmodft2.so, libmodpdfium.so, libmodpng.so, libnative-filters.so, libnative-imagetranscoder.so, libreact_codegen_rncore.so, libreact_config.so, libreact_debug.so, libreact_nativemodule_core.so, libreact_newarchdefaults.so, libreact_render_animations.so, libreact_render_attributedstring.so, libreact_render_componentregistry.so, libreact_render_core.so, libreact_render_debug.so, libreact_render_graphics.so, libreact_render_imagemanager.so, libreact_render_leakchecker.so, libreact_render_mapbuffer.so, libreact_render_mounting.so, libreact_render_runtimescheduler.so, libreact_render_scheduler.so, libreact_render_telemetry.so, libreact_render_templateprocessor.so, libreact_render_textlayoutmanager.so, libreact_render_uimanager.so, libreact_utils.so, libreactnativeblob.so, libreactnativejni.so, libreactperfloggerjni.so, libreanimated.so, librninstance.so, librnscreens.so, librrc_image.so, librrc_legacyviewmanagerinterop.so, librrc_native.so, librrc_root.so, librrc_scrollview.so, librrc_text.so, librrc_textinput.so, librrc_unimplementedview.so, librrc_view.so, libruntimeexecutor.so, libsentry-android.so, libsentry.so, libstatic-webp.so, libturbomodulejsijni.so, libucrop.so, libuimanagerjni.so, libworklets.so, libyoga.so]. Update your app to rely on libraries that support 'riscv64' architecture.
'riscv64' architecture is only supported for application targeting WearOS.
```

### Camera Device

```json
...
```

### Device

android

### VisionCamera Version

4.5.2, 4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

amir-ibrahim, 2024-09-12:
so we solved the issue by using version 4.5.2 and removing the ^ from the version

[image]

closed 2024-09-12 by amir-ibrahim

mrousavy, 2024-09-12:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/3177.