margelo/react-native-vision-camera #3201: Is it possible to hide your face in a video stream using Vision Camera?

maxskitt, 2024-09-19:
### Question

Hi everyone,

I’m currently working with react-native-vision-camera and have a requirement to mask faces in a live video stream. The goal is to apply real-time masking to faces while capturing video.

Here’s a bit more context:

I’ve integrated VisionCameraFrameProcessor for face detection.

Is it feasible to apply real-time masking to a live video stream using Vision Camera?
Any advice or insights would be appreciated!

Thanks in advance!

### What I tried

I tried drawing the mask using Skia tools, but the process is very resource-intensive and takes a lot of time.

### VisionCamera Version

"react-native-vision-camera": "^4.0.3",

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

bruno-centanaro, 2024-09-19:
I think the skia thing you mentioned with a lower fps should work for your needs

nmajumder12, 2024-09-28:
Using skia with a face detector plugin, it is possible. This like you mentioned will be heavily resource demanding. You can reduce the FPS to 30 to get somewhat of an improvement.

mrousavy, 2024-10-01:
Hey- yes you can use Skia for this.

mrousavy, 2024-10-01:
I managed to get Skia running for a 1080p video stream at 100 FPS on an iPhone 15 Pro, which is pretty fast. Even faster than Instagram and Snapchat.

closed 2024-10-01 by mrousavy