margelo/react-native-vision-camera #3204: ✨ Do we think a LockedCameraCaptureExtension would be possible in React Native?

alexfoxy, 2024-09-20:
### What feature or enhancement are you suggesting?

In the current iOS 18 and iPhone 16 there are new ways to launch "LockedCameraCapture" apps. These typically are a more limited feature set of the app which can be used without unlocking the device. Unfortunately they require a "LockedCameraCaptureUIScene" which is written in Swift. There some docs here: https://developer.apple.com/documentation/lockedcameracapture

There doesn't seem any option just to "open the app", even if it requires FaceID. This is very frustrating as my app can't be assigned to the new Camera Control button on iPhone 16s.

Could this be possible without having to re-build the whole UI in Swift for this? Is it possible to embed the React Native view into a different a Swift View? 

Thanks!

### What Platforms whould this feature/enhancement affect?

iOS

### Alternatives/Workarounds

No workarounds available.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

alexfoxy, 2024-09-20:
@mrousavy Not sure the bot can help with this one, what are your thoughts?

mrousavy, 2024-09-23:
Hi! I took a quick look at the docs, but it seems like this is impossible to do in React Native (VisionCamera) right now.

mrousavy, 2024-09-23:
(and nothing I can really do, this is something the react native app should need to do, like a widget)

closed 2024-09-23 by mrousavy