margelo/react-native-vision-camera #3205: 🐛 09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer: java.lang.IllegalStateException: maxImages (6) has already been acquired, call #close before acquiring more.

TryRevilo, 2024-09-20:
### What's happening?

App Crashes when I start Capturing Frames **ie :** _in my Example below , when I set `isActive` to `true`_

### Reproduceable Code

```tsx
<Camera
  ref={revCameraRef}
  style={StyleSheet.absoluteFill}
  device={revDevicesArr[revSelectedDeviceId]}
  isActive={isRevCameraActive}
  frameProcessor={revFrameProcessor}
/>

const revFrameProcessor = useFrameProcessor(frame => {
  'worklet';

  frame.close();
}, []);
```

### Relevant log output

```shell
09-20 19:04:14.825   906 18275 E PDTblGen: [getTbl] dev(1) scenario(0) : Table is not generated. Please check error log during generating table (1st time lunch sensor mode(0))
09-20 19:04:14.825   906 18275 E PDTblGen:  (getTbl){#778:vendor/mediatek/proprietary/hardware/mtkcam/aaa/source/common/utils/PDTblGen/PDTblGen.cpp}
09-20 19:04:14.825   906 18275 E isp_mgr_bnr: apply(): Fail to Gen PDC Table, SensorDev(1), SensorMode(0), FrontBin(0)
09-20 19:04:14.825   906 18275 E isp_mgr_bnr:  (apply){#266:vendor/mediatek/proprietary/hardware/mtkcam/aaa/source/mt6765/isp_mgr/isp_mgr_bnr.cpp}
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer: Failed to acquire image.
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer: java.lang.IllegalStateException: maxImages (6) has already been acquired, call #close before acquiring more.
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at android.media.ImageReader.acquireNextImage(ImageReader.java:527)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.AndroidImageReaderProxy.acquireNextImage(AndroidImageReaderProxy.java:86)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.SafeCloseImageReaderProxy.acquireNextImage(SafeCloseImageReaderProxy.java:86)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.ImageAnalysisBlockingAnalyzer.acquireImage(ImageAnalysisBlockingAnalyzer.java:39)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.ImageAnalysisAbstractAnalyzer.onImageAvailable(ImageAnalysisAbstractAnalyzer.java:126)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.SafeCloseImageReaderProxy.lambda$setOnImageAvailableListener$1$androidx-camera-core-SafeCloseImageReaderProxy(SafeCloseImageReaderProxy.java:211)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.SafeCloseImageReaderProxy$$ExternalSyntheticLambda0.onImageAvailable(D8$$SyntheticClass:0)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.AndroidImageReaderProxy.lambda$setOnImageAvailableListener$0$androidx-camera-core-AndroidImageReaderProxy(AndroidImageReaderProxy.java:167)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.AndroidImageReaderProxy$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
09-20 19:04:14.827 18105 18387 E ImageAnalysisAnalyzer:   at java.lang.Thread.run(Thread.java:923)
09-20 19:04:14.858   906 18275 E PDTblGen: [getTbl] dev(1) scenario(0) : Table is not generated. Please check error log during generating table (1st time lunch sensor mode(0))
09-20 19:04:14.858   906 18275 E PDTblGen:  (getTbl){#778:vendor/mediatek/proprietary/hardware/mtkcam/aaa/source/common/utils/PDTblGen/PDTblGen.cpp}
09-20 19:04:14.858   906 18275 E isp_mgr_bnr: apply(): Fail to Gen PDC Table, SensorDev(1), SensorMode(0), FrontBin(0)
09-20 19:04:14.858   906 18275 E isp_mgr_bnr:  (apply){#266:vendor/mediatek/proprietary/hardware/mtkcam/aaa/source/mt6765/isp_mgr/isp_mgr_bnr.cpp}
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer: Failed to acquire image.
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer: java.lang.IllegalStateException: maxImages (6) has already been acquired, call #close before acquiring more.
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at android.media.ImageReader.acquireNextImage(ImageReader.java:527)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.AndroidImageReaderProxy.acquireNextImage(AndroidImageReaderProxy.java:86)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.SafeCloseImageReaderProxy.acquireNextImage(SafeCloseImageReaderProxy.java:86)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.ImageAnalysisBlockingAnalyzer.acquireImage(ImageAnalysisBlockingAnalyzer.java:39)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.ImageAnalysisAbstractAnalyzer.onImageAvailable(ImageAnalysisAbstractAnalyzer.java:126)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.SafeCloseImageReaderProxy.lambda$setOnImageAvailableListener$1$androidx-camera-core-SafeCloseImageReaderProxy(SafeCloseImageReaderProxy.java:211)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.SafeCloseImageReaderProxy$$ExternalSyntheticLambda0.onImageAvailable(D8$$SyntheticClass:0)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.AndroidImageReaderProxy.lambda$setOnImageAvailableListener$0$androidx-camera-core-AndroidImageReaderProxy(AndroidImageReaderProxy.java:167)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at androidx.camera.core.AndroidImageReaderProxy$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
09-20 19:04:14.863 18105 18387 E ImageAnalysisAnalyzer:   at java.lang.Thread.run(Thread.java:923)
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 5,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Android

### VisionCamera Version

"react-native-vision-camera": "^4.5.3",

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

TryRevilo, 2024-09-20:
Hi @maintenance-hans . Thanks for the reply . How do I go about closing each image properly before acquiring more

TryRevilo, 2024-09-20:
I think I already am with :  

```
const revFrameProcessor = useFrameProcessor(frame => {
  'worklet';

  frame.close();
}, []);
```

cnavarrete, 2024-11-29:
@TryRevilo Could you ever solve this?

anderson-souza-involves, 2025-04-17:
Hi @TryRevilo, I encountered a similar issue in our project, and it was related to a specific version of the react-native-reanimated library.

In our case, we were using version 3.4.2 of Reanimated, which was conflicting with version 1.3.3 of the react-native-worklets-core plugin.

Even though we declared the Reanimated plugin in babel.config.js with the processNestedWorklets property, it still didn’t work.

To resolve the issue, we had to upgrade Reanimated to version 3.5.0.

I’m not sure if you’re using Reanimated in your project, but if you are, I recommend trying to upgrade or downgrade the library to see if that resolves the issue.