margelo/react-native-vision-camera #3206: ✨ Skia Frame Processors save with video

aaa3334, 2024-09-23:
### What feature or enhancement are you suggesting?

It seems pretty limited that we can add in Skia frame processors during the preview, but it doesn't save at all...
I only discovered this after saving the video then looking more into the docs (and noticed the text saying it is preview only is hidden in the docs)...

Would be great if the preview matches the saved video (after all, that is the vast majority of use cases).

Maybe I just don't see how to get it to save though? Or is there a timeline for when it would be ready to be saved with it? Especially when using ML and the tflite libraries, the realtime should really be saved with it...

Perhaps I am just missing the right code though

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Would have to separately process afterwards which would mean twice the work and also differences between the two

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

aaa3334, 2024-09-24:
At the bottom of the skia page it says this:

"Preview-only

INFO
Skia Frame Processors are currently preview-only. Any content drawn to the Frame will not be visible in captured photos, snapshots or videos."

How to make it so it is also in the captured videos?

mrousavy, 2024-09-25:
I'm not sure if you understood that section - it is really really really complicated to save the result back to a video file (capture texture, run through encoder, write to file etc).

[image]

We build these tailor-made solutions for our clients at Margelo, so if you need this feature you can contact us for a quote.

closed 2024-09-25 by mrousavy

aaa3334, 2024-09-25:
I did but the reason I am doing everything myself is because I have no budget - else I would just hire a developer.
You are saying it is 'currently preview-only' though? Is there an eta for it being publicly available?

mrousavy, 2024-09-30:
Nope, not planned at the moment.