margelo/react-native-vision-camera #3208: 🐛 Black screen on Pixel 8 Pro

felipejh, 2024-09-23:
### What's happening?

I have the issue reported here https://github.com/mrousavy/react-native-vision-camera/issues/2862 and solved here https://github.com/mrousavy/react-native-vision-camera/pull/2897.

My device is a Pixel 8 Pro. I have tried both `qualityPrioritization="speed"` and `qualityPrioritization="quality"`.

- react-native-vision-camera: 4.5.2
- react-native": 0.74.5

### Reproduceable Code

```tsx
const devices = useCameraDevices();

<Camera
   ref={cameraRef}
   style={StyleSheet.absoluteFill}
   device={device as CameraDevice}
   isActive={isActive}
   photo={true}
   video={false}
   onInitialized={onInitialized}
   onError={onError}
   photoQualityBalance="speed"
   lowLightBoost={device?.supportsLowLightBoost}
   enableZoomGesture={false}
   animatedProps={cameraAnimatedProps}
/>
```

### Relevant log output

```shell
Camera error: [session/invalid-output-configuration]: Failed to configure the Camera Session because the output/stream configurations are invalid! (caused by {"stacktrace":"java.lang.IllegalArgumentException: Surface was abandoned\n\tat android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:135)\n\tat android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:563)\n\tat android.hardware.camera2.params.OutputConfiguration.<init>(OutputConfiguration.java:404)\n\tat androidx.camera.camera2.internal.compat.params.OutputConfigurationCompatApi33Impl.<init>(OutputConfigurationCompatApi33Impl.java:37)\n\tat androidx.camera.camera2.internal.compat.params.OutputConfigurationCompat.<init>(OutputConfigurationCompat.java:63)\n\tat androidx.camera.camera2.internal.CaptureSession.getOutputConfigurationCompat(CaptureSession.java:385)\n\tat androidx.camera.camera2.internal.CaptureSession.openCaptureSession(CaptureSession.java:309)\n\tat androidx.camera.camera2.internal.CaptureSession.lambda$open$0$androidx-camera-camera2-internal-CaptureSession(CaptureSession.java:221)\n\tat androidx.camera.camera2.internal.CaptureSession$$ExternalSyntheticLambda3.apply(Unknown Source:8)\n\tat androidx.camera.core.impl.utils.futures.ChainingListenableFuture.run(ChainingListenableFuture.java:201)\n\tat androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:229)\n\tat androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:171)\n\tat java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)\n\tat java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)\n\tat java.lang.Thread.run(Thread.java:1012)\n","message":"Surface was abandoned"})
```

### Camera Device

```json
[
  {
    "formats": null,
    "sensorOrientation": "landscape-left",
    "hardwareLevel": "full",
    "maxZoom": 30,
    "minZoom": 0.4944782853126526,
    "maxExposure": 24,
    "supportsLowLightBoost": true,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera",
      "wide-angle-camera",
      "ultra-wide-angle-camera",
      "telephoto-camera",
      "telephoto-camera",
      "telephoto-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": true,
    "minFocusDistance": 10.500000100135805,
    "minExposure": -24,
    "name": "0 (BACK) androidx.camera.camera2",
    "hasFlash": true,
    "hasTorch": true,
    "position": "back",
    "id": "0"
  },
  {
    "formats": null,
    "sensorOrientation": "landscape-right",
    "hardwareLevel": "full",
    "maxZoom": 10,
    "minZoom": 0.8958333134651184,
    "maxExposure": 24,
    "supportsLowLightBoost": true,
    "neutralZoom": 1,
    "physicalDevices": [
      "wide-angle-camera",
      "wide-angle-camera",
      "wide-angle-camera"
    ],
    "supportsFocus": true,
    "supportsRawCapture": false,
    "isMultiCam": true,
    "minFocusDistance": 10,
    "minExposure": -24,
    "name": "1 (FRONT) androidx.camera.camera2",
    "hasFlash": false,
    "hasTorch": false,
    "position": "front",
    "id": "1"
  }
]
```

### Device

Pixel 8 Pro

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-09-25:
> No, I cannot reproduce the issue in the Example app

Sorry but then I guess this is an issue in your app..

felipejh, 2024-09-26:
> > No, I cannot reproduce the issue in the Example app
> 
> Sorry but then I guess this is an issue in your app..

Thank you for you reply. But I believe another person is facing the same issue here https://github.com/mrousavy/react-native-vision-camera/issues/2862#issuecomment-2366970218.

mrousavy, 2024-09-30:
Yea I'm sorry but that comment gives even less context. Not even logs.
There's nothing I can do at this point if it cannot be reproduced in the example app.

JesseLawler, 2024-10-03:
@felipejh Try forcing lowLightBoost to false.  I found on Pixel 6 and Pixel 9, this was necessary even though supportsLowLightBoost returns true.  No idea why, but worked for me.

felipejh, 2024-11-04:
> @felipejh Try forcing lowLightBoost to false. I found on Pixel 6 and Pixel 9, this was necessary even though supportsLowLightBoost returns true. No idea why, but worked for me.

That worked like a charm! 🥇 
Thank you!

closed 2024-11-04 by felipejh