margelo/react-native-vision-camera #3209: 🐛 [iOS] `NSRangeException` crash ending recording session

jkaufman, 2024-09-23:
### What's happening?

Caught this error using Crashlytics. The trace may be sufficient to debug. I do not have ADB logs or a minimal reproducible result. Even if the ticket isn't actionable and is closed, I want to record it in case others encounter the same issue. 

### Reproduceable Code

```tsx
<Camera
                style={…}
                ref={…}
                device={…}
                isActive={…}
                format={…}
                fps={…}
                photo={false}
                video={true}
                audio={true}
                torch={…}
                outputOrientation={'device'}
              />
```

### Relevant log output

```shell
No log output available to share.
Crash detected in production.
Will share if able to reproduce locally.
```

### Camera Device

```json
Available details:

{
	"duration": 66,
	"format": {
		"minISO": 34,
		"photoHeight": 1188,
		"fieldOfView": 106.7479019165039,
		"maxFps": 60,
		"videoHeight": 1080,
		"autoFocusSystem": "phase-detection",
		"supportsVideoHdr": false,
		"videoWidth": 1920,
		"minFps": 1,
		"supportsPhotoHdr": false,
		"photoWidth": 2112,
		"supportsDepthCapture": false,
		"maxISO": 3264,
		"videoStabilizationModes": [
			"auto",
			"cinematic",
			"off",
			"standard",
			"cinematic-extended"
		]
	},
	"deviceName": "Back Triple Camera"
}
```

### Device

iPhone 13 Pro (iOS 17.6.1)

### VisionCamera Version

4.4.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

darecki, 2025-01-23:
Hi @jkaufman, thanks for opening this issue. I’ve encountered a very similar problem reported on Sentry but haven’t been able to reproduce it either. Were you eventually able to reproduce the issue or find a workaround?  

One thing worth mentioning is that this started occurring around the time we upgraded Vision Camera from version 4.4.1 to 4.5.1, though it could also be related to other code changes.

jkaufman, 2025-01-24:
I have not found a consistent repro case, yet.