margelo/react-native-vision-camera #3210: ✨ Enable multitasking camera access on iOS

ethul, 2024-09-25:
### What feature or enhancement are you suggesting?

Would it be possible to add support for enabling multitasking camera access?

This would be useful for apps running on iPads with Stage Manager or Split View/Slide Over turned on.

From the [Apple developer documentation](https://developer.apple.com/documentation/avkit/accessing-the-camera-while-multitasking-on-ipad) I tried adding the following to the camera session, which seemed to do the trick.

```swift
if #available(iOS 16.0, *) {
  if self.captureSession.isMultitaskingCameraAccessSupported {
    // Enable use of the camera in multitasking modes.
    self.captureSession.isMultitaskingCameraAccessEnabled = true
  }
}
```

### What Platforms whould this feature/enhancement affect?

iOS

### Alternatives/Workarounds

I manually made the change in `Core/CameraSession.swift`.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

ethul, 2024-09-26:
Without the change to enable multitasking, when Stage Manager or Split View/Slide Over was turned on, the camera preview was blank, and recording a video issued an error:

```
capture/unknown: Error Domain=AVFoundationErrorDomain Code=-11803 "Cannot Record" UserInfo={AVErrorRecordingFailureDomainKey=3, NSLocalizedDescription=Cannot Record, NSLocalizedRecoverySuggestion=Try recording again.}
```

mrousavy, 2024-10-01:
Hey! Oh, pretty cool. Yea I can add that feature if I have some free time soon, thanks for posting the code here this is really helpful.

ethul, 2024-10-03:
Great, thanks!

ChristopherGabba, 2024-10-28:
+1 on this, I have `CameraRuntimeError`'s every now and then when another task interrupts the camera. Not 100% sure if this would fix the problem, but it couldn't hurt!

mrousavy, 2024-10-29:
Reading the docs wouldn't hurt - this is already `true` if you have the entitlement for it (which is required, and would crash otherwise)

[image]

closed 2024-10-29 by mrousavy

ethul, 2024-10-30:
I think it is important to note that the multitaasking-camera-access entitlement [is deprecated](https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_developer_avfoundation_multitasking-camera-access).

[image]

It seems like the recommended approach is to use `isMultitaskingCameraAccessEnabled`.