margelo/react-native-vision-camera #3211: 🐛

stevengoldberg, 2024-09-27:
### What's happening?

Using a skia frame processor disables the preview, and apparently that also breaks focusing. Calling Camera.Focus with the skiaFrameProcessor enabled gives this error:

`capture/focus-requires-preview: Focus requires preview={...} to be enabled!`

### Reproduceable Code

```tsx
// ...

const handleTap = useCallback(
    (position) => {
        cameraRef.current?.focus(position)
    },
    [cameraRef]
)

const tapGesture = Gesture.Tap()
    .onStart((e) => {
        'worklet'
        if (!device.supportsFocus) {
            return
        }
        runOnJS(handleTap)({ x: e.x, y: e.y })
})

const frameProcessor = useSkiaFrameProcessor((frame) => frame.render())

return (
    <GestureDetector gesture={composedGestures}>
                <AnimatedCamera
                    style={{ flex: 1, width: windowWidth }}
                    ref={cameraRef}
                    device={device}
                    isActive={isActive}
                    photo
                    video
                    orientation="portrait"
                    format={format}
                    animatedProps={cameraAnimatedProps}
                    frameProcessor={frameProcessor}
                    photoQualityBalance="quality"
                    outputOrientation="preview"
                /> 
            </GestureDetector>
)
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "sensorOrientation": "portrait",
  "formats": [],
  "position": "back",
  "maxZoom": 123.75,
  "hasTorch": true,
  "minZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "hardwareLevel": "full",
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "minFocusDistance": 2,
  "name": "Back Triple Camera",
  "minExposure": -8,
  "hasFlash": true,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
  "isMultiCam": true
}
```

### Device

iPhone 15 Pro

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-10-01:
This is intentional, as the error message says we cannot focus without a preview view, at least not as of right now. I worked on a temporary workaround in https://github.com/mrousavy/react-native-vision-camera/pull/3089, but nothing merged yet.

Will probably only really be possible with Nitro, as this is a really complex problem to solve.

Also, this issue is invalid as it doesn't have a proper title.

closed 2024-10-01 by mrousavy

MarianAlexandruAlecu, 2024-11-06:
@mrousavy Thanks for having a look! We were also looking to add a square around the tap area to indicate what part will try to be focused and we got the same error mentioned above.