margelo/react-native-vision-camera #3212: 🐛 Android Preview Skipping (Frame Rate Drop) with `useFrameProcessor` + `react-native-vision-camera-face-detector`

ChristopherGabba, 2024-09-30:
### What's happening?

I'm recording a video on my Google Pixel 3 & 4 Physical Device (I know these are older devices) using `frameProcessors` + `react-native-vision-camera-face-detector` and the preview + recording are very slow. This issue may be related to another issue regarding `useSkiaFrameProcessor`: https://github.com/mrousavy/react-native-vision-camera/issues/2763.

Here is a video of the low frame rate:

https://github.com/user-attachments/assets/09d906c0-1a02-4a54-bf04-28775e65c7d8

Steps to reproduce:
1. Copy the below code and paste it directly over the `CameraPage.tsx` file in the RNVC example app. 
2. Make sure to run `yarn add react-native-vision-camera-face-detector`
3. As soon as you comment out the line `frameProcessor={frameProcessor}` line, you will get a big performance bump.

Run on Physical Android device. Here is a video:
https://github.com/user-attachments/assets/ba2499b9-2593-4350-9274-8442c8b602e8

I apologize if the android `logcat` logs are not good, I've never done it before and that log chart was absolutely flying lol... Hopefully I was able to copy some relevant helpful info...

### Reproduceable Code

```tsx
import * as React from 'react'
import { useRef, useState, useCallback, useMemo } from 'react'
import type { GestureResponderEvent } from 'react-native'
import { StyleSheet, Text, View } from 'react-native'
import type { PinchGestureHandlerGestureEvent } from 'react-native-gesture-handler'
import { PinchGestureHandler, TapGestureHandler } from 'react-native-gesture-handler'
import type { CameraProps, CameraRuntimeError, PhotoFile, VideoFile } from 'react-native-vision-camera'
import {
  useCameraDevice,
  useCameraFormat,
  useFrameProcessor,
  useLocationPermission,
  useMicrophonePermission,
} from 'react-native-vision-camera'
import { Camera } from 'react-native-vision-camera'
import { CONTENT_SPACING, CONTROL_BUTTON_SIZE, MAX_ZOOM_FACTOR, SAFE_AREA_PADDING, SCREEN_HEIGHT, SCREEN_WIDTH } from './Constants'
import Reanimated, { Extrapolate, interpolate, useAnimatedGestureHandler, useAnimatedProps, useSharedValue } from 'react-native-reanimated'
import { useEffect } from 'react'
import { useIsForeground } from './hooks/useIsForeground'
import { StatusBarBlurBackground } from './views/StatusBarBlurBackground'
import { CaptureButton } from './views/CaptureButton'
import { PressableOpacity } from 'react-native-pressable-opacity'
import MaterialIcon from 'react-native-vector-icons/MaterialCommunityIcons'
import IonIcon from 'react-native-vector-icons/Ionicons'
import type { Routes } from './Routes'
import type { NativeStackScreenProps } from '@react-navigation/native-stack'
import { useIsFocused } from '@react-navigation/core'
import { usePreferredCameraDevice } from './hooks/usePreferredCameraDevice'
import { 
  Face,
  useFaceDetector,
  FaceDetectionOptions
} from 'react-native-vision-camera-face-detector'
import { Worklets } from 'react-native-worklets-core'

const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera)
Reanimated.addWhitelistedNativeProps({
  zoom: true,
})

const SCALE_FULL_ZOOM = 3

type Props = NativeStackScreenProps<Routes, 'CameraPage'>
export function CameraPage({ navigation }: Props): React.ReactElement {
  const camera = useRef<Camera>(null)
  const [isCameraInitialized, setIsCameraInitialized] = useState(false)
  const microphone = useMicrophonePermission()
  const location = useLocationPermission()
  const zoom = useSharedValue(1)
[... 558 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Google Pixel 3

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ChristopherGabba, 2024-09-30:
I do not own a newer device to test on, but referencing another issue that was closed for lack of detail: https://github.com/mrousavy/react-native-vision-camera/issues/2763, I believe this may happen on newer devices as well.

This should be very easy to reproduce in the RNVC example app.

kolias4, 2024-10-03:
I have the same issue. I've found a few things that might be causing it, and I think it has less to do with the FrameProcessor and more with the model you're loading for object detection. In my case if i don't pass the frame to the model it works fine. 
Also, it happens on Android, while on iOS this issue doesn't seem to exist.

BubbleTrouble14, 2024-10-05:
Try lowering your frame resolution.

Have a look at the [Face Blur App](https://github.com/mrousavy/FaceBlurApp/blob/845ea1c1f8ffef262eed7e1df2840dd12fa438a8/app/App.tsx)

```
const format = useCameraFormat(device, [
    {
      videoResolution: Dimensions.get('window'),
    },
    {
      fps: 60,
    },
  ]);

```

ChristopherGabba, 2024-10-08:
I've lowered my frame resolution as low as it will go and it's still skippy.

luicfrr, 2024-10-08:
The problem is because you're running a heavy face detection duty sync on each frame instead of running it on a separated thread:
```
const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    const faces = detectFaces(frame) // here is the problem
    // ... chain frame processors
    // ... do something with frame
    handleDetectedFaces(faces)
  }, [handleDetectedFaces])
```

Try using [this](https://react-native-vision-camera.com/docs/guides/frame-processors-interacting#running-asynchronously) or use face detection following [my recommended way](https://github.com/luicfrr/react-native-vision-camera-face-detector?tab=readme-ov-file#-usage)

ChristopherGabba, 2024-10-08:
@luicfrr Thank you! This makes perfect sense, sorry I must have missed that section of the docs. Closing issue!

closed 2024-10-08 by ChristopherGabba