margelo/react-native-vision-camera #3225: 🐛 [iOS] Camera always alive even after Camera component is unmounted

ibrahim-ytl, 2024-10-04:
### What's happening?

When I unmount the `Camera` component. The device camera is still active behind the scene. Below is the screenshot and video.

[Video]
<video  src="https://github.com/user-attachments/assets/9b8ce6cc-2ec3-4697-8da3-fcd8224a8e22" />

[Image]
[image]

### Reproduceable Code

```tsx
import { useEffect, useState } from 'react';
import { Pressable, StyleSheet, Text, View } from "react-native"
import { Camera, useCameraDevice, useCameraPermission } from "react-native-vision-camera"

export function VisionCamera() {
    const device = useCameraDevice('back')
    const { hasPermission, requestPermission } = useCameraPermission()
    const [isCameraActive, setIsCameraActive] = useState(true)

    useEffect(() => {
        requestPermission()
    }, [])

    const onToggleCamera = () => {
        setIsCameraActive(!isCameraActive)
    }
  
    if (!hasPermission) return <Text>No camera permission</Text>
    if (device == null) return <Text>No camera device</Text>

    return (
        <View style={{ flex: 1 }}>
            {isCameraActive && (
                <Camera
                    style={StyleSheet.absoluteFill}
                    device={device}
                    isActive={isCameraActive}
                />
            )}

            <View style={{ backgroundColor: isCameraActive ? 'red' : 'green', position: 'absolute', bottom: 24, zIndex: 999, alignSelf: 'center' }}>
                <Pressable style={{ padding: 16 }} onPress={onToggleCamera}>
                    <Text style={{ color: 'white', fontWeight: 'bold', fontSize: 24 }}>
                        {isCameraActive ? 'Stop Camera' : 'Start Camera'}
                    </Text>
                </Pressable>
            </View>
        </View>
    )
  }
```

### Relevant log output

```shell
NA
```

### Camera Device

```json
{
  "minZoom": 1,
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "position": "back",
  "hasTorch": true,
  "isMultiCam": false,
  "hardwareLevel": "full",
  "supportsRawCapture": false,
  "supportsFocus": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "maxZoom": 16,
  "minExposure": -8,
  "name": "Back Camera",
  "hasFlash": true,
  "minFocusDistance": 12,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "sensorOrientation": "portrait",
  "formats": []
}
```

### Device

iPhone XS

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ibrahim-ytl, 2024-10-07:
Please help me take a look. I have also provided the reproduceable code. @mrousavy

mrousavy, 2024-10-07:
No. You didn't provide logs, which is the most important detail here. Can't help you.

ibrahim-ytl, 2024-10-09:
> I believe this is a dupe of [this issue](https://github.com/mrousavy/react-native-vision-camera/issues/3041#issuecomment-2263720909). Below is a patch to `4.5.3` to work around the behavior:
> 
> ```diff
> diff --git a/node_modules/react-native-vision-camera/ios/React/CameraView.swift b/node_modules/react-native-vision-camera/ios/React/CameraView.swift
> index 6b85538..0cbfd44 100644
> --- a/node_modules/react-native-vision-camera/ios/React/CameraView.swift
> +++ b/node_modules/react-native-vision-camera/ios/React/CameraView.swift
> @@ -278,7 +278,9 @@ public final class CameraView: UIView, CameraSessionDelegate, PreviewViewDelegat
>      }
>  
>      // Prevent phone from going to sleep
> -    UIApplication.shared.isIdleTimerDisabled = isActive
> +    // Remove this line and instead rely on expo keep-awake.
> +    // See: https://github.com/mrousavy/react-native-vision-camera/issues/3041#issuecomment-2263720909
> +    // UIApplication.shared.isIdleTimerDisabled = isActive
>    }
>  
>    func updatePreview() {
> ```

this patch does not work. The main reason that the `isActive` need to be `false` first before the component unmounted. If the `isActive` state never change to `false` before navigating away, the green dot will be there.

ref: https://github.com/mrousavy/react-native-vision-camera/issues/3041#issuecomment-2401622720

ibrahim-ytl, 2024-10-09:
Look like this same exact issue still exist until now: https://github.com/mrousavy/react-native-vision-camera/issues/905

By the way, since I am using react-navigation. The workaround is to delay the goBack navigations.

```tsx
const onPressClose = () => {
  setIsCameraActive(false)
  setTimeout(() => goBack(), 10);
}
```

Side not: Thanks for the awesome library. ❤️

jkaufman, 2024-10-09:
I apologize, I confused two different bugs that aren't related.