margelo/react-native-vision-camera #3226: 🐛 in iOS 18, onPreviewStarted is no longer called when switching devices

stevengoldberg, 2024-10-04:
### What's happening?

In iOS 17, switching between front and back cameras results in `onPreviewStopped` being invoked, followed by `onPreviewStarted`. In iOS 18, neither method is invoked when switching cameras.

### Reproduceable Code

```tsx
const [cameraPosition, setCameraPosition] = useState('back')

const physicalDevices =
    cameraPosition === 'back'
        ? [
             'ultra-wide-angle-camera',
             'wide-angle-camera',
             'telephoto-camera',
          ]
         : ['wide-angle-camera']

const device = useCameraDevice(cameraPosition, { physicalDevices })

const formatFilter = [
    {
        photoAspectRatio: 4 / 3,
    },
        { photoResolution: 'max' },
        { videoAspectRatio: 4 / 3 },
        { videoResolution: { width: 1024, height: 768 } },
    ]
 

const format = useCameraFormat(device, formatFilter)

const onFlipCameraPressed = useCallback(() => {
        setCameraPosition((p) => (p === 'back' ? 'front' : 'back'))
    })
}, [])

return (
    <Camera
        style={{ flex: 1, width: windowWidth }}
        ref={cameraRef}
        device={device}
        isActive={isActive}
        photo
        video
        orientation="portrait"
        format={format}
        onPreviewStarted={() => { console.log('preview started') }}
        onPreviewStopped={() => { console.log('preview stopped') }}
        photoQualityBalance="quality"
        outputOrientation="preview"
    />
)
```

### Relevant log output

```shell
// Log output on switching between front and back cameras

18:12:56.605: [info] 📸 VisionCamera.didSetProps(_:): Updating 3 props: [cameraId, isMirrored, format]
18:12:56.605: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
18:12:56.605: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: false, locationChanged: false)
18:12:56.605: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Input Device...
18:12:56.605: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:7...
18:12:56.607: [info] 📸 VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
18:12:56.607: [info] 📸 VisionCamera.configureOutputs(configuration:): Configuring Outputs...
18:12:56.607: [info] 📸 VisionCamera.configureOutputs(configuration:): Adding Photo output...
18:12:56.608: [info] 📸 VisionCamera.configureOutputs(configuration:): Adding Video Data output...
18:12:56.608: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: portrait...
18:12:56.608: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: portrait...
18:12:56.608: [info] 📸 VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
18:12:56.608: [info] 📸 VisionCamera.setTargetOutputOrientation(_:): Setting target output orientation from preview to preview...
18:12:56.608: [info] 📸 VisionCamera.configureFormat(configuration:device:): Configuring Format (5712x4284 | 4032x3024@30.0 (ISO: 55.0..12320.0))...
18:12:56.608: [info] 📸 VisionCamera.configureFormat(configuration:device:): Already selected active format, no need to configure.
18:12:56.608: [info] 📸 VisionCamera.getPixelFormat(for:): Available Pixel Formats: ["420v", "420f", "BGRA", "&8v0", "-8v0", "&8f0", "-8f0", "&BGA", "-BGA"], finding best match... (pixelFormat="rgb", enableHdr={false}, enableBufferCompression={true})
18:12:56.608: [info] 📸 VisionCamera.getPixelFormat(for:): Using PixelFormat: -BGA...
18:12:56.627: [info] 📸 VisionCamera.didSetProps(_:): Updating 2 props: [exposure, zoom]
18:12:56.627: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...

18:12:57.840: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: false, outputsChanged: false, videoStabilizationChanged: false, orientationChanged: false, formatChanged: false, sidePropsChanged: false, torchChanged: false, zoomChanged: true, exposureChanged: false, audioSessionChanged: false, locationChanged: false)
```

### Camera Device

```json
{
  "hasTorch": true,
  "sensorOrientation": "portrait",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
  "isMultiCam": true,
  "minZoom": 1,
  "formats": [],
  "hardwareLevel": "full",
  "minExposure": -8,
  "hasFlash": true,
  "name": "Back Triple Camera",
  "position": "back",
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "maxZoom": 123.75,
  "minFocusDistance": 2,
  "supportsLowLightBoost": false,
  "maxExposure": 8
}
```

### Device

iPhone 15 Pro (iOS 18.0.1)

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

stevengoldberg, 2024-10-04:
I attempted to build the example app, but the build fails with the following error:
[image]

mrousavy, 2024-10-05:
Delete your `.xcode.env.local` and try again

stevengoldberg, 2024-10-05:
Thanks — the build still fails with a similar error. Here is the full build log:

```
Showing Recent Issues

Prepare build

ComputeTargetDependencyGraph

warning: Building targets in manual order is deprecated - choose Dependency Order in scheme settings instead, or set DISABLE_MANUAL_TARGET_ORDER_BUILD_WARNING in any of the targets in the current scheme to suppress this warning
note: Target dependency graph (74 targets)
    Target 'VisionCameraExample' in project 'VisionCameraExample'
        ➜ Implicit dependency on target 'Pods-VisionCameraExample' in project 'Pods' via file 'libPods-VisionCameraExample.a' in build phase 'Link Binary'
        ➜ Implicit dependency on target 'DoubleConversion' in project 'Pods' via options '-lDoubleConversion' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'MMKV' in project 'Pods' via options '-lMMKV' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'MMKVCore' in project 'Pods' via options '-lMMKVCore' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RCT-Folly' in project 'Pods' via options '-lRCT-Folly' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RCTDeprecation' in project 'Pods' via options '-lRCTDeprecation' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RCTTypeSafety' in project 'Pods' via options '-lRCTTypeSafety' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RNGestureHandler' in project 'Pods' via options '-lRNGestureHandler' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RNReanimated' in project 'Pods' via options '-lRNReanimated' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RNScreens' in project 'Pods' via options '-lRNScreens' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RNStaticSafeAreaInsets' in project 'Pods' via options '-lRNStaticSafeAreaInsets' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'RNVectorIcons' in project 'Pods' via options '-lRNVectorIcons' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-Codegen' in project 'Pods' via options '-lReact-Codegen' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-Core' in project 'Pods' via options '-lReact-Core' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-CoreModules' in project 'Pods' via options '-lReact-CoreModules' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-Fabric' in project 'Pods' via options '-lReact-Fabric' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-FabricImage' in project 'Pods' via options '-lReact-FabricImage' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-ImageManager' in project 'Pods' via options '-lReact-ImageManager' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-Mapbuffer' in project 'Pods' via options '-lReact-Mapbuffer' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-NativeModulesApple' in project 'Pods' via options '-lReact-NativeModulesApple' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTAnimation' in project 'Pods' via options '-lReact-RCTAnimation' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTAppDelegate' in project 'Pods' via options '-lReact-RCTAppDelegate' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTBlob' in project 'Pods' via options '-lReact-RCTBlob' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTFabric' in project 'Pods' via options '-lReact-RCTFabric' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTImage' in project 'Pods' via options '-lReact-RCTImage' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTLinking' in project 'Pods' via options '-lReact-RCTLinking' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTNetwork' in project 'Pods' via options '-lReact-RCTNetwork' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTSettings' in project 'Pods' via options '-lReact-RCTSettings' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTText' in project 'Pods' via options '-lReact-RCTText' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RCTVibration' in project 'Pods' via options '-lReact-RCTVibration' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RuntimeApple' in project 'Pods' via options '-lReact-RuntimeApple' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RuntimeCore' in project 'Pods' via options '-lReact-RuntimeCore' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-RuntimeHermes' in project 'Pods' via options '-lReact-RuntimeHermes' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-cxxreact' in project 'Pods' via options '-lReact-cxxreact' in build setting 'OTHER_LDFLAGS'
        ➜ Implicit dependency on target 'React-debug' in project 'Pods' via options '-lReact-debug' in build setting 'OTHER_LDFLAGS'
[... cut ...]
/Users/stevengoldberg/Documents/sample code/react-native-vision-camera/package/example/ios/Pods/Pods.xcodeproj: warning: The iOS deployment target 'IPHONEOS_DEPLOYMENT_TARGET' is set to 9.0, but the range of supported deployment target versions is 12.0 to 17.2.99. (in target 'react-native-cameraroll-RNCameraRollPrivacyInfo' from project 'Pods')

/Users/stevengoldberg/Documents/sample code/react-native-vision-camera/package/example/ios/Pods/Pods.xcodeproj: The iOS deployment target 'IPHONEOS_DEPLOYMENT_TARGET' is set to 9.0, but the range of supported deployment target versions is 12.0 to 17.2.99.

Build failed    10/5/24, 10:18 AM    3.7 seconds
```

ChristopherGabba, 2025-04-08:
+1 This is a big deal. Mine is also not being invoked all of a sudden. 

<Camera
  onPreviewStarted={()=> console.log("STARTED")}
/>

never gets called all of a sudden.