margelo/react-native-vision-camera #3229: 🐛 App crash when open camera on Android when using useFrameProcessor

diegomeire, 2024-10-06:
### What's happening?

When building for Android, whenever I use the following with `frameProcessor={frameProcessor}` the app crashes right at the start. If I remove `frameProcessor` it works normally.

Here's my dependencies on package.json:
```
  "dependencies": {
    "expo": "~51.0.28",
    "expo-status-bar": "~1.12.1",
    "metro": "^0.80.12",
    "react": "18.2.0",
    "react-native": "0.74.5",
    "react-native-asset": "^2.1.1",
    "react-native-fast-tflite": "^1.4.0",
    "react-native-fs": "^2.20.0",
    "react-native-vision-camera": "^4.5.3",
    "react-native-worklets-core": "^1.3.3",
    "vision-camera-resize-plugin": "^3.1.0"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@types/react": "~18.2.45",
    "typescript": "^5.1.3"
  },
```

### Reproduceable Code

```tsx
/* eslint-disable @typescript-eslint/no-var-requires */
import * as React from 'react'

import { StyleSheet, View, Text, ActivityIndicator } from 'react-native'
import {
  Tensor,
  TensorflowModel,
  useTensorflowModel,
} from 'react-native-fast-tflite'
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useFrameProcessor,
} from 'react-native-vision-camera'
import { useResizePlugin } from 'vision-camera-resize-plugin'

function tensorToString(tensor: Tensor): string {
  return `\n  - ${tensor.dataType} ${tensor.name}[${tensor.shape}]`
}
function modelToString(model: TensorflowModel): string {
  return (
    `TFLite Model (${model.delegate}):\n` +
    `- Inputs: ${model.inputs.map(tensorToString).join('')}\n` +
    `- Outputs: ${model.outputs.map(tensorToString).join('')}`
  )
}

export default function App(): React.ReactNode {
  const { hasPermission, requestPermission } = useCameraPermission()
  const device = useCameraDevice('back')

  // from https://www.kaggle.com/models/tensorflow/efficientdet/frameworks/tfLite
  const model = useTensorflowModel(require('./assets/lush_naked.tflite'), 'nnapi')
  const actualModel = model.state === 'loaded' ? model.model : undefined

  React.useEffect(() => {
    if (actualModel == null) return
    console.log(`Model loaded! Shape:\n${modelToString(actualModel)}]`)
  }, [actualModel])

[... 176 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Google Pixel 3

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ebrimasamba, 2024-10-08:
Am having the same problem. The moment i comment the frameprocessor line on the camera, the app doesn't crash. 

These are the current versions am using
 "react-native-vision-camera": "^4.5.3",
 "react-native-worklets-core": "^1.3.3"
 
 
   const frameProcessor = useFrameProcessor((frame) => {
    "worklet";
    console.log(`Frame: ${frame.width}x${frame.height} (${frame.pixelFormat})`);
  }, []);
  
  
     <Camera
        frameProcessor={frameProcessor}
        style={styles.camera}
        device={device}
        isActive={isActive}
        pixelFormat="yuv"
      />

mrousavy, 2024-10-10:
Does it help if you choose a smaller format? Lower `videoResolution`

ebrimasamba, 2024-10-10:
It worked after i downgraded to 4.5.2 and restarted my pc, it works now
even after i upgraded to the latest version,  i think it must’ve been a
caching problem on my end

closed 2024-10-14 by mrousavy