margelo/react-native-vision-camera #3230: 🐛 Lens switching 'broken' on iOS 18

kbrandwijk, 2024-10-06:
### What's happening?

Since upgrading to iOS 18, we are seeing a regression in lens switching behavior. It only switches lenses once when using a virtual device, but doesn't switch back, or is almost impossible to get to switch back. We verified this also happens in the ShadowLens demo app. Try getting close to an object, the lens switches, then move back away until the lens switches again. From that point on, it's impossible to get closer without the image being blurry (the lens doesn't switch back anymore).

### Reproduceable Code

```tsx
No code involved here, just a default triple back camera virtual device.
```

### Relevant log output

```shell
-
```

### Camera Device

```json
{
    "formats": [],
    "hardwareLevel": "full",
    "hasFlash": true,
    "hasTorch": true,
    "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
    "isMultiCam": true,
    "maxExposure": 8,
    "maxZoom": 123.75,
    "minExposure": -8,
    "minFocusDistance": 2,
    "minZoom": 1,
    "name": "Back Triple Camera",
    "neutralZoom": 2,
    "physicalDevices": [
      "ultra-wide-angle-camera",
      "wide-angle-camera",
      "telephoto-camera"
    ],
    "position": "back",
    "sensorOrientation": "portrait",
    "supportsFocus": true,
    "supportsLowLightBoost": false,
    "supportsRawCapture": false
  }
```

### Device

iPhone 14 Pro (18.0.1)

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kbrandwijk, 2024-10-06:
@mrousavy Marc, curious to hear if you have observed this as well on iOS 18. Getting XCode logs is a bit more involved, as I'm just seeing this using the ShadowLens app from the app store as well, and I use Expo so I don't run my RN projects through XCode. Lmk if you need any additional information!

mrousavy, 2024-10-07:
1. You can run your Expo app through Xcode as well. Without logs, I cannot help.
2. This clearly needs a video demo as well

tjohnman, 2024-12-02:
I'm having the same issue, although I don't know if iOS 18 is a factor in my case.
minZoom is being reported as 1, same as neutralZoom, and I don't see a way to switch to wide angle.

@mrousavy If you can point me to instructions on how to run my Expo project through Xcode and get those logs, I will post them. I'm afraid I haven't done this before.

mrousavy, 2024-12-02:
tbh I am almost 100% sure that this is expected iPhone behaviour. There is nothing I can do about it.

closed 2024-12-02 by mrousavy

tjohnman, 2024-12-03:
> tbh I am almost 100% sure that this is expected iPhone behaviour. There is nothing I can do about it.

I understand, but it would be unfortunate if there was no way to use the wide angle lens with react-native-vision-camera. Just to clarify, is that the case?

kbrandwijk, 2025-02-25:
> No, that is not the case. Download ShadowLens from the AppStore, everything works as expected in there. That's using VisionCamera

That's actually not the case. The ShadowLens app also has the same noticeable delay in lens switching that started with iOS 18 that the native camera doesn't have.

mrousavy, 2025-02-25:
@kbrandwijk  I think what you're referring to is the macro-lens mode? Which happens when you get close to an object.