margelo/react-native-vision-camera #3234: 🔧 Build failed

margotbignon, 2024-10-08:
### How were you trying to build the app?

i tried to run my app with run android and i tried to delete .gradle folder

### Full build logs

```tsx
> Task :react-native-vision-camera:compileDebugKotlin
'compileDebugJavaWithJavac' task (current target is 11) and 'compileDebugKotlin' task (current target is 1.8) jvm target compatibility should be set to the same Java version.

> Task :react-native-vision-camera:compileDebugKotlin FAILED
e: Incompatible classes were found in dependencies. Remove them from the classpath or use '-Xskip-metadata-version-check' to suppress errors
e: C:/Users/Margot/.gradle/caches/transforms-3/0e44cbf3ca42951555f415c70890bf38/transformed/jetified-kotlinx-coroutines-android-1.8.1.jar!/META-INF/kotlinx-coroutines-android.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.6.0.
e: C:/Users/Margot/.gradle/caches/transforms-3/c071a63723be4b190409ab571b2af037/transformed/jetified-camera-lifecycle-1.4.0-beta02-api.jar!/META-INF/camera-lifecycle_release.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.8.0, expected version is 1.6.0.
e: C:/Users/Margot/.gradle/caches/transforms-3/c2a6d7dae8a940fa5f76c864e37eb1bc/transformed/jetified-kotlinx-coroutines-core-jvm-1.8.1.jar!/META-INF/kotlinx-coroutines-core.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.6.0.
e: C:/Users/Margot/.gradle/caches/transforms-3/d16f16bd76ea7a568824ffcac304d22d/transformed/jetified-camera-core-1.4.0-beta02-api.jar!/META-INF/camera-core_release.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.8.0, expected version is 1.6.0.
e: C:/Users/Margot/.gradle/caches/transforms-3/fd49cc402b4b657cf2e03265b0eeffbf/transformed/jetified-kotlin-stdlib-1.9.21.jar!/META-INF/kotlin-stdlib-jdk7.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.6.0.
e: C:/Users/Margot/.gradle/caches/transforms-3/fd49cc402b4b657cf2e03265b0eeffbf/transformed/jetified-kotlin-stdlib-1.9.21.jar!/META-INF/kotlin-stdlib-jdk8.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.6.0.
e: C:/Users/Margot/.gradle/caches/transforms-3/fd49cc402b4b657cf2e03265b0eeffbf/transformed/jetified-kotlin-stdlib-1.9.21.jar!/META-INF/kotlin-stdlib.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.6.0.
e: C:\Users\Margot\Documents\RVO-APP\node_modules\react-native-vision-camera\android\src\main\java\com\mrousavy\camera\core\CameraConfiguration.kt: (53, 33): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.6.0.
The class is loaded from C:/Users/Margot/.gradle/caches/transforms-3/fd49cc402b4b657cf2e03265b0eeffbf/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: C:\Users\Margot\Documents\RVO-APP\node_modules\react-native-vision-camera\android\src\main\java\com\mrousavy\camera\core\CameraConfiguration.kt: (79, 4): Class 'kotlin.Suppress' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.6.0.
The class is loaded from C:/Users/Margot/.gradle/caches/transforms-3/fd49cc402b4b657cf2e03265b0eeffbf/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Suppress.class
e: C:\Users\Margot\Documents\RVO-APP\node_modules\react-native-vision-camera\android\src\main\java\com\mrousavy\camera\core\CameraDeviceDetails.kt: (32, 20): Unresolved reference: atan2
```

### Project dependencies

```json
"react-native": "0.72.4",
    "react-native-vision-camera": "4.5.1",
    "react-native-worklets-core": "^1.3.3",
```

### VisionCamera Version

4.5.1

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

margotbignon, 2024-10-08:
Thanks a lot ! I followed the guide and I changed the Kotlin version to 1.9.22 and build failed again. I've these errors : java.lang.NullPointerException: Cannot invoke "String.length()" because "<parameter1>" is null in jetified-camera files. And this in Native reanimatedModule : C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:210:30: error: explicit capture of 'this' with a capture default of '=' is a C++20 extension [-Werror,-Wc++20-extensions]
C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:39:37: note: expanded from macro 'COPY_CAPTURE_WITH_THIS'
C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:276:30: error: explicit capture of 'this' with a capture default of '=' is a C++20 extension [-Werror,-Wc++20-extensions]
C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:39:37: note: expanded from macro 'COPY_CAPTURE_WITH_THIS'
C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:290:7: error: explicit capture of 'this' with a capture default of '=' is a C++20 extension [-Werror,-Wc++20-extensions]
C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:39:37: note: expanded from macro 'COPY_CAPTURE_WITH_THIS'
C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:389:7: error: explicit capture of 'this' with a capture default of '=' is a C++20 extension [-Werror,-Wc++20-extensions]
C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:39:37: note: expanded from macro 'COPY_CAPTURE_WITH_THIS'

Do you any idea why ? 

thanks a lot again !

mrousavy, 2024-10-10:
> e: C:/Users/Margot/.gradle/caches/transforms-3/d16f16bd76ea7a568824ffcac304d22d/transformed/jetified-camera-core-1.4.0-beta02-api.jar!/META-INF/camera-core_release.kotlin_module: Module was compiled [...]

You definitely need to use kotlin 1.8.0 or higher - see Example app: https://github.com/mrousavy/react-native-vision-camera/blob/dc43cdb8231d55e0afa22007b695e2ef4e09a840/package/example/android/build.gradle#L8

mrousavy, 2024-10-10:
> C:/Users/Margot/Documents/RVO-APP/node_modules/react-native-reanimated/Common/cpp/reanimated/NativeModules/NativeReanimatedModule.cpp:276:30: error: explicit capture of 'this' with a capture default of '=' is a C++20 extension [-Werror,-Wc++20-extensions]

this is a reanimated error.

closed 2024-10-10 by mrousavy