margelo/react-native-vision-camera #3235: 🔧 Execution failed for task ':app:mergeDebugNativeLibs'.

iamharisdev, 2024-10-10:
### How were you trying to build the app?

i tried to run my app with run android and before run the app I  run npm i  via using this my package version was updated 4.4.0 to 4.5.3

before update package everything was working fine

[image]

### Full build logs

```tsx
* What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> jni extracted from path /Users/FarrukhAyyaz/.gradle/caches/transforms-3/c13432dd313fe203e8d4707c5d5b1231/transformed/jetified-camera-core-1.5.0-alpha01/jni/riscv64/libimage_processing_util_jni.so is not an ABI * Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org. BUILD FAILED in 16s.
```

### Project dependencies

```json
"react-native": "0.73.6",
    "react-native-vision-camera": "4.5.3",
```

### VisionCamera Version

4.5.3

### Target platforms

iOS, Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jibo001, 2024-10-11:
same error

BogyeumKim, 2024-10-16:
same error
react native 0.73.6
react-native-vision-camera : 4.5.3

imamrobani, 2024-10-21:
same error
react native 0.73.9
react-native-vision-camera : 4.5.3