margelo/react-native-vision-camera #3237: 🐛 <Camera> is located at top left corner on its first start - while it is expected at the center of the window

oliviermartin, 2024-10-10:
### What's happening?

In my simple app, the first time I start it, the <Camera> is located at the top left corner of the screen while I expect it to be in the middle of the screen.
But when coming back on the screen, the <Camera> is now located in the middle of the screen.

What I see:
[image]

On the screen reload:
[image]

### Reproduceable Code

```tsx
import React from 'react';
import {
  Linking,
  StyleSheet,
  View,
} from 'react-native';
import type {CameraPermissionStatus} from 'react-native-vision-camera';
import {
  Camera,
  useCameraDevice,
} from 'react-native-vision-camera';

const styles = StyleSheet.create({
  content: {flex: 1, alignItems: 'center', justifyContent: 'center'},
  camera_view: {
    paddingBottom: 20,
    alignItems: 'center',
    justifyContent: 'center',
  },
  camera: {width: 150, height: 300, zIndex: 0},
});

function App(props): React.JSX.Element {
  const [isCameraActive, setActiveCamera] = React.useState(true);
  const [cameraPosition, setCameraPosition] = React.useState<'front' | 'back'>(
    'front',
  );
  const [cameraPermissionStatus, setCameraPermissionStatus] =
    React.useState<CameraPermissionStatus>('not-determined');
  const cameraRef = React.useRef<Camera>(null);
  let cameraDevice = useCameraDevice(cameraPosition);

  const requestCameraPermission = React.useCallback(async () => {
    const permission = await Camera.requestCameraPermission();
    console.log(`Camera permission status: ${permission}`);

    if (permission === 'denied') {
      await Linking.openSettings();
    }
    setCameraPermissionStatus(permission);
  }, []);

  React.useEffect(
    () => {
      if (cameraPermissionStatus !== 'granted') {
        requestCameraPermission();
      }
    },
  );

  console.log("cameraDevice:" + JSON.stringify(cameraDevice, (k, v) => k === "formats" ? [] : v, 2))

  return (
    <View style={styles.content}>
      { cameraDevice !== undefined && (
        <View style={styles.camera_view}>
[... 250 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung Galaxy A20e (Android 11)

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

oliviermartin, 2024-10-10:
I have just tried VisionCamera Example app. The application is in full screen.

I did modify the example app to be closer to my case and it seems to work (ie: the <Camera> is always in the middle of the screen):

```
--- a/package/example/src/CameraPage.tsx
+++ b/package/example/src/CameraPage.tsx
@@ -195,11 +195,12 @@ export function CameraPage({ navigation }: Props): React.ReactElement {
   return (
     <View style={styles.container}>
       {device != null ? (
+        <View style={styles.camera_view}>
         <PinchGestureHandler onGestureEvent={onPinchGesture} enabled={isActive}>
-          <Reanimated.View onTouchEnd={onFocusTap} style={StyleSheet.absoluteFill}>
+          <Reanimated.View onTouchEnd={onFocusTap} style={styles.camera}>
             <TapGestureHandler onEnded={onDoubleTap} numberOfTaps={2}>
               <ReanimatedCamera
-                style={StyleSheet.absoluteFill}
+                style={styles.camera}
                 device={device}
                 isActive={isActive}
                 ref={camera}
@@ -232,6 +233,7 @@ export function CameraPage({ navigation }: Props): React.ReactElement {
             </TapGestureHandler>
           </Reanimated.View>
         </PinchGestureHandler>
+        </View>
       ) : (
         <View style={styles.emptyContainer}>
           <Text style={styles.text}>Your phone does not have a Camera.</Text>
@@ -289,9 +291,15 @@ export function CameraPage({ navigation }: Props): React.ReactElement {
 
 const styles = StyleSheet.create({
   container: {
-    flex: 1,
-    backgroundColor: 'black',
+    flex: 1, alignItems: 'center', justifyContent: 'center',
+    backgroundColor: 'red',
+  },
+  camera_view: {
+    paddingBottom: 20,
+    alignItems: 'center',
+    justifyContent: 'center',
   },
+  camera: {width: 150, height: 300, zIndex: 0},
   captureButton: {
     position: 'absolute',
     alignSelf: 'center',
```

But the VisionCamera Example app is much more complex than my test app. I do not know whether I missed something.

cc: @mrousavy

oliviermartin, 2024-10-11:
I have been trying to reduce VisionCamera Example app code to simpler test app. **I managed few times to duplicate the issue with a minimal version of VisionCamera Example app but not as often as with my app.**

I would guess it is a timing / race condition issue somewhere in `react-native` or `react-native-vision-camera` when the camera is initialized - more likely `react-native-vision-camera` as I would have guessed people would have seen it more if it was a generic react-native issue. When the screen is reloaded with a camera already initialized, the issue disappears.

For my test app, I used latest react-native and I have enabled new react native architecture.

timokauppinen, 2024-10-21:
I'm having the same problem that the layout is set wrong occasionally (around 20% of the time) there is some race issue going on..

timokauppinen, 2024-10-21:
I had on camera view component

const device = useCameraDevice('back');
const [ hasCameraRights, setHasCameraRights ] = useState(false);

useEffect(() => {
(async () => {
	const cameraPermitted = await PermissionsAndroid.check('android.permission.CAMERA');
	if (cameraPermitted) {
		setHasCameraRights(true);
	}
})();
}, []);

if(!hasCameraRight)
  return <Text>No camera right</Text>
  
if (!device) {
  return <Text>No camera device</Text>;
}
return ..camera view...

After I removed that hasCameraRights check it started working correctly, so it has something to do with layout change timings.

timokauppinen, 2024-11-13:
> I am also experiencing this. I am on RN `0.72.17` on version `react-native-vision-camera: 4.3.1`.
> 
> I was attempting to get the camera to sit below a header but I was experiencing this exact problem where the camera pops to the top left. This only happened on android, a Pixel 3a API 33 for me, and didnt happen every single time but it would happen 1 in 10 renders of the camera.
> 
> I created a very very simple test project on RN `0.76.1` using version `react-native-vision-camera: 4.6.1` to test if the issue occurred on the latest versions as well.
> 
> Reproducible code -> https://github.com/FraserKemp/CameraReproduceCase
> 
> Again I tested on Pixel 3a API 33.

I did not test your code, but it's similar when I had problems. I'm sure if you take that rights check away it will work. In my case I removed the right check and added it outside to the camera open event before initializing the component. You could also add some right check wrapper and only initialize the camera component after have rights.

setafd, 2024-11-13:
I just found some strange... workaround?
I have code like this
```
<View style={{ flex: 1 }} >
  { hasPermission ? <Camera {...someProps} /> : <NoPermissionView /> }
</View>
```
After some hours of debugging, I added `{ borderColor: "transparent" }` to `<View />` style, and the issue was solved. I am not sure why or how this works, but it does in my case. Maybe it will help someone.

omerts, 2024-12-17:
I am also experiencing something similiar, if i open the camera right when the screen loads i am getting:

[image]

While it should be:

[image]

Above hack with using `borderColor` didn't help in my case

yulolimum, 2025-01-02:
Not sure why, but the most consistent "fix" for me was adding `overflow: hidden` to the wrapper view. Camera mounts correctly on initial render, subsequent renders, and during HMR.

mmpdrosa, 2025-01-06:
> Not sure why, but the most consistent "fix" for me was adding `overflow: hidden` to the wrapper view. Camera mounts correctly on initial render, subsequent renders, and during HMR.

It has worked for me.

ysoftid, 2025-01-10:
> Not sure why, but the most consistent "fix" for me was adding `overflow: hidden` to the wrapper view. Camera mounts correctly on initial render, subsequent renders, and during HMR.

no luck for me 😞

Prajwaltechversant, 2025-01-16:
i also have same issue in new architecture and rn latest

realabbas, 2025-03-05:
Adding `overflow` to the parent container worked

Julien-Allard, 2025-03-26:
> I just found some strange... workaround? I have code like this
> 
> ```
> <View style={{ flex: 1 }} >
>   { hasPermission ? <Camera {...someProps} /> : <NoPermissionView /> }
> </View>
> ```
> 
> After some hours of debugging, I added `{ borderColor: "transparent", borderWidth: 0 }` to `<View />` style, and the issue was solved. I am not sure why or how this works, but it does in my case. Maybe it will help someone.
> 
> UPD 1: It looks like `borderWidth: 0` isn't necessary.

The bug still is here, which is not good, but your fix works like a charm. The { borderColor: "transparent" } prevent the camera from moving on top for no reason if the <Camera> is conditioned on authorization in my case.

`<MedifilePageContainer headerType="global" headerLabel={l.common.labelScanQrCode} isBack>
      {device && isCameraActive && (
        // borderColor: 'transparent' is used to prevent a bug where the camera stays on top above the header
        <View style={{ flex: 1, borderColor: 'transparent' }}>
          <Camera style={{ height: '100%' }} device={device} isActive={isCameraActive} codeScanner={codeScanner} />
        </View>
      )}
    </MedifilePageContainer>`

mgulfam0722, 2025-04-15:
> Adding `overflow: hidden` to the parent container worked

It does solve the problem, but what exactly is the underlying problem?

mahesh-ivy, 2025-06-06:
not sure which one worked but i applied hidden: 'overflow' and borderColor: 'transparent' to the parent and it somehow worked. I was trying to have a square qr code scanner in the centre of the screen vertically but every time the feed was going up by ignoring the style of Camera tag and its parent when given permission and after it as well randomly when coming into the screen. Something is wrong with the layout things but somehow those 2 style props fixed it.