margelo/react-native-vision-camera #3245: 🐛 iOS 18 - Taking photo quickly causes error to be thrown

tgreco, 2024-10-15:
### What's happening?

When taking a photo as quick as possible and rage tapping the capture button twice. I get an error thrown.

[image]

[image]

### Reproduceable Code

```tsx
const takePhotoResult = await cameraRef.current.takePhoto({
          flash: 'off',
          enableAutoRedEyeReduction: false,
          enableAutoDistortionCorrection: false,
});

 <ReanimatedCamera
                ref={cameraRef}
                style={styles.fill}
                device={device}
                format={format}
                fps={FRAMES_PER_SEC}
                videoHdr={format?.supportsVideoHdr}
                photoHdr={format?.supportsPhotoHdr}
                // lowLightBoost={device.supportsLowLightBoost && isNightMode}
                isActive={isActive}
                onInitialized={onInitialized}
                onError={onError}
                enableZoomGesture={false}
                animatedProps={cameraAnimatedProps}
                photo
                video={mediaType === MediaType.video}
                audio={hasMicrophonePermission && mediaType === MediaType.video}
                enableLocation
                photoQualityBalance="speed"
              />
```

### Relevant log output

```shell
description of error:
error: type for self cannot be reconstructed: type for typename "$s12VisionCamera7PromiseCD" was not found
error: Couldn't realize Swift AST type of self. Hint: using `v` to directly inspect variables and fields may still work.
```

### Camera Device

```json
{
  "position": "back",
  "maxZoom": 123.75,
  "hasTorch": true,
  "formats": [],
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "sensorOrientation": "portrait",
  "isMultiCam": true,
  "minZoom": 1,
  "minFocusDistance": 2,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
  "hasFlash": true,
  "name": "Back Triple Camera",
  "minExposure": -8,
  "hardwareLevel": "full",
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ]
}
```

### Device

iPhone 16 - ios 18.1

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iliapnmrv, 2024-11-05:
Hey! I have the same issue in Sentry (400+ times, 150+ users), cannot reproduce in dev though. I'm using code scanner

```ts
const device = useCameraDevice('back', {
  physicalDevices: [
    'ultra-wide-angle-camera',
    'telephoto-camera',
    'wide-angle-camera',
  ],
});

const format =
  device &&
  getCameraFormat(device, [
    {fps: 20},
    {videoAspectRatio: screenAspectRatio},
    {videoResolution: 'max'},
    {photoAspectRatio: screenAspectRatio},
    {photoResolution: 'max'},
    {autoFocusSystem: 'contrast-detection'},
  ]);
```

```tsx
<GestureDetector gesture={tap}>
      <Camera
        format={format}
        zoom={device.neutralZoom}
        device={device}
        ref={cameraRef}
        isActive={isActive}
        onError={onError}
        codeScanner={codeScanner}
      />
</GestureDetector>
```

```json
{
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
  "maxZoom": 123.75,
  "sensorOrientation": "portrait",
  "minFocusDistance": 2,
  "position": "back",
  "formats": [],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera",
    "telephoto-camera"
  ],
  "hardwareLevel": "full",
  "isMultiCam": true,
  "hasTorch": true,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "name": "Rear triple camera",
  "hasFlash": true,
  "minExposure": -8,
  "minZoom": 1
}
```

@tgreco Did you manage to find a solution?

tgreco, 2024-11-05:
No solution as of yet unfortunately

felixaa, 2025-01-07:
@tgreco Any updates? Have you tried updating the package? We have the same issues, mostly on iPhone 16's. Some rare occurrences on older iPhones. 

[image]

tgreco, 2025-01-08:
No solution as of yet unfortunately. 
Cc @mrousavy