margelo/react-native-vision-camera #3247: My app closes automatically after a while of processing frames

kHanHnq4901, 2024-10-16:
### What's happening?

[image]

### Reproduceable Code

```tsx
36.989  9496-9595  com.meterwater          com.meterwater                       A  java_vm_ext.cc:710] JNI ERROR (app bug): global reference table overflow (max=51200)global reference table dump:
                                                                                                    java_vm_ext.cc:710]   Last 10 entries (of 51200):
                                                                                                    java_vm_ext.cc:710]     51199: 0x173080d8 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51198: 0x173080d8 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51197: 0x17306f18 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51196: 0x172a5ce8 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51195: 0x172a4b28 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51194: 0x172a3960 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51193: 0x172a2798 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51192: 0x172a15d8 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51191: 0x172a0418 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]     51190: 0x1729f258 com.mrousavy.camera.frameprocessors.Frame
                                                                                                    java_vm_ext.cc:710]   Summary:
                                                                                                    java_vm_ext.cc:710]     50586 of com.mrousavy.camera.frameprocessors.Frame (50585 unique instances)
                                                                                                    java_vm_ext.cc:710]       453 of java.lang.Class (337 unique instances)
                                                                                                    java_vm_ext.cc:710]        58 of com.facebook.react.bridge.JavaModuleWrapper (58 unique instances)
                                                                                                    java_vm_ext.cc:710]        13 of java.lang.ref.WeakReference (13 unique instances)
                                                                                                    java_vm_ext.cc:710]         5 of com.facebook.react.bridge.ReadableType (5 unique instances)
                                                                                                    java_vm_ext.cc:710]         5 of android.view.ViewRootImpl$W (5 unique instances)
                                                                                                    java_vm_ext.cc:710]         4 of android.content.ContentProvider$Transport (4 unique instances)
                                                                                                    java_vm_ext.cc:710]         3 of java.lang.String (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         3 of dalvik.system.PathClassLoader (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of com.mrousavy.camera.frameprocessors.VisionCameraProxy (1 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of com.mrousavy.camera.frameprocessors.VisionCameraScheduler (1 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of com.swmansion.reanimated.AndroidUIScheduler (1 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of com.swmansion.reanimated.layoutReanimation.LayoutAnimations (1 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of android.hardware.camera2.CameraManager$CameraManagerGlobal (1 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of android.view.ThreadedRenderer$WebViewOverlayProvider (1 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of com.facebook.react.bridge.queue.MessageQueueThreadImpl (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of float[] (16 elements) (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of int[] (16 elements) (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of java.lang.ThreadGroup (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of android.opengl.EGLContext (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of android.opengl.EGLDisplay (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         2 of android.opengl.EGLSurface (2 unique instances)
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.react.bridge.RuntimeScheduler
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.react.bridge.RuntimeExecutor
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.react.modules.blob.BlobModule
                                                                                                    java_vm_ext.cc:710]         1 of com.swmansion.rnscreens.ScreensModule
                                                                                                    java_vm_ext.cc:710]         1 of com.reactlibrary.RNOpenCvLibraryModule
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.flipper.plugins.network.NetworkFlipperPlugin
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.flipper.plugins.inspector.InspectorFlipperPlugin
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.flipper.plugins.databases.DatabasesFlipperPlugin
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.flipper.plugins.sharedpreferences.SharedPreferencesFlipperPlugin
                                                                                                    java_vm_ext.cc:710]         1 of com.facebook.flipper.plugins.crashreporter.CrashReporterPlugin
[... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kHanHnq4901, 2024-10-16:
@mrousavy Please help me

kHanHnq4901, 2024-10-16:
source   
// Bộ xử lý khung
const frameProcessor = useSkiaFrameProcessor((frame) => {
    'worklet';

    // Hàm xử lý OpenCV
    const openCvWorklet = (frame: FrameInternal) => {
        'worklet'; // Đánh dấu hàm này là worklet
        try {
            frame.incrementRefCount();
            const map = openCv(frame); // Giả định openCv là hàm xử lý OpenCV
            return map;
        } finally {
            frame.decrementRefCount();
        }
    };

    // Hàm vẽ trên khung hình
    const drawOnFrame = (map: object) => {
        'worklet'; // Đánh dấu hàm này là worklet
        if (!map) return;
        frame.render(); // Render the frame before drawing
        const paint = Skia.Paint();
        paint.setColor(Skia.Color('red'));
        frame.drawLine(map.center_x - 10, map.center_y, map.center_x + 10, map.center_y, paint);
        frame.drawLine(map.center_x, map.center_y - 10, map.center_x, map.center_y + 10, paint);

        if (map.ptAx) {
            const paintRed = Skia.Paint();
            paintRed.setColor(Skia.Color('red'));
            const paintYellow = Skia.Paint();
            paintYellow.setColor(Skia.Color('yellow'));
            const paintBlue = Skia.Paint();
            paintBlue.setColor(Skia.Color('blue'));
            const paintBlack = Skia.Paint();
            paintBlack.setColor(Skia.Color('black'));

            frame.drawLine(map.ptAx, map.ptAy, map.ptBx, map.ptBy, paintRed);
            frame.drawLine(map.ptAx, map.ptAy, map.ptCx, map.ptCy, paintYellow);
            frame.drawLine(map.ptBx, map.ptBy, map.ptCx, map.ptCy, paintBlue);
            frame.drawLine(map.ptIx, map.ptIy, map.ptCx, map.ptCy, paintBlack);
        }
    };

    // Gọi hàm xử lý OpenCV và vẽ trên khung hình
    const nativeBuffer = frame.getNativeBuffer(); // Lưu trữ nativeBuffer
    const map = openCvWorklet(frame); // Gọi hàm xử lý OpenCV
    // Vẽ trên khung hình
    drawOnFrame(map);
    const image = Skia.Image.MakeImageFromNativeBuffer(nativeBuffer.pointer);
    image.dispose(); 
    // Xóa buffer sau khi hoàn tất
    nativeBuffer.delete(); 
}, []);
<Camera
                key={cameraKey}
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={isActive}
                zoom={zoom ? 7 : 1}
                frameProcessor={frameProcessor}
                enableFpsGraph={true}
                format={format}
                fps={[minFps, maxFps]}
                enableBufferCompression = {true}
            />

kHanHnq4901, 2024-10-18:
@mrousavy

lukachi, 2025-02-24:
Facing same issues with app crashes, described it here: https://github.com/Shopify/react-native-skia/discussions/2675#discussioncomment-12300270