margelo/react-native-vision-camera #3248: 🐛 [Android] CodeScanner doesn't work at all on Samsung Galaxy A23 (Android 14)

MaksymKuzmych, 2024-10-16:
### What's happening?

Issue: QR Code Scanner Fails on Samsung A23 (Android 14)

After updating from version 3.8.2 to 4.5.3, I've encountered an issue on the Samsung A23 running Android 14. The QR code scanner doesn't appear, and I'm receiving the following error:

```[session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]```

If I remove either the "codeScanner" or "format" prop, the camera will mount, but it won't scan any QR codes.

Code Sample (CodeScannerPipeline.kt)

```override fun analyze(imageProxy: ImageProxy) {
    val image = imageProxy.image ?: throw InvalidImageTypeError()

    try {
        val inputImage = InputImage.fromMediaImage(image, imageProxy.imageInfo.rotationDegrees)
        scanner.process(inputImage)
            .addOnSuccessListener { barcodes ->
                if (barcodes.isNotEmpty()) {
                    callback.onCodeScanned(barcodes, CodeScannerFrame(inputImage.width, inputImage.height))
                }
            }
            .addOnFailureListener { error ->
                Log.e(TAG, "Failed to process Image!", error)
                callback.onError(error)
            }
            .addOnCompleteListener {
                imageProxy.close()
            }
    } catch (e: Throwable) {
        Log.e(TAG, "Failed to process Image!", e)
        imageProxy.close()
    }
}
```

Issue Details:
- On Samsung A23 (Android 14), barcodes.isNotEmpty() always returns false.
- On other devices, the same code works as expected and barcodes.isNotEmpty() becomes true.
Before updating to 4.5.3, everything worked fine on the Samsung A23 as well.

Has anyone encountered this issue or found a workaround? Any assistance would be appreciated!

### Reproduceable Code

```tsx
const device = useCameraDevice('back');
const format = useCameraFormat(device, [{videoResolution: 'max'}]);

const isFocused = useIsFocused();
const isForeground = useIsForeground();

const isActive = isFocused && isForeground;

const codeScanner = useCodeScanner({
  codeTypes: ['qr', 'ean-13'],
  onCodeScanned: codes => {
    console.log('codes', codes);
  },
});

<Camera
 device={device}
 key={device.id}
 isActive={isActive}
 focusable
 format={format}
 style={cameraStyles}
 codeScanner={codeScanner}
[... 240 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy A23 (Android 14)

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

MaksymKuzmych, 2024-10-16:
> Guten Tag, Hans here! 🍻
> 
> This looks like a valid issue and you provided relevant logs, Danke! It seems you have detailed the behaviour on your Samsung Galaxy A23 with Android 14 after updating. However, not being able to reproduce the issue in the Example app suggests that it could be device-specific or related to your specific configuration. I recommend double-checking all your camera permissions and settings.
> 
> If the issue persists, please monitor for updates from mrousavy. Feel free to sponsor the project [here](https://github.com/sponsors/mrousavy) to support further development!
> 
> > Note: If you think I made a mistake, please ping `@mrousavy` to take a look.

But it's working with version 3.8.2 of the same model, so it shouldn't be a problem with permissions or settings.

zwerg44, 2024-10-16:
Hi guys,
Thanks for a great work on `react-native-vision-camera`

I'm also experiencing same issue with `Samsung A23 5G`  (`SM-A236B`)
I haven't opened ticket yet, trying to dig for some details firsts, but I don't have the `A23` device.
I upgraded from `2.16.1` to `4.5.2` recently and my users with `A23` complains that built-in QR-scanning not works.

I'm using now:
- `react-native-vision-camera 4.5.2`
- with built-in `codeScanner`
- VisionCamera_enableFrameProcessors=false
- VisionCamera_enableCodeScanner=true
- `format` prop not passed in my case
```jsx
<Camera
  isActive
  enableZoomGesture
  ref={cameraRef}
  device={device}
  lowLightBoost={device.supportsLowLightBoost && nightModeEnabled}
  codeScanner={codeScanner}
  torch={isCameraMounted && supportsFlash && flash ? 'on' : 'off'}
  onError={onError}
  onInitialized={onInitialized}
  style={StyleSheet.absoluteFill}
/>
```

@MaksymKuzmych do you have `A23` phone on your side to debug this?
Could you maybe try to:
- and `const format = useCameraFormat(device, [{ videoResolution: { width: 640, height: 480 } }])`
- and to use `frameProcessor` instead of `codeScanner`
  - I think `codeScanner` ignores `format` prop
    - https://github.com/mrousavy/react-native-vision-camera/blob/main/package/android/src/main/java/com/mrousavy/camera/core/CameraSession%2BConfiguration.kt#L188
    - also I found this line removed in 4.4.0 https://github.com/mrousavy/react-native-vision-camera/pull/2997/files
- here is my sample: https://gist.github.com/zwerg44/f7ebe18d4a2db0b1304188af6afd4dfa

Some details from my users:
- I was able to get a video from my users showing
  - QR reading not works
  - but 1 linear 1D barcode was recognized once at same video at corner of the screen
  - so, I think MLkit is actually receiving some frames input, but possibly frames are corrupted

I found this topic on flutter camera library: 
- https://github.com/juliansteenbakker/mobile_scanner/issues/698#issuecomment-1682999110
- https://github.com/juliansteenbakker/mobile_scanner/issues/698#issuecomment-1903491693
  - for flutter guys with A23 - it helped to hardcode to video resolution to 640x480
    - see the PR https://github.com/digitalfabrik/entitlementcard/pull/1524/files#diff-9b228a2d6b0d3af8a00f53730290cc9c2812eaa1765335d72410388486487b09R44
  - also there are suggestion there to switch encoding from `YUV_420_888` to `NV21`, that it could help

I'm digging in this direction now:
- trying to debug output frame passed from `frame-processor` to `mlkit`
- to convert frame `yuv` to `rgb` and ouput to `skia`
- trying this example https://github.com/mrousavy/vision-camera-resize-plugin/blob/main/example/src/App.tsx

zwerg44, 2024-10-16:
@MaksymKuzmych 

And here is a sample with debug output of frame to skia:
https://gist.github.com/zwerg44/2284ce92160f7c93184f2b2b899b0da1

But on my phone and also my emulator:
- resize plugin works correctly only in this case
  -  if I pass to plugin scale `320x320` and `pixelFormat: 'abgr'`
  - and then render it with skia as `updatePreviewImageFromData(data, 'rgba').then(() => data.dispose())`
- without passing `scale` or passing other scale - output has corrupted `offsets`
- with other `pixelFormat` combination - output colour not correct

Still not sure why? is it issue with resize-plugin? or 4.5.2  produces corrupted frames?

zwerg44, 2024-10-16:
Updated my gist sample with debug output to skia:
- to not use `scale` (I think I understood scale param usage wrong)
https://gist.github.com/zwerg44/2284ce92160f7c93184f2b2b899b0da1

MaksymKuzmych, 2024-10-17:
> Updated my gist sample with debug output to skia:
> 
> * to not use `scale` (I think I understood scale param usage wrong)
>   https://gist.github.com/zwerg44/2284ce92160f7c93184f2b2b899b0da1

Hi, @zwerg44 
Thanks for the answer. The first thing I tried to do was just replace my

```const format = useCameraFormat(device, [{videoResolution: 'max'}]);```

with the format in your suggestion

```const format = useCameraFormat(device, [{videoResolution: {width: 640, height: 480}}]);```

and for now it works and I don't receive the Error

```[session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]```

Yes, the quality is worse than before, but it scanning QR codes really good without any problems, so I just did the helper function just for this model:

```export const getVideoResolution = () => {
  const deviceName = DeviceInfo.getDeviceNameSync();

  return deviceName === 'Galaxy A23 5G' ? {width: 640, height: 480} : 'max';
};
```

zwerg44, 2024-10-17:
I also confirm that just passing `640x480` `format` helped all my users with `A23 5G`

I used workaround like this:
```js
// constants
const MODEL = DeviceInfo.getModel();
const MODELS_HAS_CAMERA_ISSUE = [
  'SM-A235F',
  'SM-A235M',
  'SM-A235N',
  'SM-A233C',
  'SM-A2360',
  'SM-A236B', // my users
  'SM-A236E',
  'SM-A236M',
  'SM-A236U',
  'SM-A236U1',
  'SM-S236DL',
  'SM-S237VL',
  'SM-A236V',
];
const HAS_CAMERA_ISSUE = MODELS_HAS_CAMERA_ISSUE.includes(MODEL);

...
// and then passed format like this to camera
const selectedFormat = useCameraFormat(device, [{ videoResolution: { width: 640, height: 480 } }]);
const format = HAS_CAMERA_ISSUE ? selectedFormat : undefined;
```

glecompte, 2024-11-18:
I'm unable to get the camera preview to load with codescanner enabled on any emulator using API 34. For now i've downgraded to 4.3.2 and that seems to have solved it.

olivertylsar, 2025-01-03:
I'm also facing this issue with the A23 model using the `useCodeScanner`. 

If I understand it correctly, the frame processors have to be used to be able to specify custom format? If so, should it be used only for Android and keep using `useCodeScanner` for iOS? There are strong arguments against using MLKit for iOS as the native reader is very realiable and performant - https://github.com/mrousavy/react-native-vision-camera/issues/2701#issuecomment-2031894314. And also from my personal experience, there was never a single issue with reading QR codes on iOS devices.

Can anything be done on the library side instead of implementing this workaround for specific devices? @mrousavy

zwerg44, 2025-01-03:
Hello @olivertylsar 

No, frame processors was not required to fix this.
Just to specify `640x480` resolution with `useCodeScanner` for A23 was enough to fix.

`const selectedFormat = useCameraFormat(device, [{ videoResolution: { width: 640, height: 480 } }]);`

Seams like `default` resolution and also `max` resolution for A23 is broken (frame returns cropped top-right corner of visible image).

mauripavan, 2025-02-06:
Same issue here. It had been fixed with this line:

`const selectedFormat = useCameraFormat(device, [{ videoResolution: { width: 640, height: 480 } }]);`

After upfrading RN version to 0.76 and the react-native-vision-camera package to 4.6.3, the issue came up again.

Any idea how it can be fixed? I try removing the format prop with no luck.

TrejoCode, 2025-02-20:
I have the same issue with versión 4.3.1.

vpszi, 2025-04-09:
I'm having the same problem on my A23 device.

When i try to set a format, i get the following error:
 [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]

If i set `photo={true}` the format can be set without error.

I finally solved the issue with these properties: `photo={true} photoQualityBalance={'speed'}`

hope this can help someone else.

zwerg44, 2025-05-16:
Hello @vpszi,
I have the same issue after update from `4.5.3` to `4.6.4`
With `4.5.3` we used hardcoded `640x480` format for `A23`
Did `photo={true} photoQualityBalance={'speed'}` helped for you without hardcoding `640x480` format for `A23` or did you kept `640x480` format at you side?

vpszi, 2025-05-16:
@zwerg44 exactly, the problem has been resolved without setting `640x480`