margelo/react-native-vision-camera #3251: ✨ CodeScanner QR version 40

Henus321, 2024-10-18:
### What feature or enhancement are you suggesting?

I am using react-native-vision-camera v4.5.1 with codeScanner. In my project i have to scan realy big QR code "version 40". Outside the app, the phone camera recognizes the code, but in the app with react-native-vision-camera the code is not recognized. It would be nice to implement support for this version of QR codes.

[image]

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

I haven't found any alternatives at the moment

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

xulihang, 2024-11-01:
Google MLKit and Dynamsoft can all read this high version QR code using this demo: https://github.com/tony-xlh/react-native-zxing-mlkit-dynamsoft

Henus321, 2024-11-08:
[react-native-zxing](https://github.com/rafaeltakiyama/react-native-zxing) this library and the device with the better camera helped me, for now I will use both libraries. The largest code is not always perfectly readable, but I have almost achieved the desired result.

mrousavy, 2025-02-17:
Use Dynamsoft Zxing

closed 2025-02-17 by mrousavy