margelo/react-native-vision-camera #3253: 🔧 No such a file or directory found

Fahadayyoub, 2024-10-20:
### How were you trying to build the app?

in Android, if we try to rebuild the project we face this issue

### Full build logs

```tsx
node_modules/react-native-vision-camera/android/build/intermediates/cxx/ndk_locator_record_v4yjh1kg.log (No such file or directory)
```

### Project dependencies

```json
"dependencies":{
 "react": "18.2.0",
    "react-native": "0.73.0",
    "react-native-reanimated": "^3.16.0",
    "react-native-vision-camera": "^4.5.3",
    "react-native-worklets-core": "^1.3.3"
}
```

### VisionCamera Version

4.5.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Fahadayyoub, 2024-10-20:
@mrousavy xan you please look into this I am unable to install in android

mrousavy, 2024-10-21:
> I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

Sorry, but I cannot help you if you dont try to build example app.