margelo/react-native-vision-camera #3254: 🐛 react native vision camera don't scan pdf-417 from real documents

fmoraless, 2024-10-21:
### What's happening?

I have implemented react native vision camera in my react native 0.75 application.

my objective is to scan tickets, which have the code pdf-417

Following the [documentation](https://react-native-vision-camera.com/docs/guides), and doing some tests, I have been able to verify that the pdf417 scanning works correctly with ios. But when scanning the same document with android it does not read the pdf417 of the sale receipts.

I was doing some tests to solve this problem and it turns out that when creating a pdf417 with few characters, it is possible to read it (page to create pdf417 [cognex](https://www.cognex.com/resources/interactive-tools/free-barcode-generator)).

But the problem is that the voucher or ticket ,has a standard that has a lot of characters (it is an Xml with all the data of the ballot, social reason, trade rut, amount, date, etc).

I appreciate your support
Tks

### Reproduceable Code

```tsx
import {Alert, Pressable, StyleSheet, Text, View} from 'react-native';
import React, {useCallback, useEffect, useRef, useState} from 'react';
import {
  Camera,
  Templates,
  useCameraDevice,
  useCameraFormat,
  useCameraPermission,
  useCodeScanner,
} from 'react-native-vision-camera';
import {useIsFocused} from '@react-navigation/native';
import {Loader} from '../../components/Loader';
import type {Code} from 'react-native-vision-camera';
import {Icon, IconButton} from 'react-native-paper';
import {useAppState} from '../../hooks/useAppState';

export const ScannerScreen = ({navigation}) => {
  //  const [mode, setMode] = useState('camera');

  const device = useCameraDevice('back');
  //console.log('device', device?.hardwareLevel);

  //const devices = Camera.getAvailableCameraDevices();
  //console.log(JSON.stringify(devices, null, 2));

  const format = useCameraFormat(device, Templates.Instagram);

  const showCodeAlert = (value: string, onDismissed: () => void): void => {
    console.log('showCodeAlert', value);
    isShowingAlert.current = true;
    Alert.alert('Code Scanned', value, [
      {
        text: 'OK',
        onPress: () => {
          isShowingAlert.current = false;
          onDismissed();
        },
      },
    ]);
  };

  const isShowingAlert = useRef(false);

  const onCodeScanned = useCallback((codes: Code[]) => {
    console.log('onCodeScanned', codes);
    console.log(`Scanned ${codes.length} codes!`);
    const value = codes[0].value;
    console.log('Scanned value:', value);
    if (value == null) {
      return;
    }

[... 202 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy s20FE

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fmoraless, 2024-10-21:
Here there is a log, while vision camera app it was running.

I'm using react native vision camera 4.5.3
 "react-native": "0.75.3",

Thanks

[logcat vision camera.txt](https://github.com/user-attachments/files/17455142/logcat.vision.camera.txt)

EdzonBolivar11, 2024-10-23:
Same issue here, trying to scan a barcode (pdf-417) from license driver of USA, going deeper seems like when the barcode has more than 250 characters it doesn't scan

fmoraless, 2024-11-05:
I've looked in the [documentation page ](https://react-native-vision-camera.com/docs/guides/frame-processor-plugins-community), in frame processor plugins, and I've tried with some listed plugins and I have read barcodes, 6 of 10 pdf-417 barcodes were read. I guess is an issue about the processor performance. I need to do more tests about it, but is the beginning.

Regards

EdzonBolivar11, 2024-11-05:
Which plugins did you used? @fmoraless

xulihang, 2024-12-02:
You can try other frame processor plugins:

* vision-camera-zxing
* vision-camera-dynamsoft-barcode-reader
*

felipebaisi, 2025-04-09:
vision-camera-zxing doesn't really work
vision-camera-dynamsoft-barcode-reader it's not free - if you'd use them to parse your frames you might use their own solution for scanning.