margelo/react-native-vision-camera #3255: runAsync(..) cannot be used

kHanHnq4901, 2024-10-21:
### What's happening?

[image]

### Reproduceable Code

```tsx
import React, { useCallback, useEffect, useRef, useState } from 'react';
import { StyleSheet, View, Text, Button, StatusBar, Alert, TouchableOpacity } from 'react-native';
import { onEndPress, onResetPress, onStartPress ,onSavePress, onDisconnectPress, changeSaturation} from './handle';
import { store, GetHookProps } from './controller';
import { Camera, runAtTargetFps, useSkiaFrameProcessor,  useCameraFormat, getCameraDevice,runAsync, CameraPermissionRequestResult, useFrameProcessor } from "react-native-vision-camera";
import { Points, Skia } from '@shopify/react-native-skia';
import { openCv } from './controller';
import { useIsFocused } from '@react-navigation/native';
import LinearGradient from 'react-native-linear-gradient'; 
import { onSavePressSetting } from '../setting/handle';
import Slider from '@react-native-community/slider';
import { Worklets, useSharedValue } from 'react-native-worklets-core';

const Header = () => (
  <View style={styles.header}>
    <Text style={styles.headerText}>Phần mềm kiểm sai số Đồng hồ nước</Text>
  </View>
);

const App =(props: { navigation: any }) => {
  GetHookProps();
  const { navigation } = props;
  const isFocused = useIsFocused();
  const [appState] = useState("active");
  const isActive = isFocused && appState === "active";
  const [hasCameraPermission, setHasCameraPermission] = useState<boolean>(false);
  const [cameraKey, setCameraKey] = useState(0);
  const [zoom, setZoom] = useState(false);
  const [lastPress, setLastPress] = useState(0);
  const onZoomPress = () => {
    const now = Date.now();
    if (now - lastPress < 1000) { // 3 giây
      Alert.alert('Thông báo', 'Thao tác quá nhanh');
      return;
    }
    setLastPress(now);
    setZoom(!zoom);
  };

  useEffect(() => {
    const unsubscribe = navigation.addListener('focus', () => {
      // Hàm này sẽ được gọi khi màn hình ScanQrCodeScreen được focus (hiển thị)

      console.log('App Screen is focused');
      setCameraKey(prevKey => prevKey + 1);// Update cameraKey to refresh the camera
      setZoom(false)
    });
    
    const checkPermissions = async () => {
      const status: CameraPermissionRequestResult = await Camera.requestCameraPermission();
      
      if (status === 'granted') {
          setHasCameraPermission(true);
      } else {
          setHasCameraPermission(false);
          Alert.alert('Quyền truy cập camera đã bị từ chối, vui lòng cấp quyền cho camera.');
      }
  };
    checkPermissions();
    return unsubscribe;
  }, [navigation]);
  const devices = Camera.getAvailableCameraDevices()
  const device = getCameraDevice(devices, 'back', {
[... 184 lines cut ...]
  "supportsFocus": true,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false
}
```

### Device

Samsung Galaxy A03s

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

NFSMONSTR, 2024-10-22:
May be related to #2820

Also for me helped setting babel.config like this
```
...
 plugins: [
      ['react-native-worklets-core/plugin'],
      ['react-native-reanimated/plugin', {processNestedWorklets: true}]
...      
```

kHanHnq4901, 2024-10-23:
@NFSMONSTR I have set up babel.config like that and got this error
[image]

kHanHnq4901, 2024-10-28:
That doesn't work, and it straight up crashes... this is my crash report from flipper:

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***

Unknown

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***

Build fingerprint: 'google/cheetah/cheetah:14/UQ1A.240105.004/11206848:user/release-keys'
Revision: 'MP1.0'
ABI: 'arm64'
Timestamp: 2024-02-09 19:49:09.838956168+0200
Process uptime: 19s
Cmdline: com.*****
pid: 22746, tid: 22956, name: mqt_js  >>> com.*********** <<<
uid: 10398
tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x0000006ebd02d42d
    x0  0000000000000001  x1  00000074e5511248  x2  0000000000000000  x3  0000000000000000
    x4  00000074e7805f10  x5  b400007537dc59e8  x6  0000000000000000  x7  00000073a34d88e3
    x8  0000006e7e02c4e8  x9  0000006ebd02d408  x10 0000000000000003  x11 00000000dd000000
    x12 00000074e78061c8  x13 0000000000000051  x14 0000000000000001  x15 0000006e7e022570
    x16 0000000000000051  x17 0000000000000600  x18 00000074e769c000  x19 0000006e7dc00000
    x20 00000074e7805d68  x21 00000074e7809000  x22 00000074e5511280  x23 fffa000000000000
    x24 ffff006e7e02c4e8  x25 0000000000bfef50  x26 0000000000000000  x27 b400007537dc5990
    x28 00000074e5511138  x29 00000074e7805d40
    lr  0000007448c13480  sp  00000074e7805d40  pc  0000007448c1349c  pst 0000000060001000
72 total frames
backtrace:
      #00 pc 000000000016149c  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #01 pc 00000000000ce538  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #02 pc 00000000000cc7c0  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #03 pc 0000000000106888  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #04 pc 00000000000ef104  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #05 pc 00000000000ec554  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #06 pc 00000000000ce830  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #07 pc 0000000000095060  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #08 pc 000000000005a850  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libVisionCamera.so (offset 0x11b52000) (RNWorklet::JsiWorklet::call(std::__ndk1::shared_ptr<facebook::jsi::Function>, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)+332) (BuildId: 4024460c6f4361b8a3019c1cb6706c086a77f83f)
      #09 pc 0000000000059dd8  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libVisionCamera.so (offset 0x11b52000) (RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)+140) (BuildId: 4024460c6f4361b8a3019c1cb6706c086a77f83f)
      #10 pc 00000000001a02a4  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const+68) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
      #11 pc 00000000001a023c  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
      #12 pc 00000000001a0188  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (facebook::jsi::Value std::__ndk1::__invoke_void_return_wrapper<facebook::jsi::Value>::__call<RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long>(RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*&&, unsigned long&&)+120) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
[... cut ...]
      #68 pc 0000000000179874  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
      #69 pc 000000000017919c  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.*******-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
      #70 pc 00000000000c9ccc  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+204) (BuildId: 19c32900d9d702c303d2b4164fbba76c)
      #71 pc 000000000005db00  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 19c32900d9d702c303d2b4164fbba76c)

xulihang, 2024-10-31:
This seems a conflict with react-native-reanimated as they all use the `worklet` keyword.

babldev, 2024-11-23:
Seeing this issue on iOS and Android.

```ts
const frameProcessorHandler = (frame: Frame) => {
    'worklet'
    runAsync(frame, () => {
      'worklet'
    });
  };
```

- react-native-vision-camera 4.6.1
- react-native-worklets-core 1.5.0

babldev, 2024-11-23:
Related https://github.com/margelo/react-native-worklets-core/issues/136

ChristopherGabba, 2025-02-10:
+1 on this not even sure what happened. I felt like I just rebuilt the app and it's suddenly crashing.

ncoen97, 2025-02-20:
Same issue here

predofrazao, 2025-02-24:
Facing the same issue here. My provisory solution is use `runAtTargetFps`.

```ts
// ...

 const handleDetection = Worklets.createRunOnJs(async (faces: Face[]) => { /* ... */ });

 const frameProcessor = useFrameProcessor((frame) => {
    "worklet";
    runAtTargetFps(4, () => {
      "worklet";
      const faces = detectFaces(frame);
      handleDetection(faces);
    });
  }, []);

// ...
```

wangyuxu, 2025-03-26:
> Facing the same issue here. My provisory solution is use `runAtTargetFps`.
> 
> // ...
> 
>  const handleDetection = Worklets.createRunOnJs(async (faces: Face[]) => { /* ... */ });
> 
>  const frameProcessor = useFrameProcessor((frame) => {
>     "worklet";
>     runAtTargetFps(4, () => {
>       "worklet";
>       const faces = detectFaces(frame);
>       handleDetection(faces);
>     });
>   }, []);
> 
> // ...

I want to ask, will the app crash if I use useState hooks for rendering the page inside the handleDetection method? How did you solve it?
the crash for android logcat is Abort message: 'JNI DETECTED ERROR IN APPLICATION: JNI GetObjectRefType called with pending exception java.lang.IllegalStateException: register failed, the sensor listeners size has exceeded the maximum limit 128 @

predofrazao, 2025-03-26:
> I want to ask, will the app crash if I use useState hooks for rendering the page inside the handleDetection method? How did you solve it? the crash for android logcat is Abort message: 'JNI DETECTED [...]

You mean update states inside the `handleDetection` function? If yes, I didn't have problem with this, I even update some states inside it too, like a multiple faces boolean indicator.

```tsx
const handleDetection = Worklets.createRunOnJS(async (faces: Face[]) => {
      if (faces.length > 1) {
        setHasMultipleFaces(true); // <-- state update ✅
        return;
      }

      // ...
}
```

Maybe you trying to update state inside a `'worklet'` function? AFAIK this is not possible, since they run outside JS thread.