margelo/react-native-vision-camera #3256: 🐛 Setting Camera format prop seems like disabling antialiasing on preview

NFSMONSTR, 2024-10-22:
### What's happening?

Thanks for this great library!

I tried to set different camera formats and noticed that setting any of them leads to poor preview quality(seems like some default antialiasing disabled). If remove this parameter, the preview looks great.

I was able to reproduce this in example app(At the moment I have only tested it on Android).

So if camera format is set preview look like this:
[image]
[image]

If remove camera format prop:
[image]

### Reproduceable Code

```tsx
//From example

//Screenshots 1 and 2
//...
let device = useCameraDevice(cameraPosition)
//...
 const screenAspectRatio = SCREEN_HEIGHT / SCREEN_WIDTH
  const format = useCameraFormat(device, [
    { fps: targetFps },
    { videoAspectRatio: screenAspectRatio },
    { videoResolution: 'max' },
    { photoAspectRatio: screenAspectRatio },
    { photoResolution: 'max' },
  ])
//...
<ReanimatedCamera
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={isActive}
                ref={camera}
                onInitialized={onInitialized}
                onError={onError}
                onStarted={() => console.log('Camera started!')}
                onStopped={() => console.log('Camera stopped!')}
                onPreviewStarted={() => console.log('Preview started!')}
                onPreviewStopped={() => console.log('Preview stopped!')}
                onOutputOrientationChanged={(o) => console.log(`Output orientation changed to ${o}!`)}
                onPreviewOrientationChanged={(o) => console.log(`Preview orientation changed to ${o}!`)}
                onUIRotationChanged={(degrees) => console.log(`UI Rotation changed: ${degrees}°`)}
                format={format}
                fps={fps}
                photoHdr={photoHdr}
                videoHdr={videoHdr}
                photoQualityBalance="quality"
                lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                enableZoomGesture={false}
                animatedProps={cameraAnimatedProps}
                exposure={0}
                enableFpsGraph={true}
                outputOrientation="device"
                photo={true}
                video={true}
                audio={microphone.hasPermission}
                enableLocation={location.hasPermission}
                frameProcessor={frameProcessor}
              />
//...

// Screenshoot 3
<ReanimatedCamera
                style={StyleSheet.absoluteFill}
                device={device}
[... 173 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

POCO X5 Pro 5G (Android 14)

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

NFSMONSTR, 2024-10-22:
Also here is full logs:
[issue_logs.txt](https://github.com/user-attachments/files/17475073/issue_logs.txt)

ugar0ff, 2024-11-15:
I have exactly the same problem on iPhone 8 and iPhone 14 Pro
```
const format = useCameraFormat({
  camera,
  Templates.PhotoPortrait,
})

<Camera
  ref={cameraRef}
  photo
  resizeMode="contain"
  device={camera}
  isActive={isActive}
  outputOrientation="preview"
  // format={format}
  photoQualityBalance="quality"
  style={style}
/>
```
with format={format} very bad preview format (only on the back camera), without format={format} - perfect

NFSMONSTR, 2024-11-15:
@ugar0ff Hello, try to use camera format like this
```
const format = useCameraFormat(camera, [
    {videoResolution: 'max'},
  ]);
```

is seems like on ios preview quality depends on videoResolution of format

It if helps and you want to use PhotoPortrait template you could try something like this:
```
const format = useCameraFormat({
  camera,
  [...Templates.PhotoPortrait, {videoResolution: 'max'}],
})
```

ugar0ff, 2024-11-15:
this helped, thank you!