margelo/react-native-vision-camera #3259: 🐛 In landscape mode, enabling frameProcessor causes preview image stretching and incorrect orientation

boiboif, 2024-10-24:
### What's happening?

In landscape mode

Preview without frameProcessor:
[image]

Preview with frameProcessor enabled:
[image]

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor(
    frame => {
      'worklet';
    },
    [handleDetectedFaces],
  );

<Camera
  ref={camera}
  style={StyleSheet.absoluteFill}
  device={device}
  isActive={true}
  photo
  // frameProcessor={frameProcessor}
/>
```

### Relevant log output

```shell
.
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -24,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

redmi k50

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

meleffendi, 2024-12-01:
I'm having the same issue on 4.5.2 on OnePlus5. Any ideas?

boiboif, 2024-12-02:
> I'm having the same issue on 4.5.2 on OnePlus5.我在 OnePlus5 的 4.5.2 上遇到了同样的问题。
> 
> In my case, when the phone is in portrait mode, the preview shows correct. If I rotate the phone to landscape, I get the incorrect orientation and stretching.就我而言，当手机处于纵向模式时，预览显示正确。如果我将手机旋转到横向，我会得到不正确的方向和拉伸。
> 
> Were you able to fix it?你能修好它吗？

My application is a landscape-oriented app. Currently, my  is to force the camera screen to portrait orientation.
````tsx

const portraitHoc = (Component: React.ComponentType<any>) => {
  return (props: any) => {
    const [hasLocked, setHasLocked] = React.useState(false);
    const previousOri = React.useRef<OrientationType>();

    React.useEffect(() => {
      let timer: NodeJS.Timeout;
      locker.getOrientation(o => {
        previousOri.current = o;
        locker.lockToPortrait();
        timer = setInterval(() => {
          locker.getOrientation(o => {
            if (o === OrientationType.PORTRAIT) {
              setHasLocked(true);
              clearInterval(timer);
            }
          });
        }, 100);
      });

      return () => {
        clearInterval(timer);
        if (previousOri.current === OrientationType['LANDSCAPE-RIGHT']) {
          locker.lockToLandscapeRight();
        } else if (previousOri.current === OrientationType['LANDSCAPE-LEFT']) {
          locker.lockToLandscapeLeft();
        } else {
          locker.lockToLandscape();
        }
      };
    }, []);

    return hasLocked ? <Component {...props} /> : null;
  };
};

export default portraitHoc(Facedetect);
````

charles474, 2025-02-17:
@m Just wondering if there is any update regarding this open bug?

Shubham0850, 2025-03-04:
I'm also facing the same issue. Any update @mrousavy

arnnis, 2025-03-28:
I'm having the same issue on 4.6.4, Only happens when frame processor is active in landscape mode

eebirke, 2025-04-07:
I think it has something to do with texture and surface views. When vision-camera passes the current orientation to the preview you can see an error in LogCat about rotation not being supported on SurfaceView.PERFORMANCE or something like that. When I'm testing with unlocked app rotation, it seems to behave as expected if I set the `androidPreviewType` prop on the CameraView to "texture-view" (when switching between the two I usually have to do a full reload of the app/component)

The docs describe possible differences between the two here https://react-native-vision-camera.com/docs/api/interfaces/CameraProps#androidpreviewviewtype

> "surface-view": Uses a [SurfaceView](https://developer.android.com/reference/android/view/SurfaceView) for rendering. This is more efficient and supports HDR rendering, but doesn't support masks, transparency, rotations or clipping.
> "texture-view": Uses a [TextureView](https://developer.android.com/reference/android/view/TextureView) for rendering. This is less efficient and doesn't support HDR rendering, but supports masks, transparency, rotations and clipping.

A more pressing issue in my case is that it seems like the Image/ImageProxy itself cannot be rotated by MLKit – haven't tested it yet, but that might also break CodeScanner in some phone orientations with certain sensor orientations 🤔 #3490

charles474, 2025-05-14:
> [@charles474](https://github.com/charles474) did you able to find anything yet ?

Sadly, I haven't.

charles474, 2025-05-14:
> [@charles474](https://github.com/charles474) did you able to find anything yet ?

@Shubham0850 Sadly, I haven't.

mingca, 2025-05-18:
@dalisalvador Were you able to fix this issue?

eduardorfreitas93, 2025-06-26:
https://github.com/mrousavy/react-native-vision-camera/issues/3051#issuecomment-3010647771