margelo/react-native-vision-camera #3260: 🐛 Fails to build on react-native 0.76 on Android

hsjoberg, 2024-10-24:
### What's happening?

Hi, when building using react-native 0.76, react-native-vision-camera fails with the following error:

```
C/C++: VisionCamera: Frame Processors: OFF!
C/C++: CMake Error at CMakeLists.txt:26 (add_library):
C/C++:   Target "VisionCamera" links to target "ReactAndroid::reactnativejni" but
C/C++:   the target was not found.  Perhaps a find_package() call is missing for an
C/C++:   IMPORTED target, or an ALIAS target is missing?
```

### Reproduceable Code

```tsx
N/A.

Check my repro here: https://github.com/hsjoberg/rn76-vision-camera-bug

It's simply a react-community cli project with react-native-vision-camera@4.5.3 added.
```

### Relevant log output

```shell
See above.
```

### Camera Device

```json
.
```

### Device

Emulator

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

OnurGuersoy, 2024-10-24:
Same issue here

benoitvallon, 2024-10-24:
Here is the error I am seeing, I hope this can help

```
Caused by: com.android.ide.common.process.ProcessException:
Error while executing process /YYY/Library/Android/sdk/cmake/3.22.1/bin/cmake with arguments
{
    -H/XXX/node_modules/react-native-vision-camera/android
    -DCMAKE_SYSTEM_NAME=Android
    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
    -DCMAKE_SYSTEM_VERSION=24
    -DANDROID_PLATFORM=android-24
    -DANDROID_ABI=arm64-v8a
    -DCMAKE_ANDROID_ARCH_ABI=arm64-v8a
    -DANDROID_NDK=/YYY/Library/Android/sdk/ndk/26.1.10909125
    -DCMAKE_ANDROID_NDK=/YYY/Library/Android/sdk/ndk/26.1.10909125
    -DCMAKE_TOOLCHAIN_FILE=/YYY/Library/Android/sdk/ndk/26.1.10909125/build/cmake/android.toolchain.cmake
    -DCMAKE_MAKE_PROGRAM=/YYY/Library/Android/sdk/cmake/3.22.1/bin/ninja
    -DCMAKE_CXX_FLAGS=-O2
    -frtti
    -fexceptions
    -Wall
    -Wno-unused-variable
    -fstack-protector-all
    -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/XXX/node_modules/react-native-vision-camera/android/build/intermediates/cxx/RelWithDebInfo/6f6a114v/obj/arm64-v8a
    -DCMAKE_RUNTIME_OUTPUT_DIRECTORY=/XXX/node_modules/react-native-vision-camera/android/build/intermediates/cxx/RelWithDebInfo/6f6a114v/obj/arm64-v8a
    -DCMAKE_BUILD_TYPE=RelWithDebInfo
    -DCMAKE_FIND_ROOT_PATH=/XXX/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/6f6a114v/prefab/arm64-v8a/prefab
    -B/XXX/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/6f6a114v/arm64-v8a
    -GNinja
    -DANDROID_STL=c++_shared
    -DNODE_MODULES_DIR=/XXX/node_modules
    -DENABLE_FRAME_PROCESSORS=OFF
}
```

dlalas, 2024-10-24:
Same issue here!

RicardoGuiliani, 2024-10-24:
Same issue here, after upgrading from react-native 0.75.4 to 0.76

HungGiaPham, 2024-10-25:
Same issue here, after upgrading from react-native 0.75.4 to 0.76

just-myself, 2024-10-25:
> Same issue here, after upgrading from react-native 0.75.4 to 0.76

coder-xiaomo, 2024-10-25:
Same issue here, after upgrading from react-native 0.75.4 to 0.76
using react-native-vision-camera@4.5.3

```bash
> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a] FAILED
C/C++: VisionCamera: Frame Processors: OFF!
C/C++: CMake Error at CMakeLists.txt:26 (add_library):
C/C++:   Target "VisionCamera" links to target "ReactAndroid::reactnativejni" but
C/C++:   the target was not found.  Perhaps a find_package() call is missing for an
C/C++:   IMPORTED target, or an ALIAS target is missing?
C/C++: CMake Generate step failed.  Build files cannot be regenerated correctly.
```

nazacity, 2024-10-25:
Same issue here, after upgrading from react-native 0.75.4 to 0.76

Gautham495, 2024-10-25:
@corbella83 - do you use worklets-core? If yes - please provide a fix if you have one.

Romick2005, 2024-10-25:
```
Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a] FAILED

[CXX1429] error when building with cmake using /Users/romick/repo/newamigos/na-mobile/node_modules/react-native-vision-camera/android/CMakeLists.txt: -- The C compiler identification is Clang 17.0.2

C++ build system [configure] failed while executing:
  VisionCamera: Frame Processors: OFF!
  CMake Error at CMakeLists.txt:26 (add_library):
    Target "VisionCamera" links to target "ReactAndroid::reactnativejni" but
    the target was not found.  Perhaps a find_package() call is missing for an
    IMPORTED target, or an ALIAS target is missing?
 
  CMake Generate step failed.  Build files cannot be regenerated correctly. : com.android.ide.common.process.ProcessException: -- The C compiler identification is Clang 17.0.2
```

Mhp23, 2024-10-25:
To solve the problem go to `YOUR_PROJECT/node_modules/react-native-vision-camera/android/CMakeLists` then replace the following line:

`ReactAndroid::reactnativejni`

By

`ReactAndroid::reactnative`

The build issue will be solved. If you'd like to patch until the next update is available, use [patch package](https://github.com/ds300/patch-package)

Might some of the other popular packages like [react-native-safe-area-context](https://github.com/th3rdwave/react-native-safe-area-context) have similar issue, this solution also could be helpful to fix the Android build issue in RN 0.76.0

Gautham495, 2024-10-28:
@Mhp23 - I can confirm this works and builds.

Thanks!

AjayFrancisTechversant, 2024-10-28:
@Mhp23 Its not working for me.

ajay-ranga, 2024-10-29:
https://github.com/mrousavy/react-native-vision-camera/issues/3260#issuecomment-2438632177

Thanks a lot man.
It worked.
Anyone still getting same issue can use this patch file.
[react-native-vision-camera+4.5.3.patch](https://github.com/user-attachments/files/17557123/react-native-vision-camera%2B4.5.3.patch)

mateusschenato, 2024-10-30:
> [#3260 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/3260#issuecomment-2438632177)
> 
> Thanks a lot man. It worked. Anyone still getting same issue can use this patch file. [react-native-vision-camera+4.5.3.patch](https://github.com/user-attachments/files/17557123/react-native-vision-camera%2B4.5.3.patch)

Thanks !

I also added something on here and now it's working.

`react-native-vision-camera+4.5.3.patch`
```diff --git a/node_modules/react-native-vision-camera/android/CMakeLists.txt b/node_modules/react-native-vision-camera/android/CMakeLists.txt
index 1460dbd..3d4e2ae 100644
--- a/node_modules/react-native-vision-camera/android/CMakeLists.txt
+++ b/node_modules/react-native-vision-camera/android/CMakeLists.txt
@@ -49,9 +49,9 @@ target_include_directories(
         "src/main/cpp"
         "src/main/cpp/frameprocessors"
         "src/main/cpp/frameprocessors/java-bindings"
-        "${NODE_MODULES_DIR}/react-native/ReactCommon"
-        "${NODE_MODULES_DIR}/react-native/ReactCommon/callinvoker"
-        "${NODE_MODULES_DIR}/react-native/ReactAndroid/src/main/jni/react/turbomodule" # <-- CallInvokerHolder JNI wrapper
+        "${REACT_NATIVE_DIR}/ReactCommon"
+        "${REACT_NATIVE_DIR}/ReactCommon/callinvoker"
+        "${REACT_NATIVE_DIR}/ReactAndroid/src/main/jni/react/turbomodule" # <-- CallInvokerHolder JNI wrapper
 )
 
 # Link everything together
@@ -60,7 +60,7 @@ target_link_libraries(
         ${LOG_LIB}                          # <-- Logcat logger
         android                             # <-- Android JNI core
         ReactAndroid::jsi                   # <-- RN: JSI
-        ReactAndroid::reactnativejni        # <-- RN: React Native JNI bindings
+        ReactAndroid::reactnative        # <-- RN: React Native JNI bindings
         fbjni::fbjni                        # <-- fbjni
 )
 ```

closed 2024-10-30 by mrousavy

corbella83, 2024-11-05:
This is not yet solved. Might seem solved because of caches. But if you do a clean build will have this error:
2 files found with path 'lib/arm64-v8a/libreactnative.so' from inputs

3. Add `"**/libreactnative.so",` to the _exclude_ array at build.gradle of the library

TorrCod, 2024-11-05:
@corbella83 same.. the i got the exact same error

Driveemode, 2024-11-08:
> [packagingOptions - excludes]

in which line?

TorrCod, 2024-11-11:
@corbella83  @Driveemode  this was fixed in [4.6.1](https://github.com/mrousavy/react-native-vision-camera/pull/3281)

jobaerhassan, 2024-11-15:
Just deleting the .gradle folder works for me..no need any patch.
device: mac m2 pro
rn version: 0.71

vladvlasov256, 2024-12-04:
As a workaround, turn off the [frame processor](https://react-native-vision-camera.com/docs/guides/frame-processors) if your app doesn't use one.
And use @corbella83's patch.

PR #3263 (2024-10-30): fix: Fix build for RN 0.76 by using new shared prefab

## What

## Changes

## Tested on

## Related issues

- Fixes #3260