margelo/react-native-vision-camera #3261: 🐛 App crashes (iOS 18.1) when user stops recording video. AVFCapture: CameraSession.deactivateAudioSession()

davidfeldt, 2024-10-24:
### What's happening?

When user stops recording a video, app crashes.  The error log from XCode shows the following (see image)
[image]

This happened when we upgraded to version 4.5.2 of the library.  The error seems to be happening in the Swift CameraSession+Audio.swift file 

final func deactivateAudioSession() {
    VisionLogger.log(level: .info, message: "Deactivating Audio Session...")

    audioCaptureSession.stopRunning()
    VisionLogger.log(level: .info, message: "Audio Session deactivated!")
  }

This was working fine before we upgraded to RN 0.75 with react-native-vision-camera v. 4.5.2

### Reproduceable Code

```tsx
import {Camera, getCameraDevice} from 'react-native-vision-camera';
...
const stopRecordingVideo = async () => {
    try {
      setIsRecording(false);
      console.log('stopRecordingVideo');
      timerService.stopTimer();
      await camera?.current?.stopRecording();
      setRemainingSeconds(botbVideoLength);
    } catch (error) {
      console.warn(error);
      Alert.alert('Error', error?.cause?.message);
    }
  };
...

<SafeAreaView style={styles.fullscreen}>
      <Text style={styles.headerText}>{question}</Text>
      <View style={styles.videoRecorder}>
        <Camera
          ref={camera}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          video={true}
          audio={true}
          preset="medium"
        />
      </View>
      {remainingSeconds > 0 ? displayCountdownTime() : null}
      <View style={styles.row}>
        <Icon
          type="ionicon"
          name="videocam"
          size={35}
          color={colors.cherry}
          // disabled={isRecording}
          onPress={startRecordingVideo}
        />

        <Icon
          type="ionicon"
          name="stop-circle-outline"
          size={35}
          color={colors.cherry}
          // disabled={!isRecording}
          onPress={stopRecordingVideo}
        />

...
```

### Relevant log output

```shell
See attached image above showing the error trace.
```

### Camera Device

```json
{
  "position": "front",
  "minFocusDistance": 20,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
  "maxZoom": 128.875,
  "isMultiCam": false,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "minExposure": -8,
  "name": "Front Camera",
  "hasFlash": true,
  "hardwareLevel": "full",
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "hasTorch": false,
  "minZoom": 1,
  "formats": [],
  "sensorOrientation": "portrait"
}
```

### Device

iPhone 14 Pro Max

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

davidfeldt, 2024-10-31:
Found that the error was not due to this library.

closed 2024-10-31 by davidfeldt

kylebaskin, 2025-01-10:
@davidfeldt What library was causing this issue? running into the same thing in my build.

davidfeldt, 2025-01-10:
We had a bug in an unrelated timer service while recording the video.  (We limit the length of the recording)

kylebaskin, 2025-01-10:
I see, thanks for the reply!

For anyone else having this issue, this thread was helpful -- https://github.com/mrousavy/react-native-vision-camera/issues/2132

My permissions flow was buggy and the mic wasn't getting correct permissions before initializing the camera. You could still open the camera and all, but once you try to save audio via stopRecording(), the app will crash.