margelo/react-native-vision-camera #3264: 🐛 Specific QR code is not being read by Android

Barteque, 2024-10-29:
### What's happening?

[image]
This code is not being read by the library but generic android camera app recognises this as a JSON.
Is there an issue with coding?

Upgrading an app requires react-native upgrade so it's bigger deal for me.
Is it fixed in newest library?

### Reproduceable Code

```tsx
const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: (codes) => {
      logger.debug(
        'Codes scanned:',
        codes.map((c: Code) => `${c.type}: ${c.value}`)
      );
      const readValue = codes[0].value;
      if (readValue) {
        logger.info('Code scanned:', readValue);
        if (showReadValue) setQrcode(readValue);
        props.onQRread(readValue);
      } else {
        logger.warn('Scanned empty code');
      }
    }
  });
```

### Relevant log output

```shell
2024-10-29 11:35:03.937 16410-16827 CameraView      com.es.dev  I  invokeOnAverageFpsChanged(0.0)
2024-10-29 11:35:04.439 16410-17326 Vision          com.es.dev  W  Would have logged:
                                                                    # m.aqc@5e70742b
2024-10-29 11:35:04.936 16410-16827 CameraView      com.es.dev  I  invokeOnAverageFpsChanged(0.0)
 ...
2024-10-29 11:35:14.939 16410-16827 CameraView      com.es.dev  I  invokeOnAverageFpsChanged(0.0)
2024-10-29 11:35:15.732 16410-16410 CameraSession   com.es.dev  I  Preview orientation changed! PORTRAIT
2024-10-29 11:35:15.733 16410-16410 CameraView      com.es.dev  I  invokeOnPreviewOrientationChanged(PORTRAIT)
2024-10-29 11:35:15.737 16410-16410 CameraSession   com.es.dev  I  Output orientation changed! PORTRAIT
2024-10-29 11:35:15.738 16410-16410 CameraView      com.es.dev  I  invokeOnOutputOrientationChanged(PORTRAIT)
2024-10-29 11:35:15.940 16410-16827 CameraView      com.es.dev  I  invokeOnAverageFpsChanged(0.0)
2024-10-29 11:35:18.284 16410-16586 ReactNativeJS   com.es.dev  I  [37m2024-10-29 11:35:18 | DEBUG : Codes scanned: 
                                                                        [
                                                                            [
                                                                            "qr: null"
                                                                            ]
                                                                        ]
                                                                        [0m
2024-10-29 11:35:18.286 16410-16586 ReactNativeJS  com.es.dev   I  [33m2024-10-29 11:35:18 | WARN : Scanned empty code [0m

...and many sets of DEBUG : Codes scanned: ..null...  and WARN : Scanned empty code
```

### Camera Device

```json
{
"formats": [],
  "sensorOrientation": "landscape-left",  
"hardwareLevel": "limited",
"maxZoom": 4,
"minZoom": 1,
"maxExposure": 20,
"supportsLowLightBoost": false,  
"neutralZoom": 1,\n
  "physicalDevices": [   "wide-angle-camera"  ],
 "supportsFocus": true,
"supportsRawCapture": false,
"isMultiCam": false,
 "minFocusDistance": 10,
 "minExposure": -20,
 "name": "0 (BACK) androidx.camera.camera2",
 "hasFlash": true,
  "hasTorch": true,
 "position": "back",
  "id": "0"
}'
```

### Device

Samsung A12

### VisionCamera Version

4.3.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

abdelrahman-muntaser, 2024-10-31:
Same issue here

Barteque, 2024-11-08:
we had to change the QR reading library :(

zkteco-home, 2024-11-22:
same issue ,i'v upgraded library to latest,include react native 0.76.3

zkteco-home, 2024-11-22:
> we had to change the QR reading library :(

which library can you use?please recommand

just-myself, 2025-02-18:
Same issue here

just-myself, 2025-02-18:
Important business functions