margelo/react-native-vision-camera #3270: 🔧 2 files found with path 'lib/arm64-v8a/libreactnative.so' from inputs:

torrrrzz, 2024-10-31:
### How were you trying to build the app?

Using `yarn android`

### Full build logs

```tsx
* What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.MergeNativeLibsTask$MergeNativeLibsTaskWorkAction
   > 2 files found with path 'lib/arm64-v8a/libreactnative.so' from inputs:
      - /Users/torrrrzz/Documents/blesspayments/mobileapp/node_modules/react-native-vision-camera/android/build/intermediates/library_jni/debug/copyDebugJniLibsProjectOnly/jni/arm64-v8a/libreactnative.so
      - /Users/torrrrzz/.gradle/caches/8.10.2/transforms/a86708618e37ae0e0a5c88cb1b2db70c/transformed/react-android-0.76.1-debug/jni/arm64-v8a/libreactnative.so
     If you are using jniLibs and CMake IMPORTED targets, see
     https://developer.android.com/r/tools/jniLibs-vs-imported-targets
```

### Project dependencies

```json
"dependencies": {
   "react": "^18.3.1",
   "react-native": "^0.76.1",
   "react-native-vision-camera": "^4.6.0"
  },
```

### VisionCamera Version

4.6.0

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-10-31 by maintenance-hans[bot]

mrousavy, 2024-11-05:
Heyo - thanks for reporting! I investigated this and just created a fix for this here: https://github.com/mrousavy/react-native-vision-camera/pull/3281 :)

mrousavy, 2024-11-05:
I'd appreciate a GitHub sponsor donation if you feel like this package and fix has helped you. :)