margelo/react-native-vision-camera #3274: 🔧 js files after running npm install

atheck, 2024-11-04:
### How were you trying to build the app?

After updating to react-native-vision-camera 4.6.0 when installing dependencies with npm, all Typescript files are compiled to js files.

I think this is because of the change in package/package.json. Now there is a postinstall script running tsc.

### Full build logs

```tsx
no logs
```

### Project dependencies

```json
"react-native-vision-camera": 4.6.0,
"npm": 10.9.0
```

### VisionCamera Version

4.6.0

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-11-04:
@hannojg / @chrispader this happened after https://github.com/mrousavy/react-native-vision-camera/pull/3249. Can y'all look into this asap? If not I'd have to revert all of this.

hannojg, 2024-11-04:
@atheck 

- Which node version are you using
- Which npm version are you using
- Can you send me a screenshot of the problem? i don't fully understand what you mean by " all Typescript files are compiled"

In general, the postinstall script isn't executed when you install a dependency with npm

hannojg, 2024-11-04:
@mrousavy if we want to be safe here is a PR that removes the post install hook (i would recommend to remove it since it seems it was unnecessary in the first place):

- https://github.com/mrousavy/react-native-vision-camera/pull/3275

atheck, 2024-11-04:
Hi @hannojg 

I'm using node 20.15.0 with npm 10.9.0.

When I run `npm install` all typescript files in my project are compiled into javascript. So I have hundreds of new javascript files in my repo.

[image]

closed 2024-11-04 by mrousavy

atheck, 2024-11-05:
Thank you very much for fixing this so fast.