margelo/react-native-vision-camera #3277: 🐛 Frame Processor Error: offscreenTextures.value.push is not a function (it is undefined)

tomerh2001, 2024-11-04:
### What's happening?

I am trying to open the camera, but I am getting this error:
```
 (NOBRIDGE) ERROR  [Frame Processor Error: offscreenTextures.value.push is not a function (it is undefined)]
```

### Reproduceable Code

```tsx
const {hasPermission, requestPermission} = useCameraPermission();
const device = useCameraDevice('back');
const format = useCameraFormat(device, Templates.FrameProcessing);

	const frameProcessor = useSkiaFrameProcessor(frame => {
		'worklet';
	},
	[
	]);

return <Camera
				ref={camera}
				photo
				isActive={isCameraActive}
				style={styles.camera}
				device={device}
				pixelFormat='yuv'
				frameProcessor={frameProcessor}
				format={format}
				resizeMode='contain'
				outputOrientation='landscape-right'
				{...properties}
				onLayout={event => setCameraLayout(event.nativeEvent.layout)}
				onInitialized={() => setIsCameraInitialized(true)}
			/>
```

### Relevant log output

```shell
(NOBRIDGE) ERROR  [Frame Processor Error: offscreenTextures.value.push is not a function (it is undefined)]
 (NOBRIDGE) WARN  ImmutableStateInvariantMiddleware took 503ms, which is more than the warning threshold of 32ms. 
If your state or actions are very large, you may want to disable the middleware as it might cause too much of a slowdown in development mode. See https://redux-toolkit.js.org/api/getDefaultMiddleware for instructions.
It is disabled in production builds, so you don't need to worry about that.
 (NOBRIDGE) ERROR  [Frame Processor Error: offscreenTextures.value.push is not a function (it is undefined)]
```

### Camera Device

```json
(NOBRIDGE) LOG  {
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 30,
  "minZoom": 0.5563247799873352,
  "maxExposure": 24,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "wide-angle-camera",
    "ultra-wide-angle-camera",
    "telephoto-camera",
    "telephoto-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": true,
  "minFocusDistance": 10.500000100135805,
  "minExposure": -24,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 7 Pro

### VisionCamera Version

4.6.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-11-04 by maintenance-hans[bot]

ChagitCharazi, 2025-02-03:
Same issue.

jslok, 2025-04-03:
Had the same error. Deleted and reinstalled node_modules with latest versions of vision camera and skia and the error was gone.