margelo/react-native-vision-camera #3278: 🔧 Execution failed for task ':app:mergeDebugNativeLibs'.

SulthanNK, 2024-11-04:
### How were you trying to build the app?

I usually go with commands only

npm install
npx react-native start

after that, I press "a" to run on android.

I built this project 2 months ago with the same dependencies i mentioned below and it's works fine without any issues. Today, I started the project again after 2 months to add a new feature and ended up with this build issue.

Also saw someone posted this same issue (#3235) and you mentioned a machine issue, So, I tried with 2 different machines and ended up same issue.

### Full build logs

```tsx
> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ??
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at E:\Projects\ReactNative\App\common-branches\my_app\node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core not found, Frame Processors are disabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to true!

> Task :app:generateCodegenSchemaFromJavaScript SKIPPED
> Task :app:generateCodegenArtifactsFromSchema SKIPPED
> Task :app:generatePackageList
> Task :app:preBuild
> Task :app:preDebugBuild
> Task :app:generateDebugBuildConfig
> Task :react-native-async-storage_async-storage:preBuild UP-TO-DATE
> Task :react-native-async-storage_async-storage:preDebugBuild UP-TO-DATE
> Task :react-native-ble-manager:preBuild UP-TO-DATE
> Task :react-native-ble-manager:preDebugBuild UP-TO-DATE
> Task :react-native-community_netinfo:preBuild UP-TO-DATE
> Task :react-native-community_netinfo:preDebugBuild UP-TO-DATE
> Task :react-native-device-info:preBuild UP-TO-DATE
> Task :react-native-device-info:preDebugBuild UP-TO-DATE
> Task :react-native-geolocation-service:preBuild UP-TO-DATE
> Task :react-native-geolocation-service:preDebugBuild UP-TO-DATE
> Task :react-native-ble-manager:writeDebugAarMetadata
> Task :react-native-community_netinfo:writeDebugAarMetadata
> Task :react-native-async-storage_async-storage:writeDebugAarMetadata
> Task :react-native-device-info:writeDebugAarMetadata
> Task :react-native-gesture-handler:preBuild UP-TO-DATE
> Task :react-native-gesture-handler:preDebugBuild UP-TO-DATE
> Task :react-native-geolocation-service:writeDebugAarMetadata
> Task :react-native-linear-gradient:preBuild UP-TO-DATE
> Task :react-native-linear-gradient:preDebugBuild UP-TO-DATE
> Task :react-native-gesture-handler:writeDebugAarMetadata
> Task :react-native-localize:preBuild UP-TO-DATE
> Task :react-native-localize:preDebugBuild UP-TO-DATE
> Task :react-native-linear-gradient:writeDebugAarMetadata
> Task :react-native-permissions:preBuild UP-TO-DATE
> Task :react-native-permissions:preDebugBuild UP-TO-DATE
> Task :react-native-localize:writeDebugAarMetadata
> Task :react-native-safe-area-context:preBuild UP-TO-DATE
> Task :react-native-safe-area-context:preDebugBuild UP-TO-DATE
> Task :react-native-permissions:writeDebugAarMetadata
> Task :react-native-screens:preBuild UP-TO-DATE
> Task :react-native-screens:preDebugBuild UP-TO-DATE
> Task :react-native-safe-area-context:writeDebugAarMetadata
> Task :react-native-screens:writeDebugAarMetadata
> Task :react-native-vision-camera:prepareHeaders
> Task :react-native-vision-camera:preBuild
> Task :react-native-vision-camera:preDebugBuild
> Task :react-native-wifi-reborn:preBuild UP-TO-DATE
> Task :react-native-wifi-reborn:preDebugBuild UP-TO-DATE
> Task :react-native-vision-camera:writeDebugAarMetadata
> Task :react-native-wifi-reborn:writeDebugAarMetadata
[... 558 lines cut ...]
```

### VisionCamera Version

4.0.1

### Target platforms

Android

### Operating system

MacOS, Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-11-05:
Heyo - thanks for reporting! I investigated this and just created a fix for this here: https://github.com/mrousavy/react-native-vision-camera/pull/3281 :)

mrousavy, 2024-11-05:
I'd appreciate a GitHub sponsor donation if you feel like this package and fix has helped you. :)

closed 2024-11-05 by mrousavy