margelo/react-native-vision-camera #3279: 🐛 ReactNative | V0.76.0 react-native-vision-camera not getting autlinked

sandeep36butte, 2024-11-05:
### What's happening?

I have upgraded the RN version to latest(0.76.0) to enable the RN NewArchitecture.
The React-native-vision-camera not package not seems to be autolinked.Getting getting error in the getcameradevices hooks.
[image]
[image]

### Reproduceable Code

```tsx
GitHub Repo:https://github.com/sandeep36butte/RNBrownFeldApp

This is the brownfield Application.Just clone and run the app with yarn start and run on android.
```

### Relevant log output

```shell
ERROR  Warning: TypeError: Cannot read property 'CameraDevices' of undefined

This error is located at:
    in App
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
    in ReactNativeBrownFieldApp(RootComponent), js engine: hermes
 ERROR  Warning: TypeError: Cannot read property 'CameraDevices' of undefined

This error is located at:
    in App
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
    in ReactNativeBrownFieldApp(RootComponent), js engine: hermes
```

### Camera Device

```json
Null
```

### Device

pixel 8

### VisionCamera Version

4.6.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-11-05:
Heyo - thanks for reporting! I investigated this and just created a fix for this here: https://github.com/mrousavy/react-native-vision-camera/pull/3281 :)

mrousavy, 2024-11-05:
I'd appreciate a GitHub sponsor donation if you feel like this package and fix has helped you. :)

closed 2024-11-05 by mrousavy

sandeep36butte, 2024-11-06:
@mrousavy 
This is working fine now.The issue is with the project directory structure.
**autolinkLibrariesFromCommand** needs to know where to run the config command from. In this case. By default would be .. but in your case, your package.json is in ../ReactNative.

Can refer here for solution
https://github.com/facebook/react-native/issues/47406