margelo/react-native-vision-camera #3280: ✨ Make code scanner resolution configurable using `format.videoResolution` on Android

asadwan, 2024-11-05:
### What feature or enhancement are you suggesting?

Code scanner on Android has a resolution of 1280x720 and unlike iOS, is [not configurable](https://github.com/mrousavy/react-native-vision-camera/issues/1970#issuecomment-1776789866) with `format.videoResolution`. Due to the low resolution, we are encountering issues scanning long barcodes. It would be great if scanner resolution could be configurable on Android like iOS. 

To workaround the issue are using a fork with higher resolution hardcoded https://github.com/dashdoc/react-native-vision-camera

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

To workaround the issue are using a fork with higher resolution hardcoded https://github.com/dashdoc/react-native-vision-camera

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-11-05:
Hey thanks for creating this feature request, but sorry but I don't think this is something I will be working on. I don't see a lot of value in this because you can simply create a Frame Processor which does exactly that - it uses `videoResolution` and you can do even more advanced stuff inside of it. 

I don't want to overcomplicate the QR Code Scanner because as you can see a lot of people simply want it to do "everything", and this is not what VisionCamera is about. You have so much flexibility to do this yourself in a Frame Processor and I think it should stay that way to avoid allocating all my time on a QR code scanner.

closed 2024-11-05 by mrousavy

asadwan, 2024-11-05:
> Hey thanks for creating this feature request, but sorry but I don't think this is something I will be working on. I don't see a lot of value in this because you can simply create a Frame Processor [...]

Hi @mrousavy I'd like first to thank you for your amazing work and for your reply.  Would you be open to accepting PR for it? The way I see it is that today it works for iOS but not Android and it should be consistent across the board.

mrousavy, 2024-11-06:
> Would you be open to accepting a PR for it?

Thanks for the offer, but no, for the following reasons;

1. There is no way to configure the Code Scanner resolution alone on iOS. It always chooses it's own resolution. So this is an Android-only feature, which I want to avoid.
2. Adding features always means that I have to maintain those features in the future. I know it is not that difficult to maintain a "simple" resolution, but it just adds to the Code Scanner - **and the Code Scanner is a feature I built only for the community** because I don't use it myself, so I can't really continue it's development or fix bugs in it.

mrousavy, 2024-11-06:
I'm really making sure the codebase in VisionCamera is as lean and simple and maintainable as possible to avoid VisionCamera from becoming a dead repo. A lot of repos just go dead because too much messy code lands in it.

mrousavy, 2024-11-06:
No, iOS does not let you choose a resolution for the Code Scanner afaik.

Both iOS and Android however allow you to choose a resolution for the Frame Processor.

asadwan, 2024-11-06:
> > 1. There is no way to configure the Code Scanner resolution alone on iOS. It always chooses it's own resolution. So this is an Android-only feature, which I want to avoid.
> 
> It's possible that I misunderstood but from your comments here https://github.com/mrousavy/react-native-vision-camera/issues I got the impression that it already works on iOS but not Android

I mislinked the issue in question https://github.com/mrousavy/react-native-vision-camera/issues/1970#issuecomment-1776789866

asadwan, 2024-11-06:
> No, iOS does not let you choose a resolution for the Code Scanner afaik.
> 
> Both iOS and Android however allow you to choose a resolution for the Frame Processor.

And would you be open to increasing the barcode scanner's hardcoded resolution?  1280x720 seems pretty low

mrousavy, 2024-11-06:
1280x720 is the recommended default by Android/CameraX. Anything larger is hard to keep up for slower phones and will likely increase latency.