margelo/react-native-vision-camera #3283: 🐛 Black screen when opening the camera using CodeScanner.

anderpaz, 2024-11-07:
### What's happening?

At times, when the device turns on with low battery, for example, 5-10%, I only get a black screen.
In my error log, I get MlKitContext.

### Reproduceable Code

```tsx
const CameraManager: React.FC<typ.CameraManagerProps> = ({
  photo,
  codeTypes,
  hiddenButtonClose,
  hiddenCardButton,
  isActive,
  isPreviewPhoto,
  isFull,
  isMaskCode,
  onClose,
  onReturnTakePhoto,
  onBarCodeScanned,
  onSavePhoto,
}) => {
  const cameraRef = React.useRef<Camera>(null);

  const [isFlash, setFlash] = React.useState(false);
  const [isLoading, setLoading] = React.useState(false);
  const [isFront, setFront] = React.useState(false);
  const [photoFile, setPhotoFile] = React.useState<typ.photoFile>();

  const { updateStatusBar } = useAppActions();
  const device = useCameraDevice(!isFront ? 'back' : 'front');

  React.useEffect(() => {
    console.log(JSON.stringify(device, (k, v) => (k === 'formats' ? [] : v), 2));
  }, [device, isPreviewPhoto, updateStatusBar]);

  React.useEffect(() => {
    updateStatusBar({
      active: true,
      screen: 'camera',
    });
    return () => {
      updateStatusBar({
        active: false,
        screen: 'camera',
      });
    };
  }, [isPreviewPhoto, updateStatusBar]);

  const onCodeScanned = React.useCallback(
    (codes: typ.code[]) => {
      try {
        console.log('-----------------', codes);
        if (!onBarCodeScanned) return;
        onBarCodeScanned(codes);
      } catch (e) {
        log('CameraManager').error('e from onCodeScanned', e);
      }
    },
    [onBarCodeScanned],
  );

  const codeScanner = useCodeScanner({
    codeTypes: codeTypes || [],
    onCodeScanned,
  });

  const styled = React.useMemo<typ.styled>(() => {
    if (isFull)
      return {
[... 233 lines cut ...]
                                                                                                      "hasTorch": true,
                                                                                                      "position": "back",
                                                                                                      "id": "0"
                                                                                                    }
```

### Device

Samsung A15

### VisionCamera Version

"react-native-vision-camera": "4.3.1",

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-11-07 by maintenance-hans[bot]

iliapnmrv, 2024-11-08:
hey! I think i had the same issue. Can you try this code?

```ts
const isFocused = useIsFocused();

// https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/src/hooks/useIsForeground.ts
  
const isForeground = useIsForeground();
const isActive = isFocused && isForeground;

<Camera
  isActive={isActive}
/>
```

anderpaz, 2024-11-08:
Normally, the camera works, but if the battery is low and I restart the device and try to scan a QR code, at some point after the 3rd or 5th restart, the screen goes black and I get this error in the log.

`Creating CodeScanner output...
2024-11-08 10:28:24.911  3096-3096  CameraSession           com.spacecom.mpmobile                E  Failed to configure CameraSession! Error: MlKitContext has not been initialized, Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true, orientationChanged=true, locationChanged=true)
                                                                                                    java.lang.IllegalStateException: MlKitContext has not been initialized
                                                                                                    	at com.google.android.gms.common.internal.Preconditions.checkState(com.google.android.gms:play-services-basement@@18.3.0:2)
                                                                                                    	at com.google.mlkit.common.sdkinternal.MlKitContext.getInstance(com.google.mlkit:common@@18.9.0:1)
                                                                                                    	at com.google.mlkit.vision.barcode.BarcodeScanning.getClient(com.google.android.gms:play-services-mlkit-barcode-scanning@@18.3.0:3)
                                                                                                    	at com.mrousavy.camera.core.CodeScannerPipeline.<init>(CodeScannerPipeline.kt:27)
                                                                                                    	at com.mrousavy.camera.core.CameraSession_ConfigurationKt.configureOutputs(CameraSession+Configuration.kt:219)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:135)
                                                                                                    	at com.mrousavy.camera.react.CameraView$update$1.invokeSuspend(CameraView.kt:153)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:958)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:230)
                                                                                                    	at android.os.Looper.loop(Looper.java:319)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8913)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:608)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
2024-11-08 10:28:24.911  3096-3096  CameraView              com.spacecom.mpmobile                E  invokeOnError(...):
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  java.lang.IllegalStateException: MlKitContext has not been initialized
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.google.android.gms.common.internal.Preconditions.checkState(com.google.android.gms:play-services-basement@@18.3.0:2)
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.google.mlkit.common.sdkinternal.MlKitContext.getInstance(com.google.mlkit:common@@18.9.0:1)
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.google.mlkit.vision.barcode.BarcodeScanning.getClient(com.google.android.gms:play-services-mlkit-barcode-scanning@@18.3.0:3)
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.mrousavy.camera.core.CodeScannerPipeline.<init>(CodeScannerPipeline.kt:27)
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.mrousavy.camera.core.CameraSession_ConfigurationKt.configureOutputs(CameraSession+Configuration.kt:219)
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:135)
2024-11-08 10:28:24.911  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.mrousavy.camera.react.CameraView$update$1.invokeSuspend(CameraView.kt:153)
[... cut ...]
2024-11-08 10:28:24.912  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:608)
2024-11-08 10:28:24.912  3096-3096  System.err              com.spacecom.mpmobile                W  	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
2024-11-08 10:28:24.917  3096-3096  Dialog                  com.spacecom.mpmobile                I  mIsDeviceDefault = false, mIsSamsungBasicInteraction = false, isMetaDataInActivity = false`

> hey! I think i had the same issue. Can you try this code?
>
> ```ts
> const isFocused = useIsFocused();
>
> // [...]

anderpaz, 2024-11-08:
> Guten Tag, Hans here. Unfortunately, it seems ze issue is not reproducible because you did not try it in ze VisionCamera Example app. Plus, you did not provide logs relevant to isolating ze problem, [...]

I can't test with the example, but the error is very specific to react-native-vision-camera.

EricChou517, 2024-12-12:
same issue on SM-A23 , react-native 0.76.4

anderpaz, 2024-12-16:
> same issue on SM-A23 , react-native 0.76.4

Are you still having problems?
Is your error MlKitContext has not been initialized?
And are you using the vision-camera-code-scanner lib?