margelo/react-native-vision-camera #3289: 🐛 Error when using "frameProcessor" on Samsung Galaxy S8: "Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera"

AmauryLiet, 2024-11-12:
### What's happening?

Impossible to provide a `frameProcessor` on specific device(s)

Trying to do so results in app not responding and looping over same error endlessly.

Same as closed issue https://github.com/mrousavy/react-native-vision-camera/issues/3000, but with a device at hand to provide logs

### Reproduceable Code

```tsx
import React, { FunctionComponent } from 'react';
import { Camera, useCameraDevice, useSkiaFrameProcessor } from 'react-native-vision-camera';

export const MinimalReproduceableCode: FunctionComponent = () => {
  const device = useCameraDevice('front');
  console.log(JSON.stringify(device, (k, v) => (k === 'formats' ? [] : v), 2));

  const frameProcessor = useSkiaFrameProcessor(frame => {
    'worklet';
    frame.render();
  }, []);

  if (device == null) {
    return;
  }

  return (
    <Camera device={device} isActive photo style={{ flex: 1 }} frameProcessor={frameProcessor} />
  );
};
```

### Relevant log output

```shell
2024-11-12 15:46:23.250 29893-29893 BufferQueueProducer                         E  [SurfaceTexture-0-29893-1] disconnect: not connected (req=1)
2024-11-12 15:46:23.250 29893-29893 libEGL                                      W  EGLNativeWindowType 0x7340377010 disconnect failed
2024-11-12 15:46:23.250 29893-29893 mali_winsys                                 D  EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2024-11-12 15:46:23.251 29893-29893 GLConsumer                                  E  [SurfaceTexture-0-29893-1] updateAndRelease: GLConsumer is not attached to an OpenGL ES context
2024-11-12 15:46:23.251 29893-29893 RNSkia                                      I  updateAndRelease() failed. The exception above can safely be ignored
2024-11-12 15:46:23.251 29893-30089 CaptureSession                              D  CameraCaptureSession.onConfigured() mState=OPENED
2024-11-12 15:46:23.252 29893-30089 CaptureSession                              D  CameraCaptureSession.onReady() OPENED
2024-11-12 15:46:23.267 29893-29893 ViewRootIm...nActivity]                     D  Relayout returned: old=[0,0][720,1480] new=[0,0][720,1480] result=0x1 surface={valid=true 496744218624} changed=false
2024-11-12 15:46:23.287 29893-29893 VisionCameraProxy                           D  Finding view 1309...
2024-11-12 15:46:23.287 29893-29893 VisionCameraProxy                           D  Found view 1309!
2024-11-12 15:46:23.344 29893-29893 GLConsumer                                  E  [SurfaceTexture-0-29893-1] checkAndUpdateEglState: invalid current EGLContext
2024-11-12 15:46:23.345 29893-29893 RNSkia                                      I  updateAndRelease() failed. The exception above can safely be ignored
2024-11-12 15:46:23.672 29893-30090 VideoCapture                                D  cameraCaptureResult timestampNs = 82747610709814, current system uptimeMs = 82747691, current system realtimeMs = 82747691
2024-11-12 15:46:23.696 29893-30076 skia                                        D  Could not create EGL image, err = (0x3003)
2024-11-12 15:46:23.697 29893-30076 RNSkia                                      I  Failed to convert HardwareBuffer to OpenGL Texture!
2024-11-12 15:46:23.719 29893-30076 skia                                        D  Could not create EGL image, err = (0x3003)
2024-11-12 15:46:23.719 29893-30076 RNSkia                                      I  Failed to convert HardwareBuffer to OpenGL Texture!
2024-11-12 15:46:23.728 29893-31920 ReactNativeJS                               E  Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera
2024-11-12 15:46:23.740 29893-31920 ReactNativeJS                               E  Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera
2024-11-12 15:46:23.748 29893-30076 skia                                        D  Could not create EGL image, err = (0x3003)
2024-11-12 15:46:23.748 29893-30076 RNSkia                                      I  Failed to convert HardwareBuffer to OpenGL Texture!
2024-11-12 15:46:23.753 29893-31920 ReactNativeJS                               E  Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera
2024-11-12 15:46:23.766 29893-30076 skia                                        D  Could not create EGL image, err = (0x3003)
2024-11-12 15:46:23.766 29893-30076 RNSkia                                      I  Failed to convert HardwareBuffer to OpenGL Texture!
2024-11-12 15:46:23.771 29893-31920 ReactNativeJS                               E  Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera
2024-11-12 15:46:23.805 29893-30076 skia                                        D  Could not create EGL image, err = (0x3003)
2024-11-12 15:46:23.805 29893-30076 RNSkia                                      I  Failed to convert HardwareBuffer to OpenGL Texture!
2024-11-12 15:46:23.812 29893-31920 ReactNativeJS                               E  Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera
2024-11-12 15:46:23.832 29893-30076 skia                                        D  Could not create EGL image, err = (0x3003)
2024-11-12 15:46:23.832 29893-30076 RNSkia                                      I  Failed to convert HardwareBuffer to OpenGL Texture!
2024-11-12 15:46:23.872 29893-30076 skia                                        D  Could not create EGL image, err = (0x3003)
2024-11-12 15:46:23.872 29893-30076 RNSkia                                      I  Failed to convert HardwareBuffer to OpenGL Texture!
2024-11-12 15:46:23.881 29893-31920 ReactNativeJS                               E  Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera
2024-11-12 15:46:23.904 29893-31920 ReactNativeJS                               E  Frame Processor Error: Exception in HostFunction: Failed to convert NativeBuffer to SkImage!, js engine: VisionCamera
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "limited",
  "maxZoom": 4,
  "minZoom": 1,
  "maxExposure": 4,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -4,
  "name": "1 (FRONT) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung Galxy S8 (SM-G950F)

### VisionCamera Version

4.6.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-11-12 by maintenance-hans[bot]

AmauryLiet, 2024-11-12:
@mrousavy I think the bot is a bit aggresive here :)

Happy to provide any additional information if needed

AmauryLiet, 2024-11-12:
Reproduction steps:
- start a projet with `Camera` component
- add a minimalist `frameProcessor` with `useSkiaFrameProcessor`
- run on Galaxy S8
- app crashes

mrousavy, 2024-11-13:
Hm, I agree this issue should stay open but I have no idea how to fix it right now, this involves a lot of debugging.

You can try to change `pixelFormat` from yuv to rgb and vice versa, but other than that I don't really have any ideas right now. It might also be a Skia bug.

TonchoLozev, 2025-02-09:
I have the same problem using useSkiaFrameProcessor on my Samsung Galaxy A55, but not on my Samsung Galaxy Tab A9+.
I am not sure what is happening underneath, but somehow it is device related obviously.
And changing pixelFormat does not help.

AmauryLiet, 2025-11-06:
@chrishiguto sorry but no, I ended up not using `useSkiaFrameProcessor` eventually (I implemented my feature in another way)