margelo/react-native-vision-camera #3297: 🐛 Issue: Camera Preview Showing Blank Screen When Using useSkiaFrameProcessor() in react-native-vision-camera

uasghar-smile, 2024-11-15:
### What's happening?

**Description:**
I'm encountering an issue where the camera preview is completely blank when using the useSkiaFrameProcessor() hook in react-native-vision-camera on Android. The useFrameProcessor() hook works as expected, displaying the camera feed correctly, but switching to useSkiaFrameProcessor() results in a blank screen.

**Environment:**
react-native: 0.73.0
react-native-vision-camera: 4.5.2
Platform: Android [Samsung Galaxy A05s]
react-native-reanimated: 3.15.4
Android version : 14
react-native-worklets-core: 1.3.3

**Expected Behavior:** The camera preview should show the real-time camera feed without any issues.

**Actual Behavior:** The camera preview shows a completely blank screen.

### Reproduceable Code

```tsx
const device = useCameraDevice("back");

{device != null && hasPermission && (
        <>
          <Camera
            torch={torchEnabled ? "on" : "off"}
            style={StyleSheet.absoluteFill}
            ref={camera}
            isActive={isActive}
            device={device}
            photo={true}
            format={cameraFormat}
            frameProcessor={frameProcessor}
            pixelFormat="yuv"
          />
</>
)}

const frameProcessor = useSkiaFrameProcessor((frame) => {
    "worklet";
    frame.render();
    if (takenShared.value === false && frameCount.value % 5 == 0) {
      // console.log("detect frame");
      // console.log(frame.width);
      // console.log(frame.height);
      runAtTargetFps(3, () => {
        "worklet";
        try {
          const results = DDN.detect(frame);
          console.log(results);
          frameWidth.value = frame.width;
          frameHeight.value = frame.height;
          convertAndSetResultsJS(results);
        } catch (error) {
          console.log(error);
        }
      });
    }
    frameCount.value++;
  }, []);
```

### Relevant log output

```shell
adb logcat:
11-15 13:37:02.612 18770 20017 E ReactNativeJS: { [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]
11-15 13:37:02.612 18770 20017 E ReactNativeJS:   _message: 'An unknown error occurred while creating the Camera Session, but the Camera can recover from it.',
11-15 13:37:04.437 29755 29755 E CamX    : [ERROR][SENSOR ] camxeepromdata.cpp:3642 FormatDualCameraData_HI5022() hi5022 dualcamera checksum failed: invalid data.
11-15 13:37:16.664 29755 29755 E CamX    : [ERROR][HAL    ] camxhal3.cpp:800 get_number_of_cameras() Camera-VTS  prop_vts1 user, prop_vts2: locked
[... 201 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy A05s

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

uasghar-smile, 2024-11-15:
**Hi @mrousavy,  I'm experiencing an issue where using useSkiaFrameProcessor() results in a blank camera screen, but useFrameProcessor() works as expected.

Could you please help investigate this or guide on potential fixes? Thank you for your support!**

KienDuong20, 2025-01-16:
@uasghar-smile i also have this problem on galaxy a03 device, when using useSkiaFrameProcessor, but i try some other android devices, it has no problem. have you solved this problem?