margelo/react-native-vision-camera #3303: 🔧 Build failed with 4.6.3

zkteco-home, 2024-11-26:
### How were you trying to build the app?

For Android ,i try to use yarn start,today i'v upgrade to 4.6.3,but when i run yarn start ,i got the following error

### Full build logs

```tsx
> Task :react-native-vision-camera:compileDebugKotlin FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.3/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.
523 actionable tasks: 31 executed, 492 up-to-date

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor

e: Incompatible classes were found in dependencies. Remove them from the classpath or use '-Xskip-metadata-version-check' to suppress errors
e: file:///D:/MinervaIOT/Brazil/mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:8:27 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/Admin/.gradle/caches/transforms-3/88ec9f935ee2ac1040568b164c8f97e5/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/MinervaIOT/Brazil/mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:22:38 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/Admin/.gradle/caches/transforms-3/88ec9f935ee2ac1040568b164c8f97e5/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/MinervaIOT/Brazil/mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:29:7 Class 'kotlinx.coroutines.CoroutineDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/Admin/.gradle/caches/transforms-3/88ec9f935ee2ac1040568b164c8f97e5/transformed/jetified-kotlinx-coroutines-core-jvm-1.9.0.jar!/kotlinx/coroutines/CoroutineDispatcher.class
e: file:///D:/MinervaIOT/Brazil/mobile/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraQueues.kt:29:37 Class 'kotlinx.coroutines.android.HandlerDispatcher' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 2.0.0, expected version is 1.8.0.
The class is loaded from C:/Users/Admin/.gradle/caches/transforms-3/6be3b66eeaa4e29862bd8fd4d006a3a3/transformed/jetified-kotlinx-coroutines-android-1.9.0.jar!/kotlinx/coroutines/android/HandlerDispatcher.class
```

### Project dependencies

```json
"react-native": "0.73.9",

"react-native-vision-camera": "^4.6.3"
```

### VisionCamera Version

4.6.3

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-11-26 by maintenance-hans[bot]

shubham-s3, 2024-12-04:
I am also facing the same issue, any help

litinskii, 2024-12-11:
+

zkteco-home, 2024-12-25:
4.6.3 not support react-native 0.73

D4uS1, 2025-01-22:
If 4.6.3 does not support 0.73, please correct this in your package.json file at https://github.com/mrousavy/react-native-vision-camera/blob/54df86ace1716039f690e865c97f29f87dfe43b7/package/package.json#L94 .

If someone just installs the package using npm, it will resolve to the version 4.6.3.

D4uS1, 2025-01-22:
I am also using react-native 0.73 in one project and if i just install the package using npm install it will install version 4.6.3.
I think npm is not resolving the version correct because of this line

https://github.com/mrousavy/react-native-vision-camera/blob/54df86ace1716039f690e865c97f29f87dfe43b7/package/package.json#L110

eisodev, 2025-01-27:
> I am also using react-native 0.73 in one project and if i just install the package using npm install, it will install version 4.6.3. I think npm is not resolving the version correct because of this line:
> 
> [react-native-vision-camera/package/package.json](https://github.com/mrousavy/react-native-vision-camera/blob/54df86ace1716039f690e865c97f29f87dfe43b7/package/package.json#L110)
> 
> Line 110 in [54df86a](/mrousavy/react-native-vision-camera/commit/54df86ace1716039f690e865c97f29f87dfe43b7)
> 
>  "react-native": "*", 
> The peer dependency seems to be "any version" in your package.json, hence it will just install the newest version of the package, even if they are not compatible.
> 
> Do you have any list of compatible versions?

We are on RN 0.73.7 and need to update the package because the scanner stopped scanning barcodes on Android 14.

Some version minor/patch of this package changed gradle requirements for kotlin version. According to this link Gradle 8.3 (which we use currently on vision camera 4.0.5) [installs kotlin 1.9.0](https://docs.gradle.org/8.3/release-notes.html#:~:text=%5D%20%2D%20Upgrade%20embedded-,Kotlin%20to%201.9.0,-%5B%2325492%5D%20%2D%20Enhance) .

This version is not compatible with some version of vision camera, but its hard to find out by the release notes.

Latest seems to use gradle 8.6. Will try that and come back.
https://github.com/mrousavy/react-native-vision-camera/blob/main/package/android/gradle/wrapper/gradle-wrapper.properties