margelo/react-native-vision-camera #3312: 🐛 [Android] Aspect ratio mismatch between camera format and barcode scanner frame

daankennes, 2024-11-29:
### What's happening?

Hi, thank you for this great library!

I'm encountering an issue on Android when setting a specific camera format with a defined aspect ratio. I expected the aspect ratio of the BarcodeScannerFrame to match the aspect ratio of the camera format, but this does not appear to be the case on Android.

**Primary and secondary problem:**

1) Aspect ratio mismatch:
When setting the camera format to a 16:9 aspect ratio on Android, the BarcodeScannerFrame I receive has a 4:3 aspect ratio instead. This mismatch results in distorted x-coordinates when mapping the barcode positions to the screen because the coordinates look "squeezed." On iOS, the aspect ratio of the BarcodeScannerFrame respects the aspect ratio set in the camera format, and everything works as expected.

2) Width scale factor:
I have attempted to calculate a width scale factor to map the barcode x coordinates correctly, but I'm not sure if it's the correct way (see the widthScaleFactor const in the code). I do notice however that the projected coordinates become square again (when scanning a barcode straight from above), so it seems to be kind of right.

**Current workaround:**
As a temporary solution, I am setting the camera format to 4:3 to match the aspect ratio of the BarcodeScannerFrame I get on Android. However, this feels unreliable, as I am not sure if I should expect the BarcodeScannerFrame to always be of that aspect ratio.

**Summary:**
My question boils down to two things. Could the library ensure that the BarcodeScannerFrame's aspect ratio always matches the camera format's aspect ratio as specified? If not, could there be a mechanism to determine the correct scaling factor between the camera format and the BarcodeScannerFrame?

I'm not entirely sure this is a bug, but it hard to work with the barcode scanner this way.

### Reproduceable Code

```tsx
type CornerPointsSet = { x: number; y: number }[];
type CornerPointsSets = CornerPointsSet[];

// this works:
//const ASPECT_RATIO = 4 / 3;
//const RESOLUTION_WIDTH = 960;
//const RESOLUTION_HEIGHT = 1280;

// this doesn't:
const ASPECT_RATIO = 16 / 9;
const RESOLUTION_WIDTH = 1080;
const RESOLUTION_HEIGHT = 1920;

export default function HomeScreen() {
  const device = useCameraDevice("back");
  const { hasPermission } = useCameraPermission();
  const codeScanner = useCodeScanner({
    codeTypes: [
      "code-128",
      "code-39",
      "code-93",
      "codabar",
      "ean-13",
      "ean-8",
      "upc-e",
      "qr",
    ],
    onCodeScanned,
  });

  const [viewDimensions, setViewDimensions] = useState({
    width: Dimensions.get("window").width,
    height: Dimensions.get("window").height,
  });
  const [cameraDimensions, setCameraDimensions] = useState({
    width: Dimensions.get("window").width,
    height: Dimensions.get("window").height,
  });
  const [verticalPadding, setVerticalPadding] = useState(0);
  const [cornerPointsSets, setCornerPointsSets] = useState<CornerPointsSets>();

  const processingFrame = useRef(false);
  const cameraWidth = useRef(Dimensions.get("window").width);
  const cameraHeight = useRef(cameraWidth.current * ASPECT_RATIO);

[... 194 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pocophone F1 (Android 10), iPhone 7 (Android 15.8.3), Samsung device (currently not with me, will update later)

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-11-29 by maintenance-hans[bot]

daankennes, 2024-11-29:
Guten tag @mrousavy, pinging you as requested by Hans. I included the logs via a link since they were too long for GitHub.

xiaomaodml, 2024-12-27:
Hello, has this problem been solved?

daankennes, 2024-12-27:
As you can see the issue has been closed incorrectly. I tried pinging @mrousavy but I think he's too busy.

xiaomaodml, 2025-01-01:
I'm gonna try it with a frame processor.