margelo/react-native-vision-camera #3314: 🐛 useSkiaFrameProcessor's coordinate system is incorrect and the camera frame is not available

fax1ty, 2024-11-30:
### What's happening?

There are many issues related to this specific problem.
https://github.com/mrousavy/react-native-vision-camera/issues/3301, https://github.com/mrousavy/react-native-vision-camera/issues/3297, https://github.com/mrousavy/react-native-vision-camera/issues/3296, https://github.com/mrousavy/react-native-vision-camera/issues/3169, https://github.com/mrousavy/react-native-vision-camera/issues/3160

[image]

### Reproduceable Code

```tsx
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useSkiaFrameProcessor,
} from "react-native-vision-camera";
import { StyleSheet } from "react-native";
import { useEffect } from "react";
import { Skia } from "@shopify/react-native-skia";

export default function Index() {
  const device = useCameraDevice("back");
  const { hasPermission, requestPermission } = useCameraPermission();

  const frameProcessor = useSkiaFrameProcessor((frame) => {
    "worklet";

    frame.render();

    const centerX = frame.width / 2;
    const centerY = frame.height / 2;

    const paint = Skia.Paint();
    paint.setColor(Skia.Color("red"));
    const rect = Skia.XYWHRect(centerX, centerY, 150, 150);
    frame.drawRect(rect, paint);
  }, []);

  useEffect(() => {
    requestPermission();
  }, []);

  if (!hasPermission || !device) return null;

  return (
    <>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive
        frameProcessor={frameProcessor}
      />
    </>
  );
}
```

### Relevant log output

```shell
11-30 17:33:21.758  5546  5558 W System  : A resource failed to call HardwareBuffer.close.
11-30 17:33:24.289  5546  5546 I RNSkia  : updateAndRelease() failed. The exception above can safely be ignored
11-30 17:33:24.356  5546  5546 E app.warpzone: [SurfaceTexture-0-5546-6] updateAndRelease: EGLConsumer is not attached to an OpenGL ES context
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 1,
  "minZoom": 1,
  "maxExposure": 6,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": false,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -6,
  "name": "10 (BACK) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```

### Device

Android Emulator

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-11-30 by maintenance-hans[bot]

fax1ty, 2024-11-30:
@mrousavy

Caleb-Grepp, 2024-12-04:
same issue 
react-native: 0.76.3
react-native-vision-camera: 4.6.3

agoransson, 2024-12-07:
I'm seeing the same issue.