margelo/react-native-vision-camera #3320: 🔧 Fail to build Visioncamera with Skia frameprocessor

agoransson, 2024-12-05:
### How were you trying to build the app?

1. `npx create-expo-app@latest --template`. (choose "blank (typescript)")
2. `cd my-app`
3. `npx expo install react-native-vision-camera`
4. Add required plugin block to `app.json`
6. `npx expo prebuild`
7. `eas build --platform android --local` (follow instructions to install/generate requirements)
8. `npx expo run:android`

This far it works.

9. `yarn add react-native-worklets-core`
10. `npx expo customize` (generate babel config)
11. Add plugin block to `babel.config.js`
13. `npx expo prebuild`
14. `eas build --platform android --local`
15. `npx expo run:android`

Error complaining about babel plugin missing here.

16. `yarn add -D @babel/plugin-transform-template-literals`
17.  `eas build --platform android --local`
18. `npx expo run:android`

Now it works again, with frame processors.

19. `yarn add @shopify/react-native-skia`
20. `yarn add react-native-reanimated`
21.  `eas build --platform android --local`
22. `npx expo run:android`

Here it fails to start the app with an error.

### Full build logs

```tsx
build log
```

### Project dependencies

```json
{
  "name": "my-app",
  "version": "1.0.0",
  "main": "index.ts",
  "scripts": {
    "start": "expo start",
    "android": "expo run:android",
    "ios": "expo run:ios",
    "web": "expo start --web"
  },
  "dependencies": {
    "@shopify/react-native-skia": "^1.6.0",
    "expo": "~52.0.15",
    "expo-status-bar": "~2.0.0",
    "react": "18.3.1",
    "react-native": "0.76.3",
    "react-native-reanimated": "^3.16.3",
    "react-native-vision-camera": "^4.6.3",
    "react-native-worklets-core": "^1.5.0"
  },
  "devDependencies": {
    "@babel/core": "^7.25.2",
    "@babel/plugin-transform-template-literals": "^7.25.9",
    "@types/react": "~18.3.12",
    "typescript": "^5.3.3"
  },
  "private": true
}
```

### VisionCamera Version

4.6.3

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

agoransson, 2024-12-05:
(base) agoransson@MacBook-Pro Projects % npx create-expo-app@latest --template
✔ Choose a template: › Blank (TypeScript)
✔ What is your app named? … my-app
✔ Downloaded and extracted project files.
> npm install
npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
npm warn deprecated @babel/plugin-proposal-class-properties@7.18.6: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-class-properties instead.
npm warn deprecated @babel/plugin-proposal-nullish-coalescing-operator@7.18.6: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-nullish-coalescing-operator instead.
npm warn deprecated rimraf@2.6.3: Rimraf versions prior to v4 are no longer supported
npm warn deprecated @babel/plugin-proposal-optional-chaining@7.21.0: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-optional-chaining instead.
npm warn deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated @xmldom/xmldom@0.7.13: this version is no longer supported, please update to at least 0.8.*

added 873 packages, and audited 874 packages in 5s

72 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

✅ Your project is ready!

To run your project, navigate to the directory and run one of the following npm commands.

- cd my-app
- npm run android
- npm run ios
- npm run web
(base) agoransson@MacBook-Pro Projects % npx create-expo-app@latest --template
✖ Choose a template: › Default

Please specify the template, example: --template expo-template-blank

(base) agoransson@MacBook-Pro Projects % ls
7county					ai_models				hits_and_targets			model_maker_test			score_scanner_models			vendure-instagram-basic-display
AssetForge				blueprints				katt_vännernas_värld			my-app					temp					vendure-instagram-basic-display_old
CircleDetection				ffmpeg-android				klarna-order				my-spotter-app				texture					vendure-klarna-plugin
PineTimeCompanion			firebase-function			klarna-payments				mysql					tidskriftsbutiken			vendure-postnord-plugin
ScoreKeeperProject			formhelper				media_pipe_tests			react-native-courses			tidskriftsbutiken.com			verasatelje
The-Java-Workshop			fortie					mlkit					react-native-fast-tflite		vendure-docker-compose			wificam_decompiling
(base) agoransson@MacBook-Pro Projects % cd my-app 
(base) agoransson@MacBook-Pro my-app % 
(base) agoransson@MacBook-Pro my-app % 
(base) agoransson@MacBook-Pro my-app % 
(base) agoransson@MacBook-Pro my-app % code .
(base) agoransson@MacBook-Pro my-app % npx expo install react-native-vision-camera

› Installing 1 other package using npm
> npm install --save react-native-vision-camera

added 2 packages, and audited 876 packages in 2s

72 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
› Added config plugin: react-native-vision-camera
(base) agoransson@MacBook-Pro my-app %                                            
(base) agoransson@MacBook-Pro my-app % 
(base) agoransson@MacBook-Pro my-app % npx expo prebuild
› Android package name: com.agoransson.myapp
› Apple bundle identifier: com.agoransson.myapp
✔ Created native directories
✔ Updated package.json
» android: userInterfaceStyle: Install expo-system-ui in your project to enable this feature.
[... 1298 lines cut ...]
├─ whatwg-url@5.0.0
├─ which@2.0.2
├─ wrap-ansi-cjs@7.0.0
├─ xcode@3.0.1
├─ xmlbuilder@14.0.0
├─ xtend@4.0.2
├─ y18n@5.0.8
├─ yargs-parser@21.1.1
└─ yocto-queue@0.1.0
✨  Done in 15.54s.
(base) agoransson@MacBook-Pro my-app % npx expo customize                 
✔ Which files would you like to generate? › babel.config.js
(base) agoransson@MacBook-Pro my-app % npx expo prebuild
✔ Created native directories | reusing /android, /ios
✔ Updated package.json | no changes
» android: userInterfaceStyle: Install expo-system-ui in your project to enable this feature.
✔ Finished prebuild
(base) agoransson@MacBook-Pro my-app % npx expo prebuild
✔ Created native directories | reusing /android, /ios
✔ Updated package.json | no changes
» android: userInterfaceStyle: Install expo-system-ui in your project to enable this feature.
✔ Finished prebuild
(base) agoransson@MacBook-Pro my-app % eas build --platform android --local
We detected that you are building for the "store" distribution. Resolving the environment for environment variables used during the build to "production". Set the "environment" field in the build profile if you want to specify the environment manually.
No environment variables with visibility "Plain text" and "Sensitive" found for the "production" environment on EAS servers.

agoransson, 2024-12-05:
Specified value for "android.package" in app.json is ignored because an android directory was detected in the project.
EAS Build will use the value found in the native code.
✔ Incremented versionCode from 2 to 3.
✔ Using remote Android credentials (Expo server)
✔ Using Keystore from configuration: Build Credentials EtT5LW9y38 (default)
✔ Computed project fingerprint
ANDROID_NDK_HOME environment variable was not specified, continuing build without NDK
[SETUP_WORKINGDIR] Preparing workingdir /var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/1b6859d6-a894-48b6-a952-9d60821d5497
[START_BUILD] Starting build
  "job": {
    "type": "generic",
    "platform": "android",
    "projectRootDirectory": ".",
    "projectArchive": {
      "type": "PATH",
      "path": "/var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-cli-nodejs/62222135-4330-4759-8b3b-540ae3ad29c9.tar.gz"
    },
    "builderEnvironment": {},
    "cache": {
      "disabled": false,
      "paths": [],
      "clear": false
    },
    "updates": {},
    "username": "agoransson",
    "version": {
      "versionCode": "3"
    },
    "experimental": {},
    "mode": "build",
    "triggeredBy": "EAS_CLI",
    "appId": "8d39d093-1565-4ed8-8d11-cac3afd0adb5",
    "initiatingUserId": "ec9df4ab-b561-49c4-ba84-597737e171e7"
  }
[READ_PACKAGE_JSON] Using package.json:
[READ_PACKAGE_JSON] {
  "name": "my-app",
  "version": "1.0.0",
  "main": "index.ts",
  "scripts": {
    "start": "expo start",
    "android": "expo run:android",
    "ios": "expo run:ios",
    "web": "expo start --web"
  },
  "dependencies": {
    "expo": "~52.0.15",
    "expo-status-bar": "~2.0.0",
    "react": "18.3.1",
    "react-native": "0.76.3",
    "react-native-vision-camera": "^4.6.3",
    "react-native-worklets-core": "^1.5.0"
  },
  "devDependencies": {
    "@babel/core": "^7.25.2",
    "@types/react": "~18.3.12",
    "typescript": "^5.3.3"
  },
  "private": true
}
[INSTALL_DEPENDENCIES] Running "yarn install" in /var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/1b6859d6-a894-48b6-a952-9d60821d5497/build directory
[INSTALL_DEPENDENCIES] yarn install v1.22.22
[INSTALL_DEPENDENCIES] warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies caused by unsynchronized lock files. To clear this warning, remove package-lock.json.
[INSTALL_DEPENDENCIES] [1/4] Resolving packages...
[INSTALL_DEPENDENCIES] [2/4] Fetching packages...
[INSTALL_DEPENDENCIES] [3/4] Linking dependencies...
[INSTALL_DEPENDENCIES] warning "react-native > @react-native/codegen@0.76.3" has unmet peer dependency "@babel/preset-env@^7.1.6".
[INSTALL_DEPENDENCIES] warning "expo > babel-preset-expo > @react-native/babel-preset > @react-native/babel-plugin-codegen > @react-native/codegen > jscodeshift@0.14.0" has unmet peer dependency "@babel/preset-env@^7.1.6".
[INSTALL_DEPENDENCIES] [4/4] Building fresh packages...
[INSTALL_DEPENDENCIES] Done in 1.81s.
[... 2304 lines cut ...]
[RUN_GRADLEW] C/C++: In file included from /private/var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/4b076672-6785-4933-89d6-9c5719157634/build/node_modules/@shopify/react-native-skia/android/../cpp/api/JsiSkApi.h:28:
[RUN_GRADLEW] C/C++: In file included from /private/var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/4b076672-6785-4933-89d6-9c5719157634/build/node_modules/@shopify/react-native-skia/android/../cpp/api/JsiSkParagraphBuilder.h:13:
[RUN_GRADLEW] C/C++: /private/var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/4b076672-6785-4933-89d6-9c5719157634/build/node_modules/@shopify/react-native-skia/android/../cpp/api/JsiSkParagraphStyle.h:120:10: warning: 'wstring_convert<std::codecvt_utf8_utf16<char16_t>, char16_t>' is deprecated [-Wdeprecated-declarations]
[RUN_GRADLEW] C/C++:     std::wstring_convert<std::codecvt_utf8_utf16<T>, T> convertor;
[RUN_GRADLEW] C/C++:          ^
[RUN_GRADLEW] C/C++: /Users/agoransson/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/locale:3603:28: note: 'wstring_convert<std::codecvt_utf8_utf16<char16_t>, char16_t>' has been explicitly marked deprecated here
[RUN_GRADLEW] C/C++: class _LIBCPP_TEMPLATE_VIS _LIBCPP_DEPRECATED_IN_CXX17 wstring_convert
[RUN_GRADLEW] C/C++:                            ^
[RUN_GRADLEW] C/C++: /Users/agoransson/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__config:798:41: note: expanded from macro '_LIBCPP_DEPRECATED_IN_CXX17'
[RUN_GRADLEW] C/C++: #    define _LIBCPP_DEPRECATED_IN_CXX17 _LIBCPP_DEPRECATED
[RUN_GRADLEW] C/C++:                                         ^
[RUN_GRADLEW] C/C++: /Users/agoransson/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/include/c++/v1/__config:771:49: note: expanded from macro '_LIBCPP_DEPRECATED'
[RUN_GRADLEW] C/C++: #      define _LIBCPP_DEPRECATED __attribute__((deprecated))
[RUN_GRADLEW] C/C++:                                                 ^
[RUN_GRADLEW] C/C++: 2 warnings generated.
[RUN_GRADLEW] > Task :shopify_react-native-skia:buildCMakeRelWithDebInfo[x86_64]
[RUN_GRADLEW] C/C++: ninja: Entering directory `/private/var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/4b076672-6785-4933-89d6-9c5719157634/build/node_modules/@shopify/react-nat [... cut ...]
[RUN_GRADLEW] C/C++: In file included from /private/var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/4b076672-6785-4933-89d6-9c5719157634/build/node_modules/@shopify/react-native-skia/android/../cpp/api/JsiSkApi.h:28:
[RUN_GRADLEW] C/C++: In file included from /private/var/folders/fl/dg1tx9wn3r9glcwyqkr8shk00000gn/T/eas-build-local-nodejs/4b076672-6785-4933-89d6-9c5719157634/build/node_modules/@shopify/react-native-skia/android/../cpp/api/JsiSkParagraphBuilder.h:13:
[RUN_GR

agoransson, 2024-12-05:
Build logs wouldn't fit... so I added in separate comments.

agoransson, 2024-12-05:
@mrousavy I believe Hans is incorrect in his assessment. The full logs of the entire process could not fit into the ticket, so they are added afterwards. I will edit them to make them nicer looking now too.

3210jr, 2024-12-05:
Same issue here @agoransson. Thought I was going crazy. Please share if you find a workaround, I will too.

agoransson, 2024-12-05:
There is a similar issue in worklets: https://github.com/margelo/react-native-worklets-core/issues/212

3210jr, 2024-12-05:
Installing [4.5.2](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.5.2) version seems to at least build. I am going to keep using this version for now to continue with other work until there is a fix.

agoransson, 2024-12-06:
With Skia frame processor? Would you share your dependencies please? Seems several peopl are claiming that only certain versions of all these packages work together. I still have no luck with changing versions, I end up with failure to build either standard frame processors, skia frame processors, or the tensorflow fast tflite plugin.

agoransson, 2024-12-07:
@3210jr I can't seem to build `4.5.2` at all.

agoransson, 2024-12-07:
I did manage to get it to work finally with Skia frame processor. I think the difference finally was the SDK version had to be set using `expo-build-properties`. 

```
      [
        "expo-build-properties",
        {
          "android": {
            "minSdkVersion": 30,
            "compileSdkVersion": 34,
            "targetSdkVersion": 34,
            "buildToolsVersion": "34.0.0"
          },
          "ios": {
            "deploymentTarget": "15.1"
          }
        }
      ]
```

My dependencies:

```
  "dependencies": {
    "@eva-design/eva": "^2.2.0",
    "@expo-google-fonts/inter": "^0.2.3",
    "@react-native-async-storage/async-storage": "1.23.1",
    "@react-navigation/bottom-tabs": "^7.1.3",
    "@react-navigation/native": "^7.0.13",
    "@react-navigation/stack": "^7.0.18",
    "@shopify/react-native-skia": "^1.7.0",
    "@ui-kitten/components": "^5.3.1",
    "@ui-kitten/date-fns": "^5.3.1",
    "@ui-kitten/eva-icons": "^5.3.1",
    "@ui-kitten/moment": "^5.3.1",
    "expo": "~52.0.11",
    "expo-build-properties": "~0.13.1",
    "expo-dev-client": "~5.0.5",
    "expo-font": "~13.0.1",
    "expo-status-bar": "~2.0.0",
    "firebase": "^11.0.2",
    "install": "^0.13.0",
    "react": "18.3.1",
    "react-native": "0.76.3",
    "react-native-gesture-handler": "~2.20.2",
    "react-native-paper": "^5.12.5",
    "react-native-reanimated": "^3.16.3",
    "react-native-safe-area-context": "4.12.0",
    "react-native-svg": "15.8.0",
    "react-native-vision-camera": "^4.6.3",
    "react-native-worklets-core": "^1.5.0",
    "styled-components": "^6.1.13"
  },
  "devDependencies": {
    "@babel/core": "^7.25.2",
    "@babel/plugin-transform-template-literals": "^7.25.9",
    "@types/react": "~18.3.12",
    "@types/react-native": "^0.73.0",
    "@typescript-eslint/eslint-plugin": "^8.17.0",
    "@typescript-eslint/parser": "^8.17.0",
    "eslint": "^9.16.0",
    "react-native-svg-transformer": "^1.5.0",
    "typescript": "^5.3.3"
  },
```

closed 2024-12-07 by agoransson

agoransson, 2024-12-07:
Also, I think the documentation is incorrect. 

```
const frameProcessor = useSkiaFrameProcessor((frame) => {
  'worklet'
  frame.render()

  const centerX = frame.width / 2
  const centerY = frame.height / 2
  const rect = Skia.XYWHRect(centerX, centerY, 150, 150)
  const paint = Skia.Paint()
  paint.setColor(Skia.Color('red'))
  frame.drawRect(rect, paint)
}, [])
```

Should be:
```
const frameProcessor = useSkiaFrameProcessor((frame) => {
  'worklet'
  frame.render()

  const centerX = frame.width / 2
  const centerY = frame.height / 2
  const rect = SkiaApi.XYWHRect(centerX, centerY, 150, 150)
  const paint = SkiaApi.Paint()
  paint.setColor(SkiaApi.Color('red'))
  frame.drawRect(rect, paint)
}, [])
```