margelo/react-native-vision-camera #3321: 🐛 Camera orientation not working to barcode

Ferreiraguihh, 2024-12-05:
### What's happening?

Basically, I'm using a device like a tablet, and I can't change the orientation of the component, I'm trying to use barcode, I know that in the 'react-native-camera' library, if I implement a rotate -90deg it works.
[image]

### Reproduceable Code

```tsx
const { onCodeScanned } = props;
  const device = useCameraDevice('back');
  const cameraRef = useRef<Camera>(null);

  const codeScanner = useCodeScanner({
    codeTypes: ['ean-13', 'upc-a'],
    onCodeScanned,
  });

  if (!device) {
    return null;
  }

<View>
<Camera
          ref={cameraRef}
          device={device}
          style={StyleSheet.absoluteFill}
          isActive={true}
          codeScanner={codeScanner}
          resizeMode="cover"
        />
</View>
```

### Relevant log output

```shell
.
```

### Camera Device

```json
Device: 'Back'
```

### Device

Tablet Pixel Android 12.0

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

gyanbhartip, 2025-01-07:
@Ferreiraguihh my apologies for going off-topic but could you please give me a brief idea about how do you implement the scanning within the square reticle's bounds. It used to work in RNVC 2 for me but in 4.6.3, I get different frame sizes and unable to implement the same.

jrizmal, 2025-01-16:
Hi @gyanbhartip did you manage to figure this out?
In my case, I'm using the `frame` parameter in `onCodeScanned` to figure out the scanning area, but it always returns landscape dimensions even though I'm previewing in portrait mode. I'd expect `frame` would always be the same as the preview size.

gyanbhartip, 2025-01-16:
@jrizmal I've not been able to figure the "scanning within the reticle bounds" issue.
But the frame orientation is not the same as your preview orientation.
refer: [the official docs on the Frame orientaion](https://react-native-vision-camera.com/docs/api/interfaces/Frame#orientation)