margelo/react-native-vision-camera #3323: 🐛 Camera Preview orientation does not update when screen orientation locks

dawidzawada, 2024-12-06:
### What's happening?

I have a following scenario in my application: 
- I have a pre-recording screen with camera preview - orientation here is locked to portrait
- After pressing Record I'm changing orientation to landscape and I'm moving to "Recording screen"
- In "Recording screen" I have a small camera preview - this camera preview stays at portrait orientation

You can reproduce it by running the example app like so;
1. Run the app
2. Open permissions screen, allow app to use camera
3. Rotate your device for landscape mode (screen is locked in portrait)
4. Press "Go to camera"
5. Screen will be changed to "camera" and rotated to landscape, only preview stays in wrong orientation

All videos are recorded properly, in landscape mode. I'm having issues only with preview orientation.

I'm using expo-screen-orientation for locking orientation

Example app:
https://github.com/dawidzawada/camera-bug

### Reproduceable Code

```tsx
import {lockPlatformAsync, Orientation} from "expo-screen-orientation";

export default function PermissionsScreen() {
    const {requestPermission} = useCameraPermission()

    useEffect(() => {
        void requestPermission()
    }, []);

    useFocusEffect(() => {
        // Set portrait orientation for permissions screen
        const asyncEffect = async () => {
            await lockPlatformAsync({screenOrientationArrayIOS: [
                    Orientation.PORTRAIT_UP,
                ]
            })
        }
        asyncEffect()
    })

    const handlePress = async () => {
        // Set landscape orientation, then go to camera screen
        const asyncEffect = async () => {
            await lockPlatformAsync({screenOrientationArrayIOS: [
                    Orientation.LANDSCAPE_LEFT,
                    Orientation.LANDSCAPE_RIGHT
                ]
            })
            router.push('/camera')
        }
        asyncEffect()

    }

    return (
        <View>
            <Text>Permissions</Text>
            <Button title="Go to camera" onPress={handlePress} />
        </View>
    )
}

export default function CameraScreen() {
    const device = useCameraDevice('back');
    return (
        <View>
            {device && <Camera device={device} style={{width:300, height: 160}} isActive={true} />}
        </View>
    )
}
```

### Relevant log output

```shell
16:19:25.549: [info] 📸 VisionCamera.didSetProps(_:): Updating 19 props: [onPreviewStarted, preview, onViewReady, onPreviewOrientationChanged, cameraId, isMirrored, onError, width, onShutter, onInitialized, onStarted, enableBufferCompression, onStopped, height, onCodeScanned, onOutputOrientationChanged, onPreviewStopped, enableFrameProcessor, isActive]
16:19:25.549: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: portrait...
16:19:25.549: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: portrait...
16:19:25.550: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
16:19:25.550: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true, locationChanged: true)
16:19:25.551: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Input Device...
16:19:25.551: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
16:19:25.553: [info] 📸 VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
16:19:25.553: [info] 📸 VisionCamera.configureOutputs(configuration:): Configuring Outputs...
16:19:25.553: [info] 📸 VisionCamera.configurePreviewOrientation(_:): Updating Preview rotation: portrait...
16:19:25.553: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: portrait...
16:19:25.553: [info] 📸 VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
16:19:25.553: [info] 📸 VisionCamera.setTargetOutputOrientation(_:): Setting target output orientation from device to device...
16:19:25.662: [debug] 📸 VisionCamera.deviceOrientation: Device Orientation changed from portrait -> landscapeLeft
16:19:25.662: [info] 📸 VisionCamera.configureOutputOrientation(_:): Updating Outputs rotation: landscapeRight...
16:19:25.824: [info] 📸 VisionCamera.init(frame:session:): Preview Layer started previewing.
16:19:25.824: [info] 📸 VisionCamera.configure(_:): Beginning AudioSession configuration...
16:19:25.824: [info] 📸 VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
16:19:25.825: [info] 📸 VisionCamera.configure(_:): Committed AudioSession configuration!
16:19:25.829: [info] 📸 VisionCamera.configure(_:): Beginning Location Output configuration...
16:19:25.830: [info] 📸 VisionCamera.configure(_:): Finished Location Output configuration!
```

### Camera Device

```json
"id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "minZoom": 1,
  "neutralZoom": 1,
  "position": "back",
  "name": "Back Camera",
  "maxZoom": 123.75,
  "minFocusDistance": 12,
  "maxExposure": 8,
  "minExposure": -8,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "isMultiCam": false,
  "hardwareLevel": "full",
  "sensorOrientation": "portrait",
  "formats": [],
  "hasFlash": true,
  "supportsRawCapture": false,
  "supportsLowLightBoost": false,
  "hasTorch": true
}
```

### Device

iPhone 12

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

dawidzawada, 2024-12-06:
I'm using latest `expo-screen-orientation` and there's nothing additional that could affect this preview behavior - have a look at the example app.

motoshira, 2024-12-11:
We are also using expo-screen-orientation, and we are experiencing abnormal rotation of the preview on iOS devices.

## What's happening?

You can reproduce it by running the example app like so:

1. Run `yarn ios` to launch the app.
2. Grant camera permissions.
3. Tilt the device vertically or horizontally.
4. The preview rotates 90 degrees from the correct orientation.

## Reproduceable Code

We are using expo-sensors to detect screen tilt and expo-screen-orientation to lock the screen orientation.
For the screen lock, we use setTimeout to introduce a delay, simulating cases where in-app processing takes time. The issue does not seem to occur without this setTimeout.

### App.js

```
import React, { useEffect, useState } from "react";
import { StyleSheet, Text, View, Platform } from "react-native";
import {
  Camera,
  useCameraPermission,
  useCameraDevice,
} from "react-native-vision-camera";
import {
  Orientation,
  OrientationLock,
  getOrientationLockAsync,
  lockAsync,
} from "expo-screen-orientation";
import { Accelerometer } from "expo-sensors";

const getNextOrientationFromAccelerometerEvent = (x, y, z) => {
  // when the device is flat, keep the current orientation
  if (Math.abs(z) > 0.5) {
    return undefined;
  }
  if (x >= 0.75) {
    return Orientation.LANDSCAPE_LEFT;
  }
  if (x <= -0.75) {
    return Orientation.LANDSCAPE_RIGHT;
  }
  if (y >= 0.75) {
    return Orientation.PORTRAIT_UP;
  }
  if (y <= -0.75) {
    return Orientation.PORTRAIT_DOWN;
  }
  return undefined;
};

// custom hook to get the current orientation of the device
const useCurrentOrientation = () => {
  const [orientation, setOrientation] = useState(Orientation.PORTRAIT_UP);
  useEffect(() => {
    Accelerometer.setUpdateInterval(16);
    console.log("register subscription");
    const subscription = Accelerometer.addListener((event) => {
      const nextOrientation = getNextOrientationFromAccelerometerEvent(
        event.x,
        event.y,
        event.z,
      );
      if (nextOrientation) {
        setOrientation(nextOrientation);
      }
    });
[... 231 lines cut ...]
  "sensorOrientation": "portrait",
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "supportsFocus": true,
  "neutralZoom": 1,
  "supportsRawCapture": false,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "position": "back",
  "formats": [],
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "name": "Back Camera",
  "hasFlash": true,
  "minExposure": -8,
  "minFocusDistance": 12,
  "hasTorch": true
}
```

## VisionCamera Version

4.6.3

https://github.com/user-attachments/assets/cae88d8e-8a5f-4991-b651-c65b7bc0d8e8

akhlivnoy, 2024-12-18:
After several attempts I found a solution that helped me:
```
export const CameraScreen = () => {
  const device = useCameraDevice('back');
  const isFocused = useIsFocused();
  const [orientationChanged, setOrientationChanged] = useState(false);
  const camera = useRef<Camera>(null);

  useEffect(() => {
    ScreenOrientation.addOrientationChangeListener(() => {
      setOrientationChanged(true);
    });
    ScreenOrientation.lockAsync(ScreenOrientation.OrientationLock.LANDSCAPE_RIGHT);

    return () => {
      ScreenOrientation.removeOrientationChangeListeners();
      ScreenOrientation.lockAsync(ScreenOrientation.OrientationLock.PORTRAIT_UP);
    };
  }, []);

  return (
    <View style={{ flex: 1 }}>
      {orientationChanged && device && (
        <Camera
          ref={camera}
          photo
          video
          isActive={isFocused}
          device={device}
          style={StyleSheet.absoluteFill}
        />
      )}
    </View>
  );
};
```
I hope it will help you too.

motoshira, 2024-12-20:
It appears that the OS screen rotation animation is performed after `OrientationManager.onOrientationChanged` fires.
The following patch solved the problem for the time being.
```
diff --git a/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift b/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift
index 413f79d..0d509c7 100644
--- a/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift
+++ b/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift
@@ -103,7 +103,7 @@ final class OrientationManager {
     UIDevice.current.beginGeneratingDeviceOrientationNotifications()
     NotificationCenter.default.addObserver(self,
                                            selector: #selector(onDeviceOrientationChanged),
-                                           name: UIDevice.orientationDidChangeNotification,
+                                           name: UIApplication.didChangeStatusBarFrameNotification,
                                            object: nil)
   }
 
@@ -113,7 +113,7 @@ final class OrientationManager {
     // Stop UI-orientation updates
     UIDevice.current.endGeneratingDeviceOrientationNotifications()
     NotificationCenter.default.removeObserver(self,
-                                              name: UIDevice.orientationDidChangeNotification,
+                                              name: UIApplication.didChangeStatusBarFrameNotification,
                                               object: nil)
   }
```

dawidzawada, 2024-12-20:
I've tested @motoshira patch and it is working, thanks for sharing it! 🔥 I haven't noticed any bugs, camera preview seems to flip properly every time, video output is still recorded in proper orientation.

cc @mrousavy

vladvlasov256, 2025-01-26:
Thanks @motoshira! Your workaround worked for me.

---

I’m curious about the root cause of this issue. A typical camera app uses several orientation handlers, including at a minimum:
- This package
- react-native-screens, as used in react-navigation.

Could the root cause be related to the order in which the events are handled?

eduardorfreitas93, 2025-06-26:
https://github.com/mrousavy/react-native-vision-camera/issues/3051#issuecomment-3010647771

kculmback-eig, 2025-07-25:
> It appears that the OS screen rotation animation is performed after `OrientationManager.onDeviceOrientationChanged` fires. The following patch solved the problem for the time being.
> 
> ```
> diff --git a/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift b/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift
> index 413f79d..0d509c7 100644
> --- a/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift
> +++ b/node_modules/react-native-vision-camera/ios/Core/OrientationManager.swift
> @@ -103,7 +103,7 @@ final class OrientationManager {
>      UIDevice.current.beginGeneratingDeviceOrientationNotifications()
>      NotificationCenter.default.addObserver(self,
>                                             selector: #selector(onDeviceOrientationChanged),
> -                                           name: UIDevice.orientationDidChangeNotification,
> +                                           name: UIApplication.didChangeStatusBarFrameNotification,
>                                             object: nil)
>    }
>  
> @@ -113,7 +113,7 @@ final class OrientationManager {
>      // Stop UI-orientation updates
>      UIDevice.current.endGeneratingDeviceOrientationNotifications()
>      NotificationCenter.default.removeObserver(self,
> -                                              name: UIDevice.orientationDidChangeNotification,
> +                                              name: UIApplication.didChangeStatusBarFrameNotification,
>                                                object: nil)
>    }
> ```

This works great for a development build, but doesn't seem to work for us in non-development builds. We are working in a monorepo with pnpm, which is notorious for its issues with getting Expo to work within it though.

@motoshira @dawidzawada I know this issue is a bit old, but is the above path still working for you two in production builds?