margelo/react-native-vision-camera #3332: 🐛 FATAL EXCEPTION: mrousavy/VisionCamera.video

looooool11111, 2024-12-17:
### What's happening?

When I exited the scanning page, a crash occurred
[image]

### Reproduceable Code

```tsx
let device = useCameraDevice('front');
  // 添加设备检查
  if (!device) {
    console.error('未找到相机设备');
    return null; // 或者返回一个占位符组件
  }
  const {hasPermission, requestPermission} = useCameraPermission();
  const isFocused = useIsFocused();
  const [isActive, setIsActive] = useState(false);
  const cameraRef = useRef<VisionCamera | null>(null);
  const [isForeground, setIsForeground] = useState(true);
  const targetFps = 5;
  const screenAspectRatio = 16 / 9;
  const format = useCameraFormat(device, [
    {fps: targetFps},
    {videoAspectRatio: screenAspectRatio},
    {photoAspectRatio: screenAspectRatio},
  ]);
  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: codes => {
      console.log('Scanned codes:', codes);

      if (codes[0]?.value) {
        navigation.navigate('Index', {
          code: codes[0]?.value as string,
        });
      }
    },
  });

  const renderCamera = () => {
    return isActive && device ? (
      <Camera
        style={styles.absoluteFill}
        device={device}
        format={format}
        isActive={isActive}
        enableZoomGesture={false}
        ref={cameraRef}
        resizeMode="cover"
        exposure={0}
        onStarted={() => console.log('Camera started!')}
        onStopped={() => console.log('Camera stopped!')}
        photo={true}
        video={false}
        codeScanner={codeScanner}
      />
    ) : (
      <></>
    );
  };
```

### Relevant log output

```shell
FATAL EXCEPTION: mrousavy/VisionCamera.video
                                                                                                    Process: com.airdoc.visiontrain, PID: 13485
                                                                                                    java.lang.NullPointerException: Attempt to invoke virtual method 'java.nio.ByteBuffer android.media.Image$Plane.getBuffer()' on a null object reference
                                                                                                    	at com.google.mlkit.vision.common.InputImage.zzb(com.google.mlkit:vision-common@@17.3.0:15)
                                                                                                    	at com.google.mlkit.vision.common.InputImage.fromMediaImage(com.google.mlkit:vision-common@@17.3.0:1)
                                                                                                    	at com.mrousavy.camera.core.CodeScannerPipeline._init_$lambda$4(CodeScannerPipeline.kt:53)
                                                                                                    	at com.mrousavy.camera.core.CodeScannerPipeline.$r8$lambda$J9uS5V_QrF4Y46dODUvYa6w1Wm0(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.CodeScannerPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:4)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:916)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:201)
                                                                                                    	at android.os.Looper.loop(Looper.java:288)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
{"autoFocusSystem": "contrast-detection", "fieldOfView": 0, "maxFps": 30, "maxISO": 0, "maxZoom": 4, "minFps": 1, "minISO": 0, "photoHeight": 720, "photoWidth": 1280, "pixelFormats": ["yuv", "native"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": ["off", "off"], "videoWidth": 1280}
```

### Device

Android 13

### VisionCamera Version

3.9.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

looooool11111, 2024-12-17:
Additional error messages

[image]

mrousavy, 2025-02-17:
```
                                                                                                    java.lang.NullPointerException: Attempt to invoke virtual method 'java.nio.ByteBuffer android.media.Image$Plane.getBuffer()' on a null object reference
                                                                                                    	at com.google.mlkit.vision.common.InputImage.zzb(com.google.mlkit:vision-common@@17.3.0:15)
                                                                                                    	at com.google.mlkit.vision.common.InputImage.fromMediaImage(com.google.mlkit:vision-common@@17.3.0:1)
                                                                                                    	at com.mrousavy.camera.core.CodeScannerPipeline._init_$lambda$4(CodeScannerPipeline.kt:53)
```

This is happening inside MLKit. Apparently a bug in Google's library, not VisionCamera. Sorry, nothing I can do - please create a bug report at MLKit QR Code Scanner repo.

closed 2025-02-17 by mrousavy