margelo/react-native-vision-camera #3333: 🐛 Camera component intercepts first touch on a trigger button, requiring second tap to take a photo

nahooni0511, 2024-12-19:
### What's happening?

I'm experiencing an issue where the first attempt to press my camera shutter button (to take a photo) is consistently ignored when using react-native-vision-camera. The button appears on the screen, but the very first tap does not trigger the onPress event. Instead, the second tap and all subsequent taps work as expected.

Key Points:

First tap ignored: The initial press on the capture button has no effect, as if the touch event is intercepted.

Subsequent taps recognized: After the first "ignored" tap, every following tap works fine, triggering the capture as intended.

No overlapping UI: I've tried ensuring the camera preview and the capture button do not overlap. For example, the camera component is placed above, and the button is placed in a separate view below. The issue persists even with non-overlapping layouts.

Removing the Camera fixes it: If I remove the <Camera> component completely, the first tap on the button is recognized immediately, indicating the camera component might be consuming or blocking the initial touch.

Other attempts:

Used Pressable, TouchableOpacity, and various pointerEvents settings (e.g., pointerEvents="none" on the camera container) with no success.
Disabled tap-to-focus and zoom gestures (enableTapToFocus={false}, enableZoomGesture={false}) on the camera, but the issue remains.
Environment:

React Native: [version]
react-native-vision-camera: [version]
Platform: Android (issue observed here; not thoroughly tested on iOS)
Device/Emulator: [details]
Expected Behavior:
The first tap on the capture button should immediately trigger the onPress event and capture the photo.

Actual Behavior:
The first tap is completely ignored, and the second tap finally triggers the capture.

Request for Help:
I would appreciate any guidance or solutions to prevent the camera component from intercepting the initial touch, allowing the first tap on the capture button to be recognized.

### Reproduceable Code

```tsx
import React, { useRef, useState } from 'react';
import { View, Pressable, Text, Alert, StyleSheet } from 'react-native';
import { Camera, useCameraDevice } from 'react-native-vision-camera';

export default function CameraScreen() {
  const camera = useRef(null);
  const device = useCameraDevice('back');
  const [isInitialized, setIsInitialized] = useState(false);

  const takePicture = async () => {
    if (camera.current) {
      try {
        const photo = await camera.current.takePhoto();
        Alert.alert('Photo taken', photo.path);
      } catch (e) {
        console.warn(e);
      }
    }
  };

  return (
    <View style={styles.container}>
      {device && (
        <Camera
          ref={camera}
          style={styles.camera}
          device={device}
          isActive={true}
          enableTapToFocus={false}
          enableZoomGesture={false}
          onInitialized={() => setIsInitialized(true)}
        />
      )}
      
      {/* Button placed outside camera area */}
      <View style={styles.buttonContainer}>
        <Pressable onPress={takePicture} style={styles.button}>
          <Text style={{color: '#fff'}}>Capture</Text>
        </Pressable>
      </View>
    </View>
  );
}

const styles = StyleSheet.create({
  container: { flex: 1 },
  camera: {
    width: '100%',
    height: 400, // Non-overlapping area
  },
  buttonContainer: {
    flex: 1,
    backgroundColor: 'red', // Separate area below camera
    justifyContent: 'center',
    alignItems: 'center'
  },
  button: {
    padding: 20,
    backgroundColor: 'blue'
  }
});
```

### Relevant log output

```shell
.
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 1,
  "minZoom": 1,
  "maxExposure": 6,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": false,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 999.999985098839,
  "minExposure": -6,
  "name": "10 (BACK) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```

### Device

Android emulator

### VisionCamera Version

4.6.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Rohit10701, 2025-03-18:
any update regarding this ?

sontruongna, 2025-06-20:
I was wondering if there are any updates regarding this?

Rohit10701, 2025-06-20:
It happens only in androids simulator works fine in physical device